MarlinFirmware/Marlin #6087: Marlin 1.0.2.2

TerryD15, 2017-03-21:
I have set SAFE_HOMING in Marlin 1.0.2.2 and set the position of the Z axis at the centre of the bed for safe homing.  According to the description in the comments in Config,h  SAFE_HOMING is supposed to prevent the Z axis from trying to home in an unsafe position i.e. off the bed. 

However my Z axis tries to home off the bed - I should say that it is an inductive sensor type - even when homing all axis with a G28 command issued.

This is my configuration.h for the routine can anyone spot a problem or suggest a solution-

  #define Z_SAFE_HOMING   // This feature is meant to avoid Z homing with probe outside the bed area.
                          // When defined, it will:
                          // - Allow Z homing only after X and Y homing AND stepper drivers still enabled
                          // - If stepper drivers timeout, it will need X and Y homing again before Z homing
                          // - Position the probe in a defined XY point before Z Homing when homing all axis (G28)
                          // - Block Z homing only when the probe is outside bed area.

  #ifdef Z_SAFE_HOMING

    #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28)
    #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28)

  #endif

Roxy-3D, 2017-03-21:
@thinkyhead   I saw something similar to this yesterday.   My Configuration.h files got lost and I had to recreate them.   I was bringing up UBL on a Cartesian machine.  But I had some of these enabled.  It seems we could use some more SanityChecks.h (and better comments) to catch this situation.    If  you are bringing up a Cartesian machine, these should be commented out.

```cpp
// The center of the bed is at (X=0, Y=0)
//#define BED_CENTER_AT_0_0

// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
//#define MANUAL_X_HOME_POS 100
//#define MANUAL_Y_HOME_POS 100
//#define MANUAL_Z_HOME_POS 20
```

TerryD15, 2017-03-22:
Hi,

My Configuration file indeed had *define_MANUAL_X_HOME_POS 100 etc (for Y and Z pos. as well) un-commented but when I commented them out it made no difference.  Probe was still at X0 and Y0 when trying to probe - I had to use a flat metal spatula under the probe to 'home' it before it crashed into the bed.

Here is my Configuration.h file (I have deliberately not included 'Additional Features' as I have not made any changes from original):

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// @section info

#ifdef USE_AUTOMATIC_VERSIONING
  #include "_Version.h"
#else
  #include "Default_Version.h"
#endif

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_DISTRIBUTION_DATE // will be shown during bootup in line 2

#define STRING_VERSION_CONFIG_H __DATE__ "15_03_2017_08:00am " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(Terry Diig March 2017)" // Who made the changes.

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
#define BAUDRATE 115200

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD 33
#endif

// Define this to set a custom name for your generic Mendel,
#define CUSTOM_MENDEL_NAME "My DIY Prusa i3"

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
// You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
// #define MACHINE_UUID "00000000-0000-0000-0000-000000000000"

[... 410 lines cut ...]
//#define MANUAL_Z_HOME_POS 402 // For delta: Distance between nozzle and print surface after homing.

//// MOVEMENT SETTINGS
#define NUM_AXIS 4 // The axis order in all axis related arrays is X, Y, Z, E
#define HOMING_FEEDRATE {50*60, 50*60, 4*60, 0}  // set the homing speeds (mm/min)

// default settings

#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,800.7}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {500, 500, 5, 25}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {9000,9000,100,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.

#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000   // X, Y, Z and E max acceleration in mm/s^2 for retracts

// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).
// For the other hotends it is their distance from the extruder 0 hotend.
// #define EXTRUDER_OFFSET_X {0.0, 20.00} // (in mm) for each extruder, offset of the hotend on the X axis
// #define EXTRUDER_OFFSET_Y {0.0, 5.00}  // (in mm) for each extruder, offset of the hotend on the Y axis

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK                20.0    // (mm/sec)
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
#define DEFAULT_EJERK                 5.0    // (mm/sec)

Roxy-3D, 2017-03-22:
Here is my Configuration.h file (renamed as a .txt file so I can attach it).   I have the safe homing turned on and have it home and probe at the center of the bed.   Do a diff and see how I have things set up.   Or, alternatively, you could just cross your settings over to it.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/860300/Configuration.txt)

TerryD15, 2017-03-22:
Thanks, I will do that later today.  I will test my machine with your file but of course with adaptations e.g. I have a geared extruder rather than the direct drive of the Folger.

For interest my machine is of the old Box Frame construction and the axis and motor mounts of my own design using slot and tab system made from plywood.  I originally intended to replace these with printed parts but found this unnecessary as my originals have proved to be surprisingly accurate and robust, and the plywood components appear to be able to partially absorb vibration etc.

TerryD15, 2017-03-22:
Hi Roxy-3D,
I've looked at your configuration.h file, which version of Marlin is this from?

TerryD15, 2017-03-22:
In 1.0.2-2 Marlin Configuration .h I cannot see where I define my Z axis inductive sensor for both homing and bed levelling, perhaps that is the problem,  Can anyone tell me where this may be?

closed 2017-11-12 by thinkyhead