MarlinFirmware/Marlin #6091: Q: Understanding UBL mesh offset

WheresWaldo, 2017-03-21:
I have read the wiki page several times and I do not see how to use G29 P6. I have a machine that probes consistently 1 mm off of true Z0, I know this and it is inherent in the design and cannot be changed. In the past I have used M565 then M851 then G29 S4 Z1.0 depending on which leveling technique is being used. Now with G29 P6 in UBL my understanding is that this is the proper command to Z Offset (for lack of a better term) the entire grid upward, I just can't figure out the actual Syntax and wether or not I need the C parameter.

Either I cannot read properly or there is still a bit of work that needs to be done to the Wiki page. any assistance would be appreciated.

Roxy-3D, 2017-03-21:
Please be aware that this functionality is still 'Bleeding Edge'.   The logic is super simple.  And I tested that I could move the entire mesh up and down.   But I don't use it.   I have done a lot of testing of UBL but I have always set my Z-Offset as close to 0.000mm as possible by adjusting my #define Z_PROBE_OFFSET_FROM_EXTRUDER to be exactly accurate.   If you can do that...  You will be on much better tested ground.

But with that said...  Shifting the entire mesh up or down should do what you need.  The syntax would be G29 P6 C-1.234   to move the entire mesh down 1.234 mm.     You can verify that by doing a G29 O before and after you do the command.   

Please let us know if it is doing what you need it to do.

> Either I cannot read properly or there is still a bit of work that needs to be done to the Wiki page. Any assistance would be appreciated.

Yes.   And the coding is not complete yet either.

Bob-the-Kuhn, 2017-03-21:
Can you point me to the Wiki page?  I can't find it.

WheresWaldo, 2017-03-21:
Sorry the Wiki I mentioned wasn't a wiki page, it is the readme for the UBL branch. That is all the documentation that I have found.

WheresWaldo, 2017-03-21:
@Roxy-3D I will play with both methods of adjustment

closed 2017-03-21 by WheresWaldo

Roxy-3D, 2017-03-21:
However, there is a user editable version at: https://github.com/MarlinFirmware/MarlinDocumentation/edit/master/_features/auto_bed_leveling.md