MarlinFirmware/Marlin #6092: Auto leveling bed ignoring z_min

joseshiru, 2017-03-22:
Hi guys,

Yesterday I updated Marlin to the latest version.

My printer is a Delta mini and it has a proximity sensor.

When I'm running G29, the extruder is going down until it heat the aluminum plate, which should not happen because few mm before it touches the plate, the z_min is triggered (I already checked it).

So, I'm getting lost, seems like doesn't matter if the z_min is triggered, the extruder continues going down until head the plate.

Do you have any idea why this is happening?

Many thanks.

joseshiru, 2017-03-22:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/861084/Configuration.txt)

joseshiru, 2017-03-22:
BTW is was working fine with the old version

Tannoo, 2017-03-22:
Is this kosher for a delta?

`#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

joseshiru, 2017-03-22:
I guess....

I can only choose between these two:
**Z_MIN_PROBE_ENDSTOP**: makes the z_min_probe appear with m119 and it's never triggered when the probe touches the bed, the probe is only triggering the z_min.

**Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN**: when running m119 is not displaying the z_min_probe but only z_min . But is not working neither.

Bob-the-Kuhn, 2017-03-22:
Try commenting out **#define ENDSTOPPULLUPS** and, assuming that you've tied the sensor to **Z_MIN_PIN**,  leave **Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN** enabled.

I expect that you're using a resistor divider network to interface the sensor to your controller.  Having that pullup enabled interferes with the signal level.

We've run into this once before.  If I remember correctly the default state of **#define ENDSTOPPULLUPS** has changed so it's easy to miss when switching to a newer vwersion.

You might also want to enable **PINS_DEBUGGING** in **Configuration_adv.h** and then issue **M43 E1** before testing the new setup.  That command will tell you whenever an endstop changes state (and toggle the system LED).

joseshiru, 2017-03-24:
Bob, thanks for the info, I'll try it. 👍

joseshiru, 2017-04-03:
Guys I'll close this issue as seems to be a problem with my settings.

Thanks for the help

closed 2017-04-03 by joseshiru