MarlinFirmware/Marlin #6093: question: how to 'disable' heated bed?

filipgoc, 2017-03-22:
I have a heated bed. 
Sometimes, I don't want to use it (impatience). 
But I already have my gcode with M140 S50 or something... 

### How would you go around disabling the heated bed (without incurring min temperature error :-) through the LCD? 

Thank you for your wise thoughts.

psavva, 2017-03-22:
Disable any settings in your Slicer that generates M140

filipgoc, 2017-03-22:
@psavva ... the important part for me is "THOUGH THE LCD" - by which I mean without changing the gcode at all.

filipgoc, 2017-03-22:
As in I want the gcode to have M140 ___, but the printer should disregard it.

Sebastianv650, 2017-03-22:
As this is a very special need, I think you will have to code it yourself. Marlin doesn't have an option for that.

filipgoc, 2017-03-22:
@Sebastianv650 I agree. 

I am simply trying to see what kind of approach you would guys take. In what ways add what kind of code... that's all. Crowdsourcing ideas for implementation.

psavva, 2017-03-22:
@Sebastianv650, what I suggest will basically allow you to print, regardless if M140 exists or not.  

Did I misunderstand the question?

filipgoc, 2017-03-22:
@psavva I think you may have misunderstood the question. 
- You suggested to change slicer settings to get rid of M140 (and M190).
- but my question is How to disregard M140 _while keeping it in the gcode._

psavva, 2017-03-22:
@filipgoc 
Do the following:
1) Disconnect Power from the heated bed (be safe)
2) Set the Bed Thermistor to 0
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration.h#L257
3) Comment out Thermal Protection for the Bed
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration.h#L406

This will allow M140 in the GCode, without ever allow thermal protection for the bed to halt the printer.

No menu on the LCD, or other way I can think of can work without actually implementing something specifically to what you're asking for.

Sebastianv650, 2017-03-22:
In fact it is realy much more easy to disable heat bed in slicer or just wait 1-2 minutes until it's hot :-D
But if your realy want to do it, just create a LCD menu entry which switches a boolean variable. Check for this one in all commands which would set a bed temperature and ignore them if true.

filipgoc, 2018-02-06:
@thinkyhead One of these days.