MarlinFirmware/Marlin #6094: Manual and Automatic Fan Control Issue

chevellebro1, 2017-03-22:
I have recently updated to 1.1.0 RC8 from 1.0.2 and the fan operations seem to be inverted. The extruded fans are now always on and it does it with both manual and automatic control. To turn the fan off I have to run the M106 on command. I can't seem to figure out why this is backward?

chevellebro1, 2017-03-23:
Could this be a wiring error? I've tried changing pins but whatever pin I assign to the fan controller is always on by default

Roxy-3D, 2017-03-23:
Yes.  It could be a wiring error.  Or a number of other things.

What type of electronics are you seeing the problem on?

chevellebro1, 2017-03-23:
I'm using Ramps 1.4 with an external opto-isolated relay to control the cooling fans

thinkyhead, 2018-02-06:
Presuming solved.

closed 2018-02-06 by thinkyhead