MarlinFirmware/Marlin #6095: When using G29 P6 C1.234 nozzle crashes into bed between each layer

WheresWaldo, 2017-03-22:
So, in my experiment with UBL is seem to have run into an issue setting a MESH offset using the G29 P6 C0.6 (the 0.6 was actual measurement of offset and yes the trigger happens after the nozzle hits the bed) command. This has been tested with multiple models sliced with the newest Prusa Slic3r and with S3D. every layer change seems to be preceded with an extruder movement that tries to push the nozzle into the bed. then it will return to the absolute layer height and continue printing only to try to push itself back into the bed before the next layer starts. I only let the prints go up a few layers (>10) to see how it worked. Layer height did not seem to mitigate or compound the issue, tested with fixed layer heights and variable layer heights the behavior is the same.

I know this is experimental code and your focus is integrating all the pieces into the main Marlin codebase, but I thought you would want to know this.

Any additional information you need I would be happy to provide as I am playing with this all day trying to see how it works.

WheresWaldo, 2017-03-22:
Just to test a theory I changed the offset to a negative number and the behavior does not manifest itself. But of course for my mechanical setup I can't use it that way.

WheresWaldo, 2017-03-22:
Okay, I am using today's RCBug Fix and I am not sure what is going on here is the output of G29 P5 C
```Send: G29 P5 C
Recv: Unified_Bed_Leveling_EEPROM_start=544
Recv: # of samples: 100
Recv: Mean Mesh Height: 0.600000
Recv: Standard Deviation: 0.228430
Recv: ok
```
It appears to be working printing a Deprime widget right now I tried to place the offset in firmware in Z_PROBE_OFFSET_FROM_EXTRUDER but that doesn't report anywhere with any of the G29 commands, so I also tried shoving it is M851 and that doesn't show in any UBL G29 commands either. If necessary I can enable DEBUG_LEVELING_FEATURE.

Roxy-3D, 2017-03-22:
> G29 P5 C   

Probably...   You shouldn't be using G29 P5 pr G20 P6 in your initial setup of the system.  They have their uses but not for the initial setup.    The best thing to do is to incrementally adjust 
- Z_PROBE_OFFSET_FROM EXTRUDER in Configuration.h
- load the firmware
- Do a M502 and M500 pair to get that saved to EEPROM
- Do a G28 and give a G1 X100 Y100 Z.25 and verify the nozzle is .25mm from the bed.
- if not...  Start at the top and repeat
- if you are very close to .25mm from the bed, then continue with
- G29 P1 R O
- G29 A
- G29 S1
- G26 P C O3.5

And if the Mesh Validation pattern looks good in the center of the bed...  Go ahead and print something.
You can fine to the bad spots of the Mesh Validation pattern later.

WheresWaldo, 2017-03-22:
Sorry about all the questions, UBL isn't as intuitive to me as MESH was and I have been using it since Edward added it. I will try the above instructions and report back. I was going to solve this by manually doing the probing but I am not understanding how to get G29 P2 B past the first probe point but that is for another day.

Roxy-3D, 2017-03-22:
> I was going to solve this by manually doing the probing but I am not understanding how to get G29 P2 B past the first probe point but that is for another day.

:)   That is easy!!!!   It is just like the other commands in the G29 family.   If you want it to do a bunch of them, you add an R.   If you want it to do just 3 you add an R3.     But really...   if you have a Z-Probe, you don't want to be messing around with the G29 P2 Manual Probing.   It is too time consuming and it is not very accurate because it is hard to tell the exact nozzle position.    

It is just better to fill in the un-probed locations with 'reasonable' values and use the Mesh Validation pattern to tell you which way to edit them.

WheresWaldo, 2017-03-22:
Okay, I must be having an issue. The only Z_PROBE_OFFSET_FROM_EXTRUDER I see in Configuration.h is in the Z Probe Options settings around line 580. No matter what I set it to I get exactly the same measurement when doing the G1 X100 Y100 Z.25 Even if I make it incredibly large or incredibly small it still measures the exact same. It is as if UBL ignores that particular #define. I am starting to think there is something I have enabled that is conflicting with that #define.

Roxy-3D, 2017-03-22:
- Z_PROBE_OFFSET_FROM EXTRUDER in Configuration.h
- load the firmware
- Do a M502 and M500 pair to get that saved to EEPROM

THE LAST LINE IS VERY IMPORTANT.  Marlin will use the settings stored in EEPROM unless you tell it to use something else.   Even though you modified your Configuration.h file, unless you store those numbers into the EEPROM...  You really have not made a change to how things are going to operate.

WheresWaldo, 2017-03-22:
Roxy, this isn't my first rodeo, ;) 

I have been doing that every flash, EEPROM caught me when I first started with 1.1.0, I learned that lesson a long time ago.

Just to make sure it isn't something else, since I had been playing with UBL here is what I just did, made sure the mesh was zeroed G29 P0 then redid the M502 and M500 then G28 then the G1 and it is still off exactly the same amount it was before, It still appears to be ignoring Z_PROBE_OFFSET_FROM_EXTRUDER. Are there sections of the Configuration.h you would like to see? I can also upload the whole thing to a MS OneDrive link for you to download.

WheresWaldo, 2017-03-22:
Here is what I have found out but cannot verify at the moment. Since I didn't want to fight with nearly all the settings I have enabled I started clean and only enabled what was necessary. The only apparent difference is that #define Z_MIN_PROBE_REPEATABILITY_TEST is disabled now and was enabled before. I managed to get the offset to subsequently work, it is now in the process Printing the validation grid.

Bob-the-Kuhn, 2017-03-23:
My nozzle is also up in the air.

I'll try disabling Z_MIN_PROBE_REPEATABILITY_TEST and doing Roxy's method.

I thought that maybe these were getting in the way of the big offsets but then I saw your latest post.
```
#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20
```

----

@Roxy-3D - do we really need to FLASH the new value every time?  I was expecting repeating **M851,  G28, G0 Z0.25** would get you to a value you could commit to the config file and EEPROM.

Roxy-3D, 2017-03-23:
> do we really need to FLASH the new value every time? I was expecting repeating M851, G28, G0 Z0.25 would get you to a value you could commit to the config file and EEPROM.

Probably not...   But it is important the machine homes and knows its true position.

Here is the thing...  Depending upon the shape of your bed, where the nozzle is positioned after the home operation can be misleading.   The homing is done with Z Height correction turned off (for obvious reasons).   Until the nozzle moves in either X and/or Y, you won't get a correction factor added in.      If you think your nozzle is at the wrong height...  Just tell it to move 1mm from where it is.  You may see your nozzle go up or down to the correct height.

Bob-the-Kuhn, 2017-03-23:
Mine is about 10mm in the air after G28 because I home in the +Z direction.  

I'm running dual Z steppers and dual Z endstops.

I'll report back after I've had some time to download today's RCBugFix & play with it.

Bob-the-Kuhn, 2017-03-23:
Looks like there's a bug in the UBL code.

I've downloaded RCBugFix & the problem remains.

When M501 reports a M851 offset of zero, a G0 Z0 command leaves the nozzle in the air by 7.8mm after BOTH G28 and G29 P1 commands.

When M501 reports a M851 offset of -7.8 I get exactly the same results.

 If I change my Z_MAX_POS from 450 to 435 the Z0 point is 22.8mm in the air no matter what.

If I switch to bilinear, with M501 reporting a M851 offset of zero, after G28 the Z0 point is 22.8mm in the air and after G29 Z0 is perfect.

----

As long as I'm bothering you ...  How about an option to do the probing in a grid starting at 0,0 rather than a circle starting nearest the current nozzle location?  

By accident I ended up doing a 9x9 bilinear.  It only took about 2x a UBL 3x3.  Lots of extra time & movement for no obvious benefits.

viper93458, 2017-04-03:
Seems that completely redoing my mesh has resolved the nozzle crash. It appears to have been from invalid data in the mesh outside the probe area. I am confused about 2 remaining things though.

1. The nozzle seems to go up and down by amounts larger than the mesh specifies as it travels between parts of my print. I have Z_CLEARANCE_DEPLOY_PROBE set to 0 and Z_CLEARANCE_BETWEEN_PROBES set to 0. Neither of these should impact during a print job though as I am moving up in layers... I don't know why the nozzle is moving up during travels. Thoughts?

2. I can print a mesh and it looks acceptable using G26. I have to have my Z Offset set to +.4. However when I go to print with my Simplify 3D software, I have to set a Z offset of -1.65! I have no idea why this is happening. In my firmware my Z offset is set to 0. I have the inductive probe a bit low which causes my nozzle to be a little high which is why I have to use a z offset of +4 when doing G26. I expected to set a Z offset of +4 in my S3D software too but this resulted in my nozzle be WAY too high, hence I figured out that -1.65 works. That doesn't compute for me. Any ideas?

-William

closed 2017-10-31 by thinkyhead