MarlinFirmware/Marlin #6096: Auto Bed Leveling Probe Offset

VinceHD, 2017-03-22:
Hallo

I have some trouble setting up the ABL for my Printer.
I calculate the Z-Probe_Offset as follow ...

**Run G28
Run G92 Z10
Decrease Z ( 0.1 Steps ) till the Nozle is 0.1 mm above the Bed.
Calculated Z_Probe_Offset = 1.1 mm**

So i edited the firmware .....

**z_probe_offset_from extruder -1.1**

When i try to start a print the nozzle is 0.4 mm above the Bed , much too high. It should be 0.1 mm according to the offset. Here is the Login of the G28/G29

`17:51:46.343 : Bed Level Correction Matrix:
17:51:46.359 : +1.000000 +0.000000 +0.000125
17:51:46.359 : +0.000000 +1.000000 -0.000531
17:51:46.359 : -0.000125 +0.000531 +1.000000
17:51:46.359 : X:155.00 Y:59.99 Z:11.09 E:-24.74 Count A:34400 B:15200 Z:8880
17:51:46.375 : X:155.00 Y:59.99 Z:11.09 E:0.00 Count A:34400 B:15200 Z:8898
17:51:46.390 : X:155.00 Y:59.99 Z:11.09 E:0.00 Count A:34400 B:15200 Z:8898
17:51:46.390 : Drucke Layer 1 von 199
17:51:46.390 : X:155.00 Y:59.99 Z:0.40 E:0.00 Count A:34400 B:15200 Z:8894
`

I not understand why the software raised the nozzle to 0.4 mm !

Here is the full G29 with DebugMode On

`18:06:44.578 : *** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
18:06:44.578 : > zprobe_zoffset = -1.10
18:06:44.578 : > home_offset[Z] = 0.00
18:06:44.578 : current_position=(100.00, 100.00, 1.10) :
18:06:44.578 : <<< set_axis_is_at_home(Z)
18:06:44.593 : current_position=(100.00, 100.00, 1.10) : sync_plan_position
18:06:44.593 : current_position=(100.00, 100.00, 1.10) : > AFTER set_axis_is_at_home
18:06:44.609 : current_position=(100.00, 100.00, 1.10) : set_probe_deployed
18:06:44.609 : deploy: 0
18:06:44.609 : do_probe_raise(10.00)
18:06:44.609 : >>> do_blocking_move_to(100.00, 100.00, 11.10)
18:06:44.984 : echo:busy: processing
18:06:46.984 : echo:busy: processing
18:06:47.156 : <<< do_blocking_move_to
18:06:47.156 : >>> do_blocking_move_to(100.00, 100.00, 11.10)
18:06:47.156 : <<< do_blocking_move_to
18:06:47.156 : <<< homeaxis(Z)
18:06:47.156 : <<< Z_SAFE_HOMING
18:06:47.171 : current_position=(100.00, 100.00, 11.10) : > (home_all_axis || homeZ) > final
18:06:47.171 : current_position=(100.00, 100.00, 11.10) : sync_plan_position
18:06:47.171 : current_position=(100.00, 100.00, 11.10) : clean_up_after_endstop_or_probe_move
18:06:47.187 : X:100.00 Y:100.00 Z:11.10 E:0.00 Count A:32000 B:0 Z:8880
18:06:47.187 : <<< gcode_G28
18:07:07.125 : current_position=(100.00, 100.00, 11.10) : >>> gcode_G29
18:07:07.125 : Machine Type: Core
18:07:07.125 : Probe: FIX_MOUNTED_PROBE
18:07:07.140 : Probe Offset X:0 Y:0 Z:-1.10 (Aligned With & Below Nozzle)
18:07:07.140 : Auto Bed Leveling: 3POINT
18:07:07.140 : current_position=(100.00, 100.00, 11.10) : sync_plan_position
18:07:07.156 : current_position=(100.00, 100.00, 11.10) : setup_for_endstop_or_probe_move
18:07:07.156 : current_position=(100.00, 100.00, 11.10) : set_probe_deployed
18:07:07.156 : deploy: 1
18:07:07.156 : do_probe_raise(10.00)
18:07:07.156 : >>> do_blocking_move_to(100.00, 100.00, 11.10)
18:07:07.156 : <<< do_blocking_move_to
18:07:07.171 : >>> do_blocking_move_to(100.00, 100.00, 11.10)
18:07:07.171 : <<< do_blocking_move_to
18:07:07.171 : > 3-point Leveling
18:07:07.171 : >>> probe_pt(15.00, 180.00, no stow)
18:07:07.171 : current_position=(100.00, 100.00, 11.10) :
18:07:07.187 : do_probe_raise(5.00)
18:07:07.187 : >>> do_blocking_move_to(15.00, 180.00, 11.10)
18:07:09.125 : echo:busy: processing
18:07:11.125 : echo:busy: processing
18:07:11.875 : <<< do_blocking_move_to
18:07:11.890 : current_position=(15.00, 180.00, 11.10) : set_probe_deployed
18:07:11.890 : deploy: 1
18:07:11.890 : current_position=(15.00, 180.00, 11.10) : >>> run_z_probe
18:07:11.890 : >>> do_blocking_move_to(15.00, 180.00, 6.10)
18:07:13.125 : echo:busy: processing
18:07:13.171 : <<< do_blocking_move_to
18:07:13.187 : current_position=(15.00, 180.00, 6.10) : >>> do_probe_move
18:07:13.187 : >>> do_blocking_move_to(15.00, 180.00, -210.00)
18:07:15.125 : echo:busy: processing
18:07:15.750 : <<< do_blocking_move_to
18:07:15.750 : current_position=(15.00, 180.00, 1.03) : sync_plan_position
18:07:15.750 : current_position=(15.00, 180.00, 1.03) : <<< do_probe_move
18:07:15.750 : current_position=(15.00, 180.00, 1.03) : <<< run_z_probe
18:07:15.750 : do_probe_raise(5.00)
18:07:15.750 : >>> do_blocking_move_to(15.00, 180.00, 6.10)
18:07:17.046 : <<< do_blocking_move_to
18:07:17.062 : <<< probe_pt
18:07:17.062 : >>> probe_pt(15.00, 60.00, no stow)
18:07:17.062 : current_position=(15.00, 180.00, 6.10) :
18:07:17.062 : do_probe_raise(5.00)
18:07:17.062 : >>> do_blocking_move_to(15.00, 60.00, 6.10)
18:07:17.125 : echo:busy: processing
18:07:19.140 : echo:busy: processing
18:07:21.140 : echo:busy: processing
18:07:21.890 : <<< do_blocking_move_to
18:07:21.906 : current_position=(15.00, 60.00, 6.10) : set_probe_deployed
18:07:21.906 : deploy: 1
18:07:21.906 : current_position=(15.00, 60.00, 6.10) : >>> run_z_probe
18:07:21.906 : current_position=(15.00, 60.00, 6.10) : >>> do_probe_move
18:07:21.906 : >>> do_blocking_move_to(15.00, 60.00, -210.00)
18:07:23.140 : echo:busy: processing
18:07:24.437 : <<< do_blocking_move_to
18:07:24.437 : current_position=(15.00, 60.00, 1.08) : sync_plan_position
18:07:24.453 : current_position=(15.00, 60.00, 1.08) : <<< do_probe_move
18:07:24.453 : current_position=(15.00, 60.00, 1.08) : <<< run_z_probe
18:07:24.453 : do_probe_raise(5.00)
18:07:24.453 : >>> do_blocking_move_to(15.00, 60.00, 6.10)
18:07:25.140 : echo:busy: processing
18:07:25.750 : <<< do_blocking_move_to
18:07:25.750 : <<< probe_pt
18:07:25.750 : >>> probe_pt(155.00, 60.00, no stow)
18:07:25.750 : current_position=(15.00, 60.00, 6.10) :
18:07:25.750 : do_probe_raise(5.00)
18:07:25.765 : >>> do_blocking_move_to(155.00, 60.00, 6.10)
18:07:27.140 : echo:busy: processing
18:07:29.140 : echo:busy: processing
18:07:31.140 : echo:busy: processing
18:07:31.390 : <<< do_blocking_move_to
18:07:31.390 : current_position=(155.00, 60.00, 6.10) : set_probe_deployed
18:07:31.390 : deploy: 1
18:07:31.390 : current_position=(155.00, 60.00, 6.10) : >>> run_z_probe
18:07:31.406 : current_position=(155.00, 60.00, 6.10) : >>> do_probe_move
18:07:31.406 : >>> do_blocking_move_to(155.00, 60.00, -210.00)
18:07:33.140 : echo:busy: processing
18:07:33.906 : <<< do_blocking_move_to
18:07:33.906 : current_position=(155.00, 60.00, 1.13) : sync_plan_position
18:07:33.906 : current_position=(155.00, 60.00, 1.13) : <<< do_probe_move
18:07:33.921 : current_position=(155.00, 60.00, 1.13) : <<< run_z_probe
18:07:33.921 : do_probe_raise(5.00)
18:07:33.921 : >>> do_blocking_move_to(155.00, 60.00, 6.10)
18:07:35.140 : echo:busy: processing
18:07:35.203 : <<< do_blocking_move_to
18:07:35.203 : <<< probe_pt
18:07:35.203 : current_position=(155.00, 60.00, 6.10) : set_probe_deployed
18:07:35.203 : deploy: 0
18:07:35.203 : do_probe_raise(10.00)
18:07:35.218 : >>> do_blocking_move_to(155.00, 60.00, 11.10)
18:07:36.500 : <<< do_blocking_move_to
18:07:36.500 : >>> do_blocking_move_to(155.00, 60.00, 11.10)
18:07:36.500 : <<< do_blocking_move_to
18:07:36.515 : current_position=(155.00, 60.00, 11.10) : clean_up_after_endstop_or_probe_move
18:07:36.515 : current_position=(155.00, 60.00, 11.10) : > probing complete
18:07:36.515 : Bed Level Correction Matrix:
18:07:36.515 : +1.000000 +0.000000 +0.000366
18:07:36.531 : +0.000000 +1.000000 -0.000406
18:07:36.531 : -0.000366 +0.000406 +1.000000
18:07:36.531 : current_position=(155.00, 60.00, 11.10) : G29 uncorrected XYZ
18:07:36.531 : Z from Probe:11.07  Matrix:11.06  Discrepancy:0.01
18:07:36.546 : current_position=(155.00, 60.00, 11.07) : G29 corrected XYZ
18:07:36.546 : <<< gcode_G29
18:07:36.546 : X:155.00 Y:60.00 Z:11.07 E:0.00 Count A:34400 B:15200 Z:8880
18:07:36.546 : current_position=(155.00, 60.00, 11.07) : sync_plan_position`

Bob-the-Kuhn, 2017-03-23:
I don't understand the role of G92 Z10 in your procedure.  If you'd said G0 Z10 then I'd know exactly what you were doing.

With that big of a ridge in your bed, the location of your test for Z_Probe_Offset is critical.

I suggest the following:
Use a 9x9 grid
After G29 adjust Z_Probe_Offset based on how far the nozzle is above the board when M114 says Z0.  

Unfortunately you'll need to do another G29 before the new Z_Probe_Offset takes effect.

Also, bilinear leveling isn't perfect.  Here's a write up on bilinear.
[Bilinear Auto Bed Leveling Details 0.4.zip](https://github.com/MarlinFirmware/Marlin/files/863764/Bilinear.Auto.Bed.Leveling.Details.0.4.zip)

lrpirlet, 2017-03-23:
@VinceHD

Le 22/03/17 à 20:24, VinceHD a écrit :
> When i try to start a print the nozzle is 0.4 mm above the Bed , much 
> too high. It should be 0.1 mm according to the offset

If your generated Gcode sets the first layer at .3 mm, then the math 
works .4 = .3 + .1...

I would strongly suggest to verify what the gcode tells the printer to 
do...
I would strongly suggest to try and reduce that .1 mm to ZERO when both 
the nozzle and the bed are at print temperature... I personally use 
thermic print paper (thickness = .005, I always get my receipt from the 
petrol station :-) )... I measure with it at ambient temperature to get 
the nozzle-bed distance to about ZERO at print temperature. Note, this 
works for me, your printer may react differently from mine...

VinceHD, 2017-03-23:
Hallo

Thank you for the fast reply !

I use the G92 Z10 Command only for calculating the Z-Offset.
After the G28 i tell the Software the Z is at 10.0 ( G92 Z10 ) then i decrease the hight until the Nozzle is 0.1 mm above the bed. With M114 i get the current position ! Now i only need to do the math ( Current Z ) - 10 = Offset of the Probe.

The G-Code is an interesting Point !
I use a Layer-Hight of 0.2 mm and im printing the first Layer with 200 % = 0.4 mm ?
I use Slicer to generate the G-Code. But it would make no sense if Slicer would put the first Layer on 0.4 mm Z-Height because you editing the first Layer with 200 % becasuse you want the Layer stick better to the bed. In other word more PLA on the same Layer-Hight ( 0.2 mm )

dvseliteus, 2017-03-24:
Not sure if this helps you but I found this way very easy and fast to dial in my Z-Offset quickly through the G-Code Terminal and save to the EEPROM.
M851 Z0 ; set the offset to zero
M851 Z-X.X ; to set the offset (X=offset number)
M500 ; Stores the values in EEPROM
Adjust as needed
I use this method so I don't need to add it to the start scripts or re-flash.

pflannery, 2017-03-27:
@Bob-the-Kuhn 
> I suggest the following:
Use a 9x9 grid
After G29 adjust Z_Probe_Offset based on how far the nozzle is above the board when M114 says Z0.

- For me G29 (5x5 BILINEAR) always finishes in the far corner of my bed. Is this where I should test and adjust the Z_Probe_Offset or should I home first with g28? or does it even matter?
- M114 gives me two Z outputs. I guess it's the first Z result here -> `17:59:19.711 : X:171.00 Y:192.00 Z:-0.00 E:0.00 Count A:29040 B:-1680 Z:140 . 
- There seems to be a bug when I comment out `// #define MIN_SOFTWARE_ENDSTOPS` as I can't go below 0 nor will M211 s0. So I'm forced to use g92 to allow me to go under 0.

Bob-the-Kuhn, 2017-03-27:
G29 bilinear always finishes in the far corner by design.

I just noticed a change in the software endstops.  1-3 weeks ago it changed from **min_software_endstops** to **MIN_SOFTWARE_ENDSTOPS**.  

**min_software_endstops** should be either true or false.  You want false.  

**MIN_SOFTWARE_ENDSTOPS** is either commented out or enabled.  You want commented out.

The system will warn you if you use the lower case version when it's expecting upper case.

There's no need to run G28 after G29.  G28 turns off bilinear bed leveling.  G28 doesn't modify the grid/table so a M420 S1 will turn it back on if you run G28 after G29.

They've also recently changed the M851 command.  Now when you enter a new value for z offset the bilinear grid/table is automatically updated.  No more re-running G29 to see how the new z offset affects things.

Where to do your paper test?  Ideally at one of the probed points but that's a pain to determine precisely.  Just get close and do the paper test. That should get you really close to determining the final z offset.

----

Eventually I expect that you'll want to switch to the UBL bed leveling system.  It's basically a superset of the bilinear system.  It allows you to manually enter the offsets/compensation for the areas that the probe can't get to.  The other improvement is on long extrudes.  Bilinear just does one long extrude with the head moving in a straight line from the old to the new position. If the bed has waves in it in between then they'll show up in the first layer thickness.  UBL breaks up the move at mesh boundaries and provides a new compensation at each boundary.

Right now there's still changes being done and we don't have a good write up for beginners (me included).  I expect it'll be a week or so before the first draft of the write up is available.

Roxy-3D, 2017-03-27:
@VinceHD   When you change your Configuration.h file's #define Z_PROBE_OFFSET_FROM_EXTRUDER,  you flash the new image into your controller board.  Are you doing a M502 followed by a M500?   

If not...   That could explain all the difficulty you are having.

closed 2018-02-06 by thinkyhead