MarlinFirmware/Marlin #6098: [Bug?] CCoreXY printer move axis after G28 weird behavior

soimy, 2017-03-23:
I'm running a CoreXY printer with Marlin-RCBugfix 2017.3.20 version.
My setup on homing is X-min, Y-min, Z-max, and my configuration is as followed. 
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/863536/Configuration.txt)

**Approach to represent the problem:**

- Enable CoreXY in configuration.h
- Home all axis in menu or G28
- Then move x/y axis in menu->Move Axis.

Now printer will first move Z platform very quickly upward to Zero ( Almost crush into the hotend ... ), after that, the printer will then move the X/Y axis as demanded.
I can hardly understand the corexy mechanism in code, pls help to confirm whether it's a setting problem or potential bug?

Bob-the-Kuhn, 2017-03-23:
I don't use an LCD so I may be off in left field on this.

You're homing in the Z+ direction with Z_MAX_POS 225, X_MIN_POS 0, Y_MIN_POS 0.  That means after G28 M114 will report X0, Y0, Z225.

If you then do a G0 X10 Y10 Z10, the Z axis will move much faster than X & Y because the planner sets the move up such that all axis arrive at the destinations at the same time.

Maybe I'm reading your post wrong.  Are you saying that immediately after G28 the very first move you command is something like G0 X20 but the Z platform moves even though there was no command to move it?

soimy, 2017-03-23:
> Maybe I'm reading your post wrong. Are you saying that immediately after G28 the very first move you command is something like G0 X20 but the Z platform moves even though there was no command to move it?

Yes Bob, even when I move a single axis, the Z move first, from Z-max position all the way up to Z0.

Bob-the-Kuhn, 2017-03-23:
I expect that from G28. It'll move to the Z max first and then do the X & Y.

I don't expect that from G0 X1.

If you do G28, then M114, then G0 X1 and then M114 - what do the M114 commands report?

Dreamloom, 2017-03-23:
In the config file, there is no probe type selected.

mperdue, 2017-03-23:
I had this same problem. After homing the first move would drop the Z axis to near zero then back to whatever the move. I reloaded the firmware then did an M502 followed by M500. That resolved the problem so I'm assuming something changed in the EEPROM.

soimy, 2017-03-24:
> If you do G28, then M114, then G0 X1 and then M114 - what do the M114 commands report?

Here is the log of above G-code command:

```
SENT: G28
READ: X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:156800
READ: ok
SENT: M114
READ: X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:156800
READ: ok
SENT: G0 X1
READ: ok
SENT: M114
READ: X:1.00 Y:0.00 Z:0.00 E:0.00 Count A:161 B:161 Z:9942
READ: ok
```

soimy, 2017-03-24:
@mperdue Thanks, Restore failsafe then Store memory work!

closed 2017-03-24 by soimy