MarlinFirmware/Marlin #6102: Bltouch clone and endstop inverting problem.

generaldawid, 2017-03-23:
HI, im using Bltouch clone, MKS gen v1.4 board and Marlin RC8 version.

My problem is  when BLtouch pin is in up position M119 show Open and whem pin is down is trigger.

// coarse Endstop Settings
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors. Firmware no responding for any changes like  
 //#define ENDSTOPPULLUP_ZMIN,//#define ENDSTOPPULLUP_ZMIN_PROBE enable or disable this section.

#define Z_MIN_ENDSTOP_INVERTING  true or false on this and #defineZ_MIN_PROBE_ENDSTOP_INVERTING true or false on this
Change pin to any other didnt give me any change.

Every time my BLtouch show trigger  in down position

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true// set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true// set to true to invert the logic of the endstop.

Bob-the-Kuhn, 2017-03-23:
To run a clone you need the following settings:
```
//#define BLTOUCH 
#define Z_ENDSTOP_SERVO_NR 0
#define Z_SERVO_ANGLES {10,90} // Z Servo Deploy and Stow angles
#define NUM_SERVOS 1
#define SERVO_DELAY 750  // milliseconds 500 is marginal
```

You also to set **#define Z_MIN_ENDSTOP_INVERTING true** or **Z_MIN_PROBE_ENDSTOP_INVERTING true**, depending what pin the probe is tied to.

When all is configured properly then M119 will report OPEN when the clone is deployed and TRIGGERED when it is stowed.

generaldawid, 2017-03-23:
Bob-the-Kuhn what do you mean "depending what pin the probe is tied to"

generaldawid, 2017-03-23:
My Bltouch is set to z_min pin, but in firmware i have this 
// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 do i neet to // this #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN then set true on  Z_MIN_PROBE_ENDSTOP_INVERTING, and what about #define Z_MIN_ENDSTOP_INVERTING

Bob-the-Kuhn, 2017-03-23:
OK - you're using the Z_MIN_PIN.

In that case you need 
```
//#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN 
#define Z_MIN_ENDSTOP_INVERTING true
```

Marlin will ignore the setting for Z_MIN_PROBE_ENDSTOP_INVERTING

generaldawid, 2017-03-23:
Ok i can now comfirm bltouch is working.
Thanks for your help.

closed 2018-02-06 by thinkyhead

Nitrosito, 2019-01-31:
thank you both, this was useful for me :+1:

darap89, 2019-11-16:
<3