MarlinFirmware/Marlin #6103: Setting Z offset via LCD.

KevMags, 2017-03-23:
I am using RCBugFix downloaded today.  I don't want bed leveling but I would like to set just the Z offset via the LCD.  I have always adjusted my z endstop to get the proper clearance but with my new printer it uses nozzle contact to trigger the z endstop therefore I need a z offset.  Thanks in advance.

Tannoo, 2017-03-23:
Control - Motion - Z Offset

KevMags, 2017-03-23:
I just flashed a fresh copy and only changing LCD type (ReRap Full Graphic) and encoder direction and it's not there.  Do I need to enable something?

Tannoo, 2017-03-23:
What version did you flash?

Tannoo, 2017-03-23:
I have the same display and it's there for me. I'm using RCBugFix.

Tannoo, 2017-03-23:
I know it was there back in RC7 also.

Tannoo, 2017-03-23:
Maybe you have to have a probe inabled?

KevMags, 2017-03-23:
I flashed RCBugFix that I downloaded 2 hours ago.  I have no probe.  My Z endstop is triggered when my nozzle touches the build plate (Voron).  I enabled both Mesh bed leveling and manual bed leveling and it is in the menu now.  I have no need for bed leveling but I thought that you could set a z offset from the LCD without enabling bed leveling.  I can just use it this way and comment out the menu entry.  Thanks.

unixunion, 2017-07-25:
Hi, I have noticed that setting the Z offset is no longer available on my MKS. I'm running 1.1.4, and my probe is on Z- endstop. Is there some other way to enable the z-offset on the LCD?

KevMags, 2017-07-25:
I find it in Control>Motion>Z offset.  

Note: 
I have used 1.1.2 through 1.1.4Bugfix.  As stated in my previous post, enabling mesh and manual bed leveling is not necessary since I don't use it.

unixunion, 2017-07-25:
I don't have  Control>Motion>Z offset. Any idea what could be disabling it?

closed 2018-02-09 by thinkyhead