MarlinFirmware/Marlin #6105: UBL Boundries question

Tannoo, 2017-03-23:
With `AUTO_BED_LEVELING_UBL` defined,
these are the probe boundries:
```
  #define UBL_PROBE_PT_1_X 39       // These set the probe locations for when UBL does a 3-Point leveling
  #define UBL_PROBE_PT_1_Y 180      // of the mesh.
  #define UBL_PROBE_PT_2_X 39
  #define UBL_PROBE_PT_2_Y 20
  #define UBL_PROBE_PT_3_X 180
  #define UBL_PROBE_PT_3_Y 20
```
And not these?
```
  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 34
  #define RIGHT_PROBE_BED_POSITION 185
  #define FRONT_PROBE_BED_POSITION 37
  #define BACK_PROBE_BED_POSITION 180
```

Tannoo, 2017-03-23:
And wouldn't these be the one in the same?
```
  // 3 arbitrary points to probe.
  // A simple cross-product is used to estimate the plane of the bed.
  #define ABL_PROBE_PT_1_X 34
  #define ABL_PROBE_PT_1_Y 37
  #define ABL_PROBE_PT_2_X 185
  #define ABL_PROBE_PT_2_Y 37
  #define ABL_PROBE_PT_3_X 110
  #define ABL_PROBE_PT_3_Y 180
```

Tannoo, 2017-03-23:
Also, not sure how to get UBL to do a 3-point probing.

Blue-Marlin, 2017-03-23:
There is now way UBL could work with only 3 points.

Tannoo, 2017-03-23:
```
Marlin_main.cpp:5171: error: 'set_bed_leveling_enabled' was not declared in this scope

       set_bed_leveling_enabled(false);
```
UBL not working for me.
I tried this:
```
#if ENABLED(AUTO_BED_LEVELING_UBL)
  #include "UBL.h"
  extern void set_bed_leveling_enabled(bool enable=true);
#endif
```
and I now get this:
```
In function `gcode_M48()':
sketch/Marlin_main.cpp:5171: undefined reference to `set_bed_leveling_enabled(bool)'
sketch/Marlin_main.cpp:5324: undefined reference to `set_bed_leveling_enabled(bool)'
```
ABL compiles and works fine.

Tannoo, 2017-03-23:
Well, I found that it had to do with the `Z_MIN_PROBE_REPEATABILITY_TEST` enabled. I disabled it and now it compiles.

Tannoo, 2017-03-23:
`   *   T     3-Point    Perform a 3 Point Bed Leveling on the current Mesh`

Tannoo, 2017-03-23:
G29 is not doing anything.
G29 P1 is not doing anything.
G29 P2 is not doing anything.
G29 G3 P1 is not doing anything.

M111 S1, 2, 3, 4, 5, 6, 7 are of no help

I have already M502'ed and M500'ed. Still nothing.
What am I missing?

Tannoo, 2017-03-23:
I have only `AUTO_BED_LEVELING_UBL` enabled.
These are not an issue?
```
  #define PLANNER_LEVELING      ((HAS_ABL && DISABLED(AUTO_BED_LEVELING_UBL)) || ENABLED(MESH_BED_LEVELING))
```
```
      #if PLANNER_LEVELING
        case 29: // G29 Detailed Z probe, probes the bed at 3 or more points,
                 // or provides access to the UBL System if enabled.
          gcode_G29();
          break;
```

How do I get UBL to work?

Tannoo, 2017-03-23:
G29 (anything) only gives me a response of `ok`. As if everything I ask of it is being ignored.

Roxy-3D, 2017-03-23:
> There is no way UBL could work with only 3 points.
> You could try 4 - but then its worse than linear levelling (borders).
> Also neither bilinear- nor mesh- levelling can work with only 3 points.

Au-contraire!  Give it a try and see.    The high resolution mesh is still there.   But based on a 3-Point leveling (just as the planar leveling works) the mesh is tilted based on the cross product of the vectors.

A place where this would be useful is if you have a perfectly tuned piece of glass, but you some times get a spec of plastic that keeps the glass from laying perfectly flat on the metal frame.  Well...   You just 'level' the mesh.

Tannoo, 2017-03-23:
I got UBL to work by:
```
      #if PLANNER_LEVELING || HAS_ABL
        case 29: // G29 Detailed Z probe, probes the bed at 3 or more points,
                 // or provides access to the UBL System if enabled.
          gcode_G29();
          break;
      #endif // PLANNER_LEVELING
```

Roxy-3D, 2017-03-23:
> I got UBL to work by:

@thinkyhead 
Yeah...  There are problems in M48 and G28 in today's code.   I'm fixing them right now.  Give me an hour or two.

Tannoo, 2017-03-23:
It really should be re-written here somehow. I'm not sure what was intended.
```
  /**
   * Set ABL options based on the specific type of leveling
   */
  #define ABL_PLANAR (ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_3POINT))
  #define ABL_GRID   (ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR))
  #define HAS_ABL    (ABL_PLANAR || ABL_GRID || ENABLED(AUTO_BED_LEVELING_UBL))

  #define PLANNER_LEVELING      ((HAS_ABL && DISABLED(AUTO_BED_LEVELING_UBL)) || ENABLED(MESH_BED_LEVELING))
  #define HAS_PROBING_PROCEDURE (HAS_ABL || ENABLED(Z_MIN_PROBE_REPEATABILITY_TEST))

  #if HAS_PROBING_PROCEDURE
    #define PROBE_BED_WIDTH abs(RIGHT_PROBE_BED_POSITION - (LEFT_PROBE_BED_POSITION))
    #define PROBE_BED_HEIGHT abs(BACK_PROBE_BED_POSITION - (FRONT_PROBE_BED_POSITION))
  #endif
```

Roxy-3D, 2017-03-23:
> It really should be re-written here somehow. I'm not sure what was intended.
Thinky made a bunch of changes to get the UBL code into compliance with the Marlin coding standards...  And a couple of things have to be tweaked...   I'm working on it....

Tannoo, 2017-03-23:
I got my first UBL mesh though:
[image]

Roxy-3D, 2017-03-23:
@Tannoo    I think I have the 'corrections' in place.   It would probably be beneficial for you to pull it down and lay your configuration files on top of it.   (Seriously!!!)

How are you generating that topographical map?   I have a similar phenomena with the edges dropping dramatically.   But I also have peaks and valleys in the glass in the center area of the glass.  It would be interesting to see just how bad my glass is.

Tannoo, 2017-03-23:
>How are you generating that topographical map? 

Get mesh data by `G29 O`.
```
       .                 .           -0.18151    -0.16146    -0.18359    -0.16953    -0.18073    -0.19349    -0.28307    -0.76094   
 
       .                 .           -0.00990     0.03568     0.07083     0.06536     0.05443     0.02604    -0.10078    -0.74870   
 
       .                 .            0.02500     0.10104     0.15339     0.19453     0.16094     0.10182    -0.02812    -0.67344   
 
       .                 .            0.06953     0.15938     0.22734     0.26563     0.24505     0.17995     0.02526    -0.59479   
 
       .                 .            0.06120     0.18906     0.28255     0.31042     0.29870     0.23516     0.09661    -0.49714   
 
       .                 .            0.06380     0.20469     0.27839     0.32578     0.32031     0.27083     0.11458    -0.38958   
 
       .                 .            0.04974     0.20573     0.26771     0.30964     0.32422     0.32135     0.18099    -0.32187   
 
       .                 .            0.04662     0.19505     0.26510     0.30885     0.33099     0.34714     0.24844    -0.16771   
 ```
Copy that info.

I then paste into Notepad++ and clean it up to this:
```
-0.18151    -0.16146    -0.18359    -0.16953    -0.18073    -0.19349    -0.28307    -0.76094
-0.00990     0.03568     0.07083     0.06536     0.05443     0.02604    -0.10078    -0.74870
0.02500     0.10104     0.15339     0.19453     0.16094     0.10182    -0.02812    -0.67344
0.06953     0.15938     0.22734     0.26563     0.24505     0.17995     0.02526    -0.59479
0.06120     0.18906     0.28255     0.31042     0.29870     0.23516     0.09661    -0.49714
0.06380     0.20469     0.27839     0.32578     0.32031     0.27083     0.11458    -0.38958
0.04974     0.20573     0.26771     0.30964     0.32422     0.32135     0.18099    -0.32187
0.04662     0.19505     0.26510     0.30885     0.33099     0.34714     0.24844    -0.16771
```
The save it as a .CSV file and import it into Excel.
Then select the data cells in the worksheet and press F11.
Then change the chart type to `Surface`.

Tannoo, 2017-03-23:
Can you make the `G29 O` output cleaner? Then it would just be a copy and paste into excel.

Roxy-3D, 2017-03-23:
The problem with G29 O is both PronterFace and Repetier host use a proportional font.  But to your point...   Would a G29 S -1 give you a cleaner format?    I'm not sure if you will have to do this to try it, but I just changed my local copy to be:
```cpp
    //
    // Store a Mesh in the EEPROM
    //

    if (code_seen('S')) {     // Store (or Save) Current Mesh Data
      storage_slot = code_has_value() ? code_value_int() : ubl.state.eeprom_storage_slot;

      if (storage_slot == -1) {                     // Special case, we are going to 'Export' the mesh to the
        SERIAL_ECHOLNPGM("G29 I 999");              // host in a form it can be reconstructed on a different machine
        for (uint8_t x = 0; x < UBL_MESH_NUM_X_POINTS; x++)
          for (uint8_t y = 0;  y < UBL_MESH_NUM_Y_POINTS; y++)
            if (!isnan(z_values[x][y])) {
              SERIAL_ECHOPAIR("M421 I ", x);
              SERIAL_ECHOPAIR(" J ", y);
              SERIAL_ECHOPGM(" Z ");
              SERIAL_ECHO_F(z_values[x][y], 6);
	      idle();
	      delay(50);
            }
        return;
      }
```
I'll be pushing that change soon...  But that might just give you a 'cleaner' version of the data that you can use.

Tannoo, 2017-03-23:
Well, `G29 O` is easier as it is.

`G29 S -1` gave me this:
```
M421 I 2 J 2 Z 0.027865
M421 I 2 J 3 Z 0.024219
M421 I 2 J 4 Z 0.038542
M421 I 2 J 5 Z 0.061979
M421 I 2 J 6 Z 0.060156
M421 I 2 J 7 Z 0.051302
M421 I 2 J 8 Z 0.015625
M421 I 2 J 9 Z -0.122656
M421 I 3 J 2 Z 0.141927
M421 I 3 J 3 Z 0.163542
M421 I 3 J 4 Z 0.172396
M421 I 3 J 5 Z 0.194531
M421 I 3 J 6 Z 0.146094
M421 I 3 J 7 Z 0.129948
M421 I 3 J 8 Z 0.053646
M421 I 3 J 9 Z -0.131771
M421 I 4 J 2 Z 0.191667
M421 I 4 J 3 Z 0.210938
M421 I 4 J 4 Z 0.247656
M421 I 4 J 5 Z 0.251042
M421 I 4 J 6 Z 0.213021
M421 I 4 J 7 Z 0.169792
M421 I 4 J 8 Z 0.077865
M421 I 4 J 9 Z -0.144791
M421 I 5 J 2 Z 0.222396
M421 I 5 J 3 Z 0.270313
M421 I 5 J 4 Z 0.272917
M421 I 5 J 5 Z 0.286198
M421 I 5 J 6 Z 0.242709
M421 I 5 J 7 Z 0.189584
M421 I 5 J 8 Z 0.091928
M421 I 5 J 9 Z -0.138802
M421 I 6 J 2 Z 0.235417
M421 I 6 J 3 Z 0.243490
M421 I 6 J 4 Z 0.257813
M421 I 6 J 5 Z 0.255469
M421 I 6 J 6 Z 0.217709
M421 I 6 J 7 Z 0.154428
M421 I 6 J 8 Z 0.071875
M421 I 6 J 9 Z -0.156770
M421 I 7 J 2 Z 0.229167
M421 I 7 J 3 Z 0.223438
M421 I 7 J 4 Z 0.201563
M421 I 7 J 5 Z 0.188802
M421 I 7 J 6 Z 0.156771
M421 I 7 J 7 Z 0.108854
M421 I 7 J 8 Z 0.037240
M421 I 7 J 9 Z -0.179427
M421 I 8 J 2 Z 0.149740
M421 I 8 J 3 Z 0.095834
M421 I 8 J 4 Z 0.062500
M421 I 8 J 5 Z 0.055209
M421 I 8 J 6 Z 0.016667
M421 I 8 J 7 Z -0.041927
M421 I 8 J 8 Z -0.088802
M421 I 8 J 9 Z -0.255208
M421 I 9 J 2 Z -0.314323
M421 I 9 J 3 Z -0.389323
M421 I 9 J 4 Z -0.531770
M421 I 9 J 5 Z -0.610416
M421 I 9 J 6 Z -0.680989
M421 I 9 J 7 Z -0.733333
M421 I 9 J 8 Z -0.734635
M421 I 9 J 9 Z -0.756771
```

Tannoo, 2017-03-23:
>I could add an option to the Map to make it suppress the variable spaces it prints in an attempt to line up the columns.

That would be nice. Have it output something like the ABL mesh does. Tab or comma separation is best.

Roxy-3D, 2017-03-23:
> Well, G29 O is easier as it is.

OK...   Let me take a look at how to do this cleanly with the minimum of extra code....    If it is comma separated, you don't care if the columns line up, right?

Tannoo, 2017-03-23:
No, that doesn't matter. Just separation by a tab or comma.

Blue-Marlin, 2017-03-23:
> The problem with G29 O is both PronterFace and Repetier host use a proportional font.

Would be nice i you could distinguish between a proportional font and omitting leading spaces.

Bob-the-Kuhn, 2017-03-23:
FYI on Repetier Host.  

It tries to save you from entering invalid commands.  It doesn't know about the new UBL format so it does things like truncate commands, not send them or even send "G9" instead of "G29".

I've had to go to Pronterface to get around this.

Tannoo, 2017-03-23:
I just got this from Repetier Host when I sent `G29 O` from it's Manual control:
```
17:31:14.195 : N19 G29 O*69
17:31:14.195 : ubl_eeprom_start=544
17:31:14.195 : echo:Home XYZ first
17:31:27.183 : X:0.00 Y:0.00 Z:6.10 E:0.00 Count X:0 Y:0 Z:23424
17:31:27.198 : Bed Topography Report:
17:31:27.198 : (0, 9)                                                                                                                                                             (9,9)
17:31:27.214 : (1,184)                                                                                                                                                         (199,184)
17:31:27.251 : .                 .           -0.12266    -0.13177    -0.14479    -0.13880    -0.15677    -0.17943    -0.25521    -0.75677
17:31:27.251 : 
17:31:27.251 : .                 .            0.01563     0.05365     0.07787     0.09193     0.07188     0.03724    -0.08880    -0.73464
17:31:27.251 : 
17:31:27.267 : .                 .            0.05130     0.12995     0.16979     0.18958     0.15443     0.10885    -0.04193    -0.73333
17:31:27.267 : 
17:31:27.267 : .                 .            0.06016     0.14609     0.21302     0.24271     0.21771     0.15677     0.01667    -0.68099
17:31:27.267 : 
17:31:27.283 : .                 .            0.06198     0.19453     0.25104     0.28620     0.25547     0.18880     0.05521    -0.61042
17:31:27.283 : 
17:31:27.283 : .                 .            0.03854     0.17240     0.24766     0.27292     0.25781     0.20156     0.06250    -0.53177
17:31:27.283 : 
17:31:27.298 : .                 .            0.02422     0.16354     0.21094     0.27031     0.24349     0.22344     0.09583    -0.38932
17:31:27.298 : 
17:31:27.298 : .                 .            0.02786     0.14193     0.19167     0.22240     0.23542     0.22917     0.14974    -0.31432
17:31:27.298 : 
17:31:27.314 : .                 .                 .                 .                 .                 .                 .                 .                 .                 .
17:31:27.314 : 
17:31:27.314 : [      .       ]        .                 .                 .                 .                 .                 .                 .                 .                 .
17:31:27.335 : (1,1)                                                                                                                                                             (199,1)(0,0)                                                                                                                                                                (9,0)
```

Tannoo, 2017-03-23:
That was not much different than using the serial monitor from the IDE.

Tannoo, 2017-03-23:
But, what about the boundry issue?

Roxy-3D, 2017-03-24:
> But, what about the boundary issue?

Huh?

Tannoo, 2017-03-24:
Well, the original issue is three different boundaries to set. It was as mud. I think I have waded throught that.

But new boundary issue. I'm trying to back off of the right side of the bed for probing.
I changed all three boundary settings:
`#define RIGHT_PROBE_BED_POSITION 155`  <-- from 180.
`#define ABL_PROBE_PT_2_X 155`  <-- from 180.
and
`#define UBL_PROBE_PT_2_X 155  <-- also from 180.

UBL is still probing the same spots. I will go more drastic to verify:
I am going to try:
```
  #define LEFT_PROBE_BED_POSITION 50
  #define RIGHT_PROBE_BED_POSITION 100
  #define FRONT_PROBE_BED_POSITION 50
  #define BACK_PROBE_BED_POSITION 100
```

Roxy-3D, 2017-03-24:
> UBL is still probing the same spots. I will go more drastic to verify:
> I am going to try:
```cpp
  #define LEFT_PROBE_BED_POSITION 50
  #define RIGHT_PROBE_BED_POSITION 100
  #define FRONT_PROBE_BED_POSITION 50
  #define BACK_PROBE_BED_POSITION 100
```

No!  Those are not even looked at by UBL.   At least...  Not yet...  Just define your bed the correct size and use a UBL_MESH_INSET of 1   (or 0!)

Use the whole bed.   You want the mesh to stretch across any printable area.

Tannoo, 2017-03-24:
Where do I set the size of the bed for UBL?

Tannoo, 2017-03-24:
From this?
```
// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 185
#define Z_MAX_POS 190
```

Roxy-3D, 2017-03-24:
You don't.  It already knows the size from 
```
#define MAX_X_POS
#define MIN_X_POS
#define MAX_Y_POS
#define MIN_Y_POS
```
Yes....

Tannoo, 2017-03-24:
Alright. But that "dip" in the topo, I believe is my sensor reading too close to the right edge of the bed.
So, just change the `UBL_MESH_INSET`?

Tannoo, 2017-03-24:
Is the INSET in mm? or %?

Roxy-3D, 2017-03-24:
No!   Don't worry about it!   It will be fine!   That gets addressed with the G26 P R O3.5 and the interactive mesh edit of G29 P4 R O     Trust me!   It is all going to be OK.   You will get adhesion across 100% of the bed 100% of the time.

Adhesion is in mm.

Tannoo, 2017-03-24:
Ok. I haven't edited any mesh yet. Uncharted territory for me. lol

Tannoo, 2017-03-24:
Or, Git squash commands.

Tannoo, 2017-03-24:
I have to go catch some Magikarp, I'll be back to tackle this a little later.

Roxy-3D, 2017-03-24:
Just do the commands
- G28 
- G29 P1 O
- G29 P3 R C .5
- G26 P C O3.5
- G29 P4 R O    To edit....

That sequence will get you very close to a perfect mesh.   You just repeat the last two commands and you will have everything perfect.

Bob-the-Kuhn, 2017-03-24:
@Roxy-3D - do you have anything like a beginners guide for UBL?

Also, if you have something you'd like put onto the Wiki I can start that process.

Roxy-3D, 2017-03-24:
@Bob-the-Kuhn    You have UBL limping along, right?   I was thinking of adding a P command to automatically fill in the un-probed (after a G29 P1 command) regions.   I was going to have it do a spline fit outwards and populate the unprobed areas of the mesh.   

If we did that...  The G26 would go smoother on the first iteration.   The sequence would be 
- G28
- G29 P1 O
- G29 P3 R MAGIC_TO_INVOKE_SPLINE_FIT
- G26 P C O3.5
- G29 P4 R O To edit....

What I'm hoping is I can get the process automated enough that you could write directions to use it....

Bob-the-Kuhn, 2017-03-24:
Limping is the correct word.  I still need to get my nozzle close to the bed.

I'll play with the P6 Cxx command tomorrow & see if I can get to the point where it makes sense to try G26.

Roxy-3D, 2017-03-24:
> & see if I can get to the point where it makes sense to try G26.

Pretty much...   All of the automated movement is designed to be stopped if the 'Wrong Thing' starts to happen.   On the both the G29 P1 and the G26 if you press and hold the encoder wheel...  It will stop within 1 point.   If you don't give those two commands the 'C' parameter which says try to go to the next closest point, you will only branch out so fast from 'known good' territory.   The nozzle goes back and forth a bunch without the C parameter, but you can easily catch it and stop it before the nozzle gets too close to the glass.

Tannoo, 2017-03-24:
`G26 P C O3.5` goes into a "User-controlled Prime". But, clicking the screen does nothing. It won't stop priming.
Also, had to find where to change the temp. 205C is a little warm for PLA. It was leaving Hersheys on the table. lol
I saw there were some updates, so I applied those.
Now, `G26 P C O3.5` does nothing except for respond with `ok`.

Tannoo, 2017-03-24:
`G26 P10 C O3.5 B50 H195` still does nothing.

Roxy-3D, 2017-03-24:
I'm seeing something similar...   I'm seeing the Press & Hold of the encoder wheel not abort the G26 or the G29 P1.   Something got broke in the ultralcd.cpp and the way those functions hook into it.

Tannoo, 2017-03-24:
In `G26_Mesh_Validation_Tool.cpp`, I changed `#if ENABLED(AUTO_BED_LEVELING_UBL) && ENABLED(UBL_MESH_EDIT_ENABLED)` to ``#if ENABLED(AUTO_BED_LEVELING_UBL)` and had to change:
```
  bool keep_heaters_on = false;
       //g26_debug_flag = false;
```
Also, in `marlin_main.cpp`, I changed
```
      #if ENABLED(AUTO_BED_LEVELING_UBL) && ENABLED(UBL_MESH_EDIT_ENABLED)
        case 26: // G26: Mesh Validation Pattern generation
          gcode_G26();
          break;
```
to
```
      #if ENABLED(AUTO_BED_LEVELING_UBL)
        case 26: // G26: Mesh Validation Pattern generation
          gcode_G26();
          break;
```
and G26 is back to working.

Tannoo, 2017-03-24:
Maybe `#if ENABLED(AUTO_BED_LEVELING_UBL) && ENABLED(UBL_MESH_EDIT_ENABLED)`

should be `#if ENABLED(AUTO_BED_LEVELING_UBL) || ENABLED(UBL_MESH_EDIT_ENABLED)`?

Roxy-3D, 2017-03-24:
OK...   I better start a fresh branch so I can get the PR's in place to revert that stuff.   I haven't looked at the history, but it probably happened at the same time G29_lcd_clicked() got its name changed....    

Can you abort the G29 P1 and/or the G26 by doing a Press & Hold of the encoder wheel.   If so, good!  But if not...   More needs to be understand and reverted.

Tannoo, 2017-03-24:
I just tried....no. It is constantly beeping... so, the loop for purging is not waiting for user click. It is also not purging smoothly. That section is getting repeated endlessly. Causing no time for a user to click.

Roxy-3D, 2017-03-24:
I'm sure what you posted up above to get it running again is needed.   But I think the problem involves the re-work that was done to put ubl_lcd_clicked() in place.    What has to happen is if UBL has taken control of the LCD Panel...  The normal lcd_clicked() or LCD_CLICKED needs to return 0; regardless of whether or not a click is present.   That is necessary to keep the LCD code from thinking a menu item has been selected and hopping to it.

Tannoo, 2017-03-24:
I am only getting it working. You can fix it. lol

Roxy-3D, 2017-03-24:
> I am only getting it working. You can fix it. lol

I fully understand!

Roxy-3D, 2017-03-24:
If you can back up to the version on 03-20-2017   I think things will work better for you.

Tannoo, 2017-03-24:
hmmm.. nope. I'm not good enough to do that.

Roxy-3D, 2017-03-24:
Well....    It probably is going to take me the rest of the day to get things back to where they were.   Take a break!

Bob-the-Kuhn, 2017-03-24:
Would 3-21 RCBugFix do?  I have a copy of one from the middle of the afternoon on that day.

Tannoo, 2017-03-24:
Yeah. I'll be going to work in a few hours anyway.

I did do `G26 P10 C O3.5` that worked fine. Although, the nozzle wanted to plow into the bed and my nozzle started jamming. I could not cancel it from clicking or holding. So...I reset Marlin. That stopped it.

[21 more replies]

closed 2017-11-01 by thinkyhead