MarlinFirmware/Marlin #6112: Black Screen with  USE_BIG_EDIT_FONT ?!

fred2088, 2017-03-24:
when 
#define USE_BIG_EDIT_FONT

and comment out the following two lines
// #define XYZ_HOLLOW_FRAME
// #define MENU_HOLLOW_FRAME

the edit item will be blank screen ?

thinkyhead, 2018-02-06:
Still a problem today with 1.1.8?

boelle, 2019-02-18:
@fred2088 still a problem with bugfix 2.0?

if not please close this one

boelle, 2019-03-12:
@thinkyhead i think we can close this one, no activity from OP

boelle, 2019-03-16:
i just did the edits suggested in first post but i dont think it will have effect

[image]

seems that HAS_GRAPHICAL_LCD  needs to be active first and i have a full graph display

Phr3d13, 2019-06-10:
Can this be closed?

closed 2019-06-11 by fred2088