MarlinFirmware/Marlin #6113: PrintrBoard Rev F missing DAC_STEPPER_DFLT information

lucacri, 2017-03-24:
Hello, I just downloaded the latest RC-BugFix and it fails to compile on a PrintrBoard Rev.F with the following 

```
stepper_dac.cpp:52: error: 'DAC_STEPPER_DFLT' was not declared in this scope
   uint16_t dac_channel_pct[XYZE] = DAC_STEPPER_DFLT;
                                    ^
'DAC_STEPPER_DFLT' was not declared in this scope
```

I checked and the only place where DAC_STEPPER_DFLT is defined is in the pins_RIGIDBOARD_v2.h . 

The file pins_PRINTRBOARD_REVF.h needs it too

Bob-the-Kuhn, 2017-03-25:
PR #5792 on 9 FEB implemented that change.

Here's a version of the file that has the PR #5792 changes made as conditional.

Hope you're not running an LCD.  That also has an issue when stepper DACs are enabled.

[stepper_dac.zip](https://github.com/MarlinFirmware/Marlin/files/869772/stepper_dac.zip)

Bob-the-Kuhn, 2017-03-25:
I took a harder look at the DAC_STEPPER_DFLT changes.  One of the changes was to initialize the current percents at power up.  The only other way to initialize them is via the LCD.

You are correct, the better long term fix is to add the defaults to the pins_xxxx.h files or maybe even the configuration file.

Bob-the-Kuhn, 2017-03-25:
I just looked at the mcp4728 data sheet and it comes from the factory with the DAC outputs set to zero.  If you want something else it's either the LCD or DAC_STEPPER_DFLT

Would it be better to have DAC_STEPPER_DFLT in the configuration file?  People are used to editing that file when they want to change an option.

Bob-the-Kuhn, 2017-03-25:
I've opened PR #6124  to move this into Configuration_adv.h & to fix the LCD issue.

fiveangle, 2017-03-26:
I'm not sure if this is so cut-n-dried as your PR #6124.  I posted to #4289 but really should be here it seems. Are there really 3 different methods for setting motor current ? I remember being confused when originally configuring RC6 for Printrbot 1403 and asking myself, "how do I know if my DAC is connected via I2C or not ?"  There's definitely some confusion around this that we should tackle if possible. What I posted to the other issue:

I'm trying to undwind this too, due to recent conditional added in #5792 (as Bob identified in #6113,, which I also hit when attempting to move from @Sebastianv650 's unified timer branch (RC7-based) to RCBugFix).

In Configuration_adv.h we have:

> // Motor Current setting (Only functional when motor driver current ref pins are connected to a digital trimpot on supported boards)
> **_#define DIGIPOT_MOTOR_CURRENT {135,135,135,135,135} // Values 0-255 (RAMBO 135 = ~0.75A, 185 = ~1A)_**
> 
> // Motor Current controlled via PWM (Overridable on supported boards with PWM-driven motor driver current)
> //#define PWM_MOTOR_CURRENT {1300, 1300, 1250} // Values in milliamps
> 
> **_// uncomment to enable an I2C based DIGIPOT like on the Azteeg X3 Pro
> //#define DIGIPOT_I2C_**
> // Number of channels available for I2C digipot, For Azteeg X3 Pro we have 8
> #define DIGIPOT_I2C_NUM_CHANNELS 8
> // actual motor currents in Amps, need as many here as DIGIPOT_I2C_NUM_CHANNELS
> #define DIGIPOT_I2C_MOTOR_CURRENTS {1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0}

And in pins_PRINTRBOARD_REVF.h we have:

> **_// uncomment to enable an I2C based DAC like on the Printrboard REVF
> #define DAC_STEPPER_CURRENT_**
> // Number of channels available for DAC, For Printrboar REVF there are 4
> #define DAC_STEPPER_ORDER { 3, 2, 1, 0 }
> 
> #define DAC_STEPPER_SENSE    0.11
> #define DAC_STEPPER_ADDRESS  0
> #define DAC_STEPPER_MAX   3520
> #define DAC_STEPPER_VREF     1 // internal Vref, gain 1x = 2.048V
> #define DAC_STEPPER_GAIN     0
> #define DAC_OR_ADDRESS    0x00
> about:blank

Printrboard RevF has potential to support 5 digital trim pots (including second extruder when expansions board installed as on Plus with dual extrusion upgrade)

Nowhere does **_DAC_STEPPER_DFLT_** exist outside of supporting just the RIGIDBOT that I can find.

Thoughts ?

-=dave

Bob-the-Kuhn, 2017-03-26:
I haven't been into the world of DACS/digi-pots before.  Now that you've pointed out the digi-pots I'll have to do some investigation.

At first glance it appears that DAC_STEPPER_DFLT and DIGIPOT_MOTOR_CURRENT perform similar functions.  Right now I have no idea why one would be used and not the other or if one is a superset of the other.

"Motor Current controlled via PWM" is also something I have not looked at.  I don't even know if the PWMs are the ones on the 2560 or are on dedicated chips.

It's time for me to wind down & call it a night.  Tomorrow's busy for me so it may be late Sunday or Monday morning before I have enough knowledge to get back to you with confidence.

What DAC is on the Printrboard rev F?

Bob-the-Kuhn, 2017-03-26:
Here's some top level stuff.

DIGI_POT is aimed at single or dual 4 channel mcp4451 DAC based systems.

There is no LCD support for it.

It uses the digipot_mcp4451.cpp software.

PRINTERBOARD and V2 use the stepper_dac.cpp and dac_mcp4728.cpp software

No idea if the mcp4451 and the mcp4728 are register compatible.

Bob-the-Kuhn, 2017-03-26:
If DAC_STEPPER_CURRENT is defined then the mcp4728 RIGIDBOARD_v2 & PRINTRBOARD_REVF code is used (stepper_dac).  M907, M908, M909, M910 & LCD can set the current.

If DIGIPOT_I2C is defined then DIGI_POT mcp4451 AzteegX3pro code is used (digipot_mcp4451). M907 & M908 can set the current.

The two chips are different enough that they need different software.

A4JP, BQ_ZUM_MEGA_3D & RAMBO also have DACs.  I only have the schematic for one & it uses the Ad5206.  This is enabled if the pin DIGIPOTSS_PIN is enabled.  It uses the M907 to set the current.  Power up current is set by DIGIPOT_MOTOR_CURRENT in Configuration_adv.h.  DIGIPOT_MOTOR_CURRENT is also used by the AzteegX3pro.

I'm not touching the code more than what I've already done.  

I will try to re-org the Configuration_adv.h section to make it clear which boards use which items.

Bob-the-Kuhn, 2017-03-26:
Just finished rewording & testing to be sure I had it right.  The PR has been updated with the new wording.

fiveangle, 2017-03-27:
What mess, but I think that's the best you can do w/o re-architecting a more universal fix.  What probably needs to happen is to collapse Configuration*.h to a single "STEPPER_CURRENT" that is set as 0%-100%, then push the actual calcs for setting both the DAC type, bus, etc to pins_*.h for each board.

thinkyhead, 2017-04-02:
#6124

cscott, 2017-07-13:
With the fix in #6124, this seems to be working fine again.  (Tested on printrboard rev f with https://github.com/cscott/Marlin/tree/xilka2 ).  I think this issue can be closed.

fiveangle, 2017-07-14:
I confirm it's resolved too. Whoever has god mode, please do the needful.

closed 2017-07-14 by Roxy-3D