MarlinFirmware/Marlin #6116: compiler error after eab7854

christianh17, 2017-03-24:
I just updated rcbugfix (eab7854) and wanted to compile and update my printer. I got the following error (unfortunately I can't mark as code it changes back too early):
--------START
sketch\Marlin_main.cpp: In function 'bool shove_and_echo_command(const char*, bool)':

Marlin_main.cpp:825: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
     echo_command(cmd);
                     ^
Marlin_main.cpp:799: error: initializing argument 1 of 'void echo_command(char*)' [-fpermissive]

 inline void echo_command(char * const cmd) {` 
There are more errors like this further on
--------END
Am I the only person with this problem? Than I would have to  investigate further my own changes.
Regards
Christian

NiklasOlofsson, 2017-03-24:
Not only you. 

`  // Disable the leveling matrix before homing
  #if PLANNER_LEVELING
    #if ENABLED(AUTO_BED_LEVELING_UBL)
      const bool bed_leveling_state_at_entry = ubl.state.active; <- It does not like this among other warnings.
    #endif
    set_bed_leveling_enabled(false);
  #endif`

benlye, 2017-03-24:
@christianh17, take a look at PR #6111 which I created to fix the problem you (and I) are having.  The fix is simple, I think.

christianh17, 2017-03-24:
@benlye: Cool, it works :-). 
Thanks!

closed 2018-02-06 by thinkyhead