MarlinFirmware/Marlin #6117: BLTouch fails intermittantly Marlin 1.1.0-RCBugFix

dvseliteus, 2017-03-24:
I finally got my [Geeetech BLTouch](https://www.aliexpress.com/item/Free-Shipping-Geeetech-Auto-Leveling-Sensor-3D-Touch-for-3D-Printer-Improve-Printing-Precision/32761924885.html?spm=2114.13010608.0.0.lVuWUV) working with the Marlin 1.1.0-RCBugFix version.
It was working great but after about 15 prints the sensor is failing now about 80% of the time.
See the video here: https://youtu.be/n-434AEZeSQ
It fails during the second touch on G28 and also fail on random points during G29.
When I run the M280 P0 S120 ; Self test it either works fine or fails after the 8th interval or so.
Here is my configuration: [Configuration.h](http://pastebin.com/GgDpej4K)
I have tested 2 sensors and they both have the same issue.
Anyone else having this issue?

DuoDreamer, 2017-03-24:
I have been running RCBugFix for a couple of weeks, printing for several hours every day, with 9-point bed probing before every print, and have not had a single problem with my Genuine BLTouch. Perhaps your knockoff is not functioning correctly? If your unit flakes out during self-test, that sounds like it's not functioning correctly and is not a Marlin issue.

dvseliteus, 2017-03-25:
@DuoDreamer thanks for the quick reply. Looks like I'll be buying a [Genuine BLTouch](http://a.co/4xXELmo).

Anyone running a Geeetech successfully?

DuoDreamer, 2017-03-25:
One thing I can suggest you try is to make sure that the wiring for the unit is thick and the power supply for it is strong. The genuine unit is known to draw upwards of 300mA at times while operating the pin. It's entirely possible that your wiring or connectors are super-thin in spots and are unable to supply the full current load.

dvseliteus, 2017-03-25:
I wouldn't have thought it was a power issue but that makes a lot of sense on why it's acting up.
I replaced the power supply with a Mean Well 350W 12V (NES-350-12) a while back.
I have some 24AWG 5050 LED wire. I'll make a new harness on Monday and see if that resolves the issue.
I also have a Genuine BLTouch on the way Monday so I will also test that and see if the same thing happens.
Thanks @DuoDreamer

Bob-the-Kuhn, 2017-03-25:
I ran a Geeetech successfully until I clicked the wrong button and crushed it.  More are on order.

I don't see any issues with your configuration.h file.

I think your pin needs to be cleaned.  That happened on my BLTouch after a couple of hours.

If it won't pass M280 P0 S120 for several hours it's not reliable enough for probing.  

I'm not sure I understand what I saw in the video.  If the pin were sticky the Geeetech should have gone into the error state almost immediately.  It's as if the unit didn't get the deploy command.  Check your 3 pin cable to see if there is a poor connection somewhere.

Dreamloom, 2017-03-25:
Over-heating smaller gauge coil in the knock-off than in the real item?

dvseliteus, 2017-03-27:
I upgraded the wiring harness to 24AWG and there seems to be an improvement. I had the fault happen so far 1 out of 5 prints. I ran the self-test for about 10 mins and everything looked okay.
- still waiting for original bltouch to arrive and test
- contacted Geeetech waiting for their support

I do not see the issue to be in Marlin and I assume it's the hardware at fault here.
@Bob-the-Kuhn weird right? The pin is not stuck and moves freely but it mimics that effect which is really strange behavior. I'm not sure if this is an overheating issue as the self-test runs now with the new wiring.

Bob-the-Kuhn, 2017-03-27:
I'm pretty sure the coil isn't overheating.  That current pulse is only there when deploying & stowing.  If it were a problem you'd see it with the self test.

I've seen reports of other BLTouch problems going away after upgrading the wiring and/or going to a separate 5V supply..

dvseliteus, 2017-03-27:
@Bob-the-Kuhn just got the Original BLTouch and it's performing with no issues.
I'll keep the thread open until I hear back from Geeetch support then close it.
I think your right it has to do with the current pulse is only there when deploying & stowing.

Roxy-3D, 2017-03-28:
> I ran a Geeetech successfully until I clicked the wrong button and crushed it. More are on order.

Bob...  If you want a Geeetech probe...  I can't help you.   But I can send you more authentic BL-Touch probes.

> I'm pretty sure the coil isn't overheating. That current pulse is only there when deploying & stowing. If it were a problem you'd see it with the self test.

> I think your right it has to do with the current pulse is only there when deploying & stowing.

I know for a fact the authentic BL-Touch probe does a monster current spike when it deploys and retracts the pin.  I have some debug LED's wired up to the same voltage source as the BL-Touch and they dim noticeably during the deploy and retract.

DuoDreamer, 2017-03-28:
I should note that I am using a 24V printer that has a secondary 24-to-5V 8A regulator to provide power for my electronics and LED lighting. I'm not attempting to sap 5V from the *tiny* 1A regulator included with most arduino-compatible controller boards (and USB power is disconnected). This is probably why I haven't had the unit sap enough power to break itself or the arduino mega.

Roxy-3D, 2017-03-28:
> I'm not attempting to sap 5V from the tiny 1A regulator included with most arduino-compatible controller boards (and USB power is disconnected). 

Ditto.   I have a LM7805 connected to my printer's 12 volt supply to power my debug LED's and servos.    It should be plenty of current.   And yet...  I see the LED's dim.

DuoDreamer, 2017-03-28:
I gave up and went with a dedicated high-amperage 5V regulator due to the status LEDs and head illumination (ws2812 strips). I'm getting a fairly steady 4.95V out of it most of the time, and slightly less when the bed and head are at 100/200C.  I should have a second genuine BLTouch arriving in a couple of weeks to test against as well, to see if there are any significant differences between the old units and the current units.

DuoDreamer, 2017-03-28:
> Ditto. I have a LM7805 connected to my printer's 12 volt supply to power my debug LED's and servos. It should be plenty of current. And yet... I see the LED's dim.

I think that if the BL draws 200-300ma normally and spikes, plus you have LEDs (20-60ma each), plus actual servos (9 gram minis can suck 200-300ma each), and powering the controller board/driver logic, you're probably pumping most of what the 7805 can push under normal circumstances (1-1.5A). Maybe try an LM323-like regulator? (3A+)

dvseliteus, 2017-03-28:
I'm running my BLTouch off an Azteeg X3 and Viki LCD. I did notice that both the  BLTouch Led and LCD backlight dims quite a bit when the pin drops and raises. This is proven for both ANTCLABS and Geeetech BLTouch sensors.

So basically the options are to drop an in-line regulator like a 7805 or LM323 into the wire harness of the Sensor?
Would another other alternative would be to run a separate dedicated PSU like a meanwell [NES-35-12](http://a.co/em95l7h)?
Shouldn't ANTCLABS have built this into their sensor to prevent such high current draw or is this issue not as common as it seems?

DuoDreamer, 2017-03-28:
The NES-35-12 is a 12V regulator, not a 5V regulator. A 7805 or LM323 just for the BLTouch would probably be a little overkill, but would work, but wouldn't be in-line on the existing connector, it would be an offshoot between the 12V power supply and the 5V pin on the BLTouch.

dvseliteus, 2017-03-28:
Thanks @DuoDreamer I have no experience in these electronics as you can tell.

elfag, 2017-03-28:
I use 3D touch on my geeetech pro B. Got some problem before but work great when i got the latest RC8

dvseliteus, 2017-03-28:
I'm not sure if this will resolve the Geeetech issue but here is what I got from the support rep.
https://youtu.be/Ix7lXvfkPEU

dvseliteus, 2017-03-29:
The Original BLTouch is performing flawlessly. Thanks all for the help and I will still be working on the Geeetech Sensors on my other printer to see if the regulators will do the trick.

closed 2017-03-29 by dvseliteus