MarlinFirmware/Marlin #6120: The new functions in Marlin_main.cpp won't compile except at a very low warning level.

Roxy-3D, 2017-03-25:
The new functions in Marlin_main.cpp won't compile except at a very low warning level.

>  inline void echo_command(char * const cmd) {
>              ^
> sketch\Marlin_main.cpp: In function 'bool enqueue_and_echo_command(const char*, bool)':
> sketch\Marlin_main.cpp:905:21: warning: invalid conversion from 'const char*' to 'char*' [-fpermissive]
>      echo_command(cmd);
>                      ^
> sketch\Marlin_main.cpp:799:13: note: initializing argument 1 of 'void echo_command(char*)'
>  inline void echo_command(char * const cmd) {
>              ^

Tannoo, 2017-03-25:
#6116

Tannoo, 2017-03-25:
The way Thinkyhead was talking, they might all need to be `(char * const cmd)`.

Roxy-3D, 2017-03-25:
OK!   Its being addressed....    I'll close it!

THANKS!

closed 2017-03-25 by Roxy-3D