MarlinFirmware/Marlin #6127: X/Y/Z flash to Question Marks on Power On

stampede-arm, 2017-03-25:
When I power my machine on (1.1.0 RC-8) the spaces for X-Y-Z co-ordinates flash the XYZ to ? (question marks) and back and forth. I cant move my motors until I do an auto home, even if endstops and software endstops are disabled. I don't have any endstops and this recent functionality is not documented anywhere that I can find. How do I work around this?

My machine is a pain to use at this point because of this.

stampede-arm, 2017-03-26:
And also, since disabling the stepper pins in the ramps pins file like http://ohmeye.com/2013/disablingenabling-endstops-in-marlin/ , auto home raises my Z up 5mm and then still flashes the ? and doesn't let me move motors with Prepare > Move Axis.

Bob-the-Kuhn, 2017-03-26:
The question marks are just saying that the machine doesn't know where it's at.  

It sounds like you had your machine working and only had problems after switching to a newer version of Marlin.  Is that correct?  What did you go from and what did you go to?

No endstops - that's the way I ran when I first got my machine running.  I'm trying to remember what it took to do that.

I just tried to do that but I can't get by the sanity check.  It's insisting on an X, Y & Z endstop.

If you do a M119 command and it says that both the MIN and MAX endstops are triggered then that axis isn't going to move.  In that case enable the endstop pullups and set all the endstop invertings to true.  After that M119 should only report open.

----

If you want you can drop your config files (both old/working and new/problem versions) onto your reply & I'll see if I can spot anything.

You'll either need to ZIP them up or rename them to xxx.h.txt before dropping them.  GitHub only allows certain file extensions when dropping.

stampede-arm, 2017-03-26:
=( This is my R & D machine that has never worked up until just recently, when I finally got a new hotend for it. Shortly afterwords I fried the Ramps Board. I never had the original firmware. I'm trying to upgrade to a MKS Base 1.0, which is why I'm using RC8 (includes functionality) My other machine works on an older version of Marlin with no issues with endstops. (Don't use them on that one either.)

Bob-the-Kuhn, 2017-03-26:
I have a 2560 based controller so I could try to reproduce your problem if I had your config files.

I'd be really interesting to see what M119 reports.

stampede-arm, 2017-03-26:
M119 reports back
SENT: M119
RECEIVED: Reporting endstop status
RECEIVED: ok

Bob-the-Kuhn, 2017-03-26:
I'm out of ideas until I try to reproduce your problem.

K3nt, 2017-04-07:
Can you help me?  I'm having the same trouble.  And my machine stuck  for a year and counting. How can I attach my config here?

Bob-the-Kuhn, 2017-04-07:
If the file extension is .txt or .zip you can just drop the file on the comment window. 

I'll need both your config files and which version of Marlin that you're using.

Once we get your printer working I hope you upgrade to a newer version of Marlin.  Many more features and better/faster LCD support.

When using the USB port - how functional is your printer?  Does it respond to any commands?  

What does your LCD report for temperatures?

K3nt, 2017-04-07:
Im using anger a8 3d printer. And I change the board with ramps 1.4a and
ever since I haven't succeeded yet.

Marlin v1.1.0-rc8

Via usb,  I don't get any voltage operation and no movement on motor

Heat bed and filament heater are working fine.

/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
[... 1372 lines cut ...]
 * 34 - RAMPS1.4 - uses Analog input 5 on the AUX2 connector
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H

closed 2017-10-31 by thinkyhead

willytwo, 2020-04-09:
> When I power my machine on (1.1.0 RC-8) the spaces for X-Y-Z co-ordinates flash the XYZ to ? (question marks) and back and forth. I cant move my motors until I do an auto home, even if endstops and [...]

> The question marks are just saying that the machine doesn't know where it's at.
>
> It sounds like you had your machine working and only had problems after switching to a newer version of Marlin. Is [...]

I have the same issue. Were you ever able to resolve it?
I have an Anet A8 and just upgraded from stock firmware to Marlin ver 4. 
Any help would be appreciated.

thinkyhead, 2020-04-27:
@willytwo — 
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).