MarlinFirmware/Marlin #6128: SD card print reboot MEGA

Chegevarko, 2017-03-26:
Hi
when i print some by SD card  RAMPS+MEGA go to reboot, print by usb work fine
some thing is bed work

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/870627/Configuration.h.txt)

BillyBob54, 2017-04-03:
+1 - I use two different SD cards, one is a 128M and one is a 16G card.  Never have problems with the 128M.
configuration: Ramps 1.4, RCBUGFix (25 April down load- but also have seen problem on RC 8) RepRap discount LCD Controller
Symptom:  if using the 16G SD Card-
1. SD card in SD reader, printer powered on.  Everything seems to work OK.  If I then connect Repetier Host via USB, the printer goes into a repeating reset mode.  If I connect the printer w/o the SD card in the reader, everything works OK, but when I insert the SD card- things go "haywire"- generally this manifests as a freeze up of Repetier Host as well as LCD Control- even the reset on the mega 2650 does not work.  The 128M SD card seems to work OK.  I tried setting the  SD buffer in the advanced config file to 32; didn't correct and I received a memory warning during compile so I set it back to 16.

BillyBob54, 2017-04-26:
I have been printing almost entirely from the SD card lately, I understand that RCBugFix will be released in the next couple of days as 1.1.0.  After the release, I will load up 1.1.0 and see if this is still a problem.

BillyBob54, 2017-05-15:
Loaded up RCBug Fix 1.1.0 on 13 May- problem persists.

closed 2018-02-06 by thinkyhead