MarlinFirmware/Marlin #6129: HOMING_FEEDRATE_Z resetting on every print

mattogodoy, 2017-03-26:
Hello!

I'm using **1.1.0-RC8** and I've set `#define HOMING_FEEDRATE_Z  (3*60)` because that's the fastest my motors can go.

When I turn on the printer and check the value on the menu in the LCD, I can see `Vmax Z: 3`. So far so good. The problem is that when I select a file to be printed from the SD card, this value returns to **4**, causing my motors to stop while trying to cope up with the speed.

I've found a way around it and it's to select a file from the SD card to print, and while the hotbed and the nozzle warm up, I change the `Vmax Z` value back to **3**. Then the motors work good.
I have to do this in every print.

I've checked if it's caused by some old value stored in the EEPROM, but I don't think this is the case, because each time I turn the printer on, the value is correctly set to **3**. It only changes to **4** when it starts to print.

This is my Configuration.h just in case: [http://pastebin.com/K5ywnwNr](http://pastebin.com/K5ywnwNr)

Any ideas?
Thanks

fiveangle, 2017-03-29:
Slicer sending M203 in gcode ?

mattogodoy, 2017-03-29:
I took a look again into the GCODE and as you say, Slic3er was actually sending a 203 code:
`M203 X192.00 Y208.00 Z4.00 E25.00`
but I did not pay attention to it before because I thought the maximum values set by firmware in the printer where immutable, some sort of failsafe. Is this right, or M203 overrides the firmware maximums?

thinkyhead, 2017-04-02:
```cpp
/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 200, 200, 20, 20 }
```

closed 2017-04-02 by thinkyhead