MarlinFirmware/Marlin #6130: How to abort a failing G26 P C On.nnn

WheresWaldo, 2017-03-26:
So you go through the U.B.L. process and want to verify the mesh. You issue a proper G26 command to do just that, then you realize that something is wrong. Short of pulling the plug on your printer how do you stop the MESH Validation Print?

Roxy-3D, 2017-03-26:
For both G26 and G29 P1 commands....     You Press and Hold the encoder wheel.   The abort is not instantaneous.   But it will happen within 1 mesh point of the current operation.    For both of those commands, you can limit 'danger' by starting the nozzle in a corner and not doing the 'C' on G26.   Without the 'C',   the algorithm will try to work in an increasing diameter from the starting point.  

So...   if the nozzle is getting too close to the glass...  It can only get closer at a rate where you have time to stop the operation.

WheresWaldo, 2017-03-26:
Not the issue of gradually gouging the glass, rather the same issue as before, ignoring M851 no matter what it is set to. Also a question about whether or not the MESH needs to be reinitialized if you change only the offset.

Roxy-3D, 2017-03-26:
> Also does the MESH need to be remeasured if all you change is the Z Offset?

If you mean Z_PROBE_OFFSET_FROM_EXTRUDER, then practically speaking... "Yes!"   The reason is that number enters directly into the numbers that are generated during the mesh generation.

If you mean z_offset, then "No."   The z_offset just gets added blindly to any generated Z Height Correction that is calculated.

Roxy-3D, 2017-03-26:
It should not be ignoring M851 if you are adjusting the Z_PROBE_OFFSET_FROM_EXTRUDER via that mechanism.   You are saving the M851 value in EEPROM, right?   You should be able to verify the number really did get saved and is being used by resetting the printer and doing an M503 to display all the values.   You should see the value that you entered...   

UPDATE:    Huh!!!   You found a bug!!!   The G29 P1 code looks like this:
```cpp
        const float measured_z = probe_pt(xProbe, yProbe, stow_probe, g29_verbose_level);
        z_values[location.x_index][location.y_index] = measured_z + Z_PROBE_OFFSET_FROM_EXTRUDER;
      }
```
it should be
```cpp
        const float measured_z = probe_pt(xProbe, yProbe, stow_probe, g29_verbose_level);
        z_values[location.x_index][location.y_index] = measured_z +  zprobe_zoffset;
      }
```
I keep my EEPROM identical to my Configuration.h settings...   That is why I didn't see the problem.   If you search in UBL_G29.cpp and change that line...    I think it is going to start acting like you want it to behave.   Please report back with what you see happen!

Sorry about that!

WheresWaldo, 2017-03-26:
Okay, I knew I wasn't being dense, yay! I will test it in a bit need to get rid of this IRL headache I got while working on this issue this morning.

When I started working on MESH, I had a few exchanges with Ed Patel and I know about what affect EEPROM has on your settings. I also have a small group of people following my progress with Marlin 1.1.0 using a Robo3D R1. Most of them I have converted to MESH as I have been a big proponent of it since its inception. I see U.B.L. as MESH on steroids and will push all my Robo3D friends toward it once I figure it out.

Funny you mentioned the gMax 1.5+ in another issue, I just helped a buddy upgrade his gMax to Marlin 1.1.0 RCBF using MESH. I think he would love U.B.L. and its ability to make life easier with so many probe points. I will report back sometime after the NC/Kent NCAA game.

Roxy-3D, 2017-03-26:
> Funny you mentioned the gMax 1.5+ in another issue, I just helped a buddy upgrade his gMax to Marlin 1.1.0 RCBF using MESH. I think he would love U.B.L. and its ability to make life easier with so many probe points. 

Send him over here!   :)   I'll help him get it up and running.    I'm putting together a gMax 1.5+ version right now so he should be able to just load it and go.

WheresWaldo, 2017-03-26:
Working I am closing this ticket!

closed 2017-03-26 by WheresWaldo