MarlinFirmware/Marlin #6135: AUTO_BED_LEVELING_*** ignores boundaries and points

iurobpn, 2017-03-27:
Hi, I'm using Marlin 1.1.0 RC8, and a RAMPS 1.4. Each AUTO_BED_LEVELING on marlin ignores points or boundaries set by using constants:

 ```
 #define LEFT_PROBE_BED_POSITION 45
  #define RIGHT_PROBE_BED_POSITION 170
  #define FRONT_PROBE_BED_POSITION 30
  #define BACK_PROBE_BED_POSITION 160
```

or, for 3 points:

```
#define ABL_PROBE_PT_1_X 45
#define ABL_PROBE_PT_1_Y 160
#define ABL_PROBE_PT_2_X 45
#define ABL_PROBE_PT_2_Y 30
#define ABL_PROBE_PT_3_X 170
#define ABL_PROBE_PT_3_Y 30

```
The printer always start at a point out of boundary, making ir try to probe at a point where the inductive sensor is out of bed, possibly causing harm to the extruder and or bed.

I'm trying to make the autoleveling feature works for 3 days, and I think it is possible that its a Marlin's bug. My Configuration.h is in the [pastebin](pastebin.com/cMHT69P9). Any help? Is it a bug, or I'm missing something? 

ps: I have look extensively on google, and in other issues.

Roxy-3D, 2017-03-27:
Are you clearing the EEPROM with M502 and M500?   That pastebin link isn't working for me.

iurobpn, 2017-03-27:
I fixed the link above. No, I didn't clear the EEPROM, I will try and post the result later.

wookie666, 2017-04-02:
Did this work? If you have the correct offsets set for the z probe, it should work.

thinkyhead, 2017-04-02:
Be sure to test the `RCBugFix` branch to see if the newer code resolves the issue.

iurobpn, 2017-04-03:
It worked, Thank you. Sorry I took a long time to answer...

closed 2017-10-31 by thinkyhead