MarlinFirmware/Marlin #6136: Suggestion:  Add Travis check

Roxy-3D, 2017-03-27:
It would seem to me we should have a Travis check that ensures 
```cpp
// #define MIN_SOFTWARE_ENDSTOPS 
```
is turned off when either Mesh or UBL is enabled.   I suspect this should be turned off for Bi-Linear, 3-Point and Grid leveling also.    But it would also seem to me we should have more granularity.   I think we want to insist this is turned off for the Z-Axis if bed leveling is enabled.   But we probably want to leave it active on the X & Y Axis.

Comments and suggestions are welcome!

Blue-Marlin, 2017-03-27:
> It would seem to me we should have a Travis check that ...

I'd agree if you'd talk about a __Sanity__ check.

teemuatlut, 2017-03-27:
Wouldn't this be something to be handled on the stepper/planner level?

Roxy-3D, 2017-03-27:
>> It would seem to me we should have a Travis check that ...

> I'd agree if you'd talk about a Sanity check.

Yes.   I presume that if it is in the SanityCheck.h file, it will cause an error and Travis would blow up.  But "Yes!" what I really want is for the user to know that the nozzle can't fully compensate for a bed that has a dip in it if this option is turned on.

thinkyhead, 2018-02-06:
I guess we all now know that software endstops only constrain movement before leveling and skew, so the nozzle is allowed to go outside these endstops in the physical space, so if you have very tilted bed, you will not end up with a leaning tower with one flat side.

closed 2018-02-06 by thinkyhead