MarlinFirmware/Marlin #6137: UBL Mesh adjust encoder wheel doesn't move the hotend G29 P4

WheresWaldo, 2017-03-27:
Tried to adjust the mesh, have about 10 points all on one side of the bed and adjacent to each other that need adjustment. Tried to issue the G29 P4 R 10 X219 Y115 (that is about the middle of the right side). LCD told me it was in adjustment mode and told me the current measurement of the nearest mesh point. Tried turning the encoder wheel and nothing happens. Turned it both directions. Pushing the encoder button moves to the next point with no other up or down movement. Basically can't adjust the mesh. As an experiement I went to OctoPrint and tried to move the Z axis there with the control menu, because that is more intuitive than trying to figure out what commands to put in terminal. still no movement. Then tried to abort by holding the encoder wheel. Held the wheel over 30 seconds and nothing. pressed the wheel and nothing. Printer was locked up. Only the kill button would exit, then of course all is lost, but since nothing actually happened it was not a big deal.

Other sort of related info. Its nice that you go to the nearest spot to the hotend when doing these adjustments, but working with calculated coordinates that are never displayed make it very hard to make this anykind of easy task. Ether report the actual X/Y corordinates or let me specify that I want to edit matrix point 10,5 if I want the middle of the Y at the far end of X on a 100 point grid.

Roxy-3D, 2017-03-28:
> Tried turning the encoder wheel and nothing happens. Turned it both directions. Pushing the encoder button moves to the next point with no other up or down movement. Basically can't adjust the mesh. 

I can't check this out for several hours...  My printer is doing a big print.   But I'll check it ASAP.   However, the Interactive Mesh Editing got broke during some merging of code 3 days or so ago.  It got fixed promptly, but if you grabbed your version of RC-8's RCBugFix during that time window, that is what you are seeing.

If your version of Marlin is roughly 3 days old...  Please refresh it and the problem you are seeing should go away.

WheresWaldo, 2017-03-28:
I am pretty sure I am using Sunday's files, but I will download them fresh tomorrow morning to check if it's fixed.

Bud

Roxy-3D, 2017-03-28:
Yikes!!!!   My print finished and I checked out what you said.   You are correct.   That behavior is back.

I'll get that fixed as fast as I can.   Realistically...   It will be the middle of the afternoon before I have an updated version available.

Roxy-3D, 2017-03-28:
@WheresWaldo    I'm making good progress resolving the issue.   But I just wanted to give you a "Heads Up" that it will be later this afternoon before I have the updates to the code in RCBugFix for you.

Roxy-3D, 2017-03-28:
@WheresWaldo    I think I have everything working for you.    The numbers don't roll as nicely as I want them to in the Interactive Mesh Editor.   Depending upon how fast you have your encoder wheel set, some times it is hard to land on a .005 in the third decimal place.   But for right now, you can hold it between stops and do the click if you really want a mesh point set at that level of accuracy.

And G26 aborts with a Press and Hold also...

WheresWaldo, 2017-03-29:
Cool, I can check it much later today as its past my bedtime.

Roxy-3D, 2017-03-29:
> Cool, I can check it much later today as its past my bedtime.

Not a problem!   And like I said...  I'm not 100% happy with how the numbers roll.   But it does work.   You don't edit the mesh that often, and very seldom do you need to edit it to .005 mm resolution.   What we really care about is....   Do you get good adhesion across the entire bed?

WheresWaldo, 2017-03-29:
Actually I use extrusion multiplier to get good adhesion on layer one. I know that traditionally you would squeeze layer 1 to get good adhesion but I think that is ass backwards. If I want a .25 layer 1, I want it to be .25 not .20 to get adhesion. So both Slic3r and S3D allow me to adjust layer 1 extrusion multiplier or extrusion width to extrude enough filament to get good adhesion and I don't have to mess with layers smaller than what I actually asked for.  Anyway I would think 0.005 mm is pretty close to most peoples physical limits, especially since my setup has each Z axis step = 0.01 mm can't have fractional steps so that is my lower limit anyway.

Is there a place other than MBL_Z_STEP to define the minimum step size for adjustments?

Roxy-3D, 2017-03-29:
> Is there a place other than MBL_Z_STEP to define the minimum step size for adjustments?

If you adjust this number....  UBL won't even know it.   It doesn't use that.    Instead, the concept is UBL takes over the LCD Panel for certain operations.  It inherits the configuration of the LCD Panel as set up.   So...   How you set up ENCODER_RATE_MULTIPLIER is going to affect how the display 'feels' when adjusting mesh numbers.

WheresWaldo, 2017-03-29:
Downloaded RCBF an hour ago, here is the result. Encoder spins, numbers scroll on the LCD, Axis doesn't actually move.

Roxy-3D, 2017-03-29:
> Downloaded RCBF an hour ago, here is the result. Encoder spins, numbers scroll on the LCD, Axis doesn't actually move.

Yes.  That is correct.

WheresWaldo, 2017-03-29:
Okay then what was the point? Isn't that how you adjust the mesh?

Roxy-3D, 2017-03-29:
No...     If instead you do a G29 P4 R O    You will see the mesh point raise or lower to what ever you specified.    The nozzle does not need to move up or down during the mesh editing cycle.   It is only positioned in (X,Y) so you know which part of the mesh validation pattern you are correcting.   And it is pretty easy to look at that and say "This mesh point should be .1mm lower because I'm not getting good adhesion here."

WheresWaldo, 2017-03-29:
Okay then I am going to have to rethink this whole process. Speak at me like I am a five year old and explain the thought process, because I am not seeing how to determine how much adjustment I actually need to move other than guessing based on the numbers displayed.

Roxy-3D, 2017-03-29:
Well....  If you do a G26 Mesh Validation Pattern across the bed, You will see sections or areas that are real good and the filament is squished just the right amount.   Probably the area where you homed the machine's Z against the bed.   But other areas will be too squished or maybe the filament doesn't stick at all.

Depending upon what you see for a mesh validation pattern....   It should be fairly obvious whether the nozzle needs to be closer to the bed at that location or further away from the bed.    In the bad areas, you just edit the mesh point up or down the 'Right' amount.   

If you don't know how much to raise or lower a mesh point based on the G26 pattern....   Just always move the mesh point .1 mm  in the right direction.   Run another G26 and repeat the process.   You will get a perfect mesh pretty quickly even with a fixed .1 mm movement.

WheresWaldo, 2017-03-29:
Okay I understand what you did there with the process. 

This is totally different than I thought the process would work. Since auto generating the mesh basically determines Z0 all across the bed (maybe this is incorrect), if I have a section where the bed flexed or is warped a bit more when heated I need to adjust. My thinking was that I could just go to the section of the mesh that needed adjustment and reuse the nozzle to re-determine where Z0 is for that point. Kinda like baby-stepping to fix the suspect area.

Roxy-3D, 2017-03-29:
Probably...  You should heat the nozzle and bed before auto probing.   But it is possible to do this without auto probing.   If you want, you can do this without a Z-Probe.  You would just print a Mesh Validation Pattern and start editing it.     It will take an extra iteration or two because you are starting with no extra information gathered by the auto probing....   But that does work.

I think what you will find is after printing 2 or 3 mesh validation patterns and editing the mesh points....  The process converges very quickly.      As long as you don't drop your printer off of the table or mess with end stop positions, or switch pieces of glass....   Things stay very stable for many months.

I don't even have a G29 in my Slic3r's start up code.   I home and print.....

WheresWaldo, 2017-03-29:
Just give some thought to the way I outlined it, think of it as UBL auto and UBL manual had a baby and the baby is the way to adjust. I think it would be more intuitive for an enduser than editing and reprinting. I think that users of auto-generation would think that there would be a way to auto-adjust what they have done. Just a thought. 

I have used MESH since Ed PR'ed it into the mainline code and I also don't have a G29 command in my startup code.  

In the meantime I am going to try and outline the steps I need to do to get UBL running on a Robo R1. Configuration is pretty easy, it will be the actual implementation and adjustment that will require documentation. I can almost guarantee there will be a few of us putting it through it's paces.

Roxy-3D, 2017-03-29:
>  I can almost guarantee there will be a few of us putting it through it's paces.

Great!    The higher level process I envisioned is:

-  Auto Probe to generate a mesh that is close.
- G26 to generate a mesh validation pattern.    If the pattern prints nicely every where, you are done!
- If not, you move the nozzle to the problem areas and do some small edits of the mesh points.
- go back to step 2 and print the G26 Mesh Validation Pattern.   If the pattern prints nicely ever where, you are done....   Other wise...  a little more editing.   The areas you need to edit should shrink in both size and amount.

WheresWaldo, 2017-03-29:
I have been maintaining configuration files for 1.1.0 since RC6 for the Robo community. There is no support from Robo3D. There are currently quite a few moved to RC8 and a few of us brave souls using RCBF. I generally provide files twice a month, beginning and middle. I will gather a new source zip April 1 or 2 for the first set of April files. It is one of the largest threads on the Robo forums.

Roxy-3D, 2017-03-29:
I'll try to give you good support and a good experience bringing up UBL.    I think you will find a 10 x 10 Mesh that is edited to your piece of glass really does a good job!    And once you have gone through the mesh editing process once....    It is very easy and quick to bring up a new printer or a new piece of glass.

I don't dread switching printers or putting a new piece of glass on my printer.  Or for that matter, switching mesh dimensions.    It takes 30 minutes to re-probe.   But I don't even have to watch it do that.   And then 10 or 15 minutes to print a new mesh validation pattern...   But I don't even have to watch that.   I edit what ever has to be tweaked...   And I never touch it again.

WheresWaldo, 2017-04-03:
I'm closing this and will retest when UBL is more mature.

closed 2017-04-03 by WheresWaldo