MarlinFirmware/Marlin #6140: Bug using old configuration file.

batata004, 2017-03-28:
Hi my friends,

I have this configuration file that is used in old versions of marlin. I would like to migrate it to the new version/format of marlin firmware. Does any of you have any idea of some automated script that can do that or I would have to go line by line checking what changed from old version of marlin to the new and change?

Take a look at the file attached which belongs to my printer manufacturar that refuses do provide an up to date configuration file of the new marling firmware.

For example, in the old file I have this:

MOTHERBOARD 33

And I know in the new file of the newest marlin firmware I should use this:

define MOTHERBOARD BOARD_RAMPS_14_EFB

Unfortunately I am not too smart to find all the changes necessary.

So I ask you: is there any tool that allows me to migrate some old marlin configuratino file (provided by the manufacter of my printer) to the new format?

I really want to use the new version of marlin mostly beucase it now allows me to use the latest ARDUINO IDE software instead of having to use the old 1.0.6! I use a lot arduino and it's terrible to have this old software installed on my computer only because of this old firmware.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/874412/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/874411/Configuration_adv.txt)

Roxy-3D, 2017-03-28:
@batata004   You should load up NotePad++ and turn on the comparison plug in.   It will let you map everything over fairly painlessly.    

I don't feel like doing this right now.   But check back tomorrow afternoon.    I might just do the first pass for you.

batata004, 2017-03-28:
Wow man thanks a loot! I didnt know it was simple like that, only comparing the old and new file. Anyway I think some configurations may have changed a lot or even merged and I think I am not too smart into this 3D printing stuff to know how to solve this problems of configurations merged.

As a start point, if you could to the first pass I appreciate a lot, this will certainly point me in the right direction so I can tweak the things from there!

Could you just answer one more question, please? Is it really true that the new version of marlin supports the newest Arduino IDE?

Thanks a LOT! I really thank you so much!

Tannoo, 2017-03-28:
Yes, it does.

batata004, 2017-03-28:
Great that's great news! Unfortunatelly with this old config file I am stuck with the version 1.06 cause this manufacturer wont update it for me.

One thing that amazes me is that the new configuration file of marlin is twice the size of the old configuration file! So I think I could never migrate this file to the new version without any proper help.

Tannoo, 2017-03-28:
Get the important basic machine settings set first (in `configuration.h`): 
- Motherboard type
- Temp sensor types
- Steps per mm
- Accdeleration settings
- Endstop settings
- Travel limits

All else are options.

batata004, 2017-03-28:
@Tannoo I appreciate your kindness of listing the most impotant configuration, however I am a complete complete beginner with 3d printing, I have no idea which settings belong to each item in your list and have no idea how they changed in the new version.

For now I will stick with @Roxy-3D help in order to get a decent base file to start print. If I have some problem I can tweak it till I get what is wrong!

Tannoo, 2017-03-28:
Leave these disabled until you have your probe points set correctly.
```
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING
```

Tannoo, 2017-03-28:
I saw there is a new version of the IDE (1.8.2). Trying that now.

batata004, 2017-03-28:
Thanks for your feedback, good to know the new IDE works fine with the most recent marlin! :)

Tannoo, 2017-03-28:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/876088/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/876089/Configuration_adv.txt)

Please look them over (at least the important settings) before you try them.

denmorru, 2017-03-28:
I support topic starter idea: the upgrade procedure needs to be developed some way, 
with script executed over settings file or similar automated procedure. 
This is not convenient to use text compare and contrast method on each upgrade...

Tannoo, 2017-03-28:
> I have no idea which settings belong to each item in your list and have no idea how they changed in the new version.

- Motherboard type
`  #define MOTHERBOARD BOARD_RAMPS_14_EFB`
- Temp sensor types
```
//===========================================================================
//============================= Thermal Settings ============================
//===========================================================================

```
- Steps per mm
- Acceleration settings
```
//=============================================================================
//============================== Movement Settings ============================
//=============================================================================
```
- Endstop settings
```
//===========================================================================
//============================== Endstop Settings ===========================
//===========================================================================
```
- Travel limits
```
// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 205
#define Y_MAX_POS 205
#define Z_MAX_POS 200
```

Roxy-3D, 2017-03-28:
> This is not convenient to use text compare and contrast method on each upgrade...

We are almost done with a major code and feature clean up.   It won't be as difficult to upgrade as this transition is.   And really...   This transition is not that difficult.

Tannoo, 2017-03-28:
>I support topic starter idea: the upgrade procedure needs to be developed some way,
with script executed over settings file or similar automated procedure.
This is not convenient to use text compare and contrast method on each upgrade...

This has been brought up multiple times.
The upgrade from that old of a version is the only headache.
Subsquent upgrades are not nearly as much of a pain as the config files are not being heavily shuffled anymore.

batata004, 2017-03-28:
Thank you so much! I will give it a try to the file you kindly generated! Thaaaaanks a lot!

I agree with you guys that this upgrade should not be difficult for smart people like all of you who already digged deep into this subject and know all the inner workings! Without people like you this community could never grow. I hope some day I learn a lot about this 3D print stuff and be able to retribute your attention!

ANYWAY I THINK an upgrade script is a must! Everyday more and more beginners are diving in this "3D printing world" and I bet you many of those people will get stuck with old firmware/softwares just because their manufacturers wont release updates.

IMHO you should develop some script that parses the source file and migrate all the settings to the latest firmware available. I know this takes time and I assume this is not that easy, but it would certainly help people to use the latest firmware available.

emartinez167, 2017-03-28:
Perhaps you could try using this migration tool:

https://github.com/cabbagecreek/Marlin3DprinterTool

Regards,

Ernesto.

Tannoo, 2017-03-29:
@batata004, if you got it going, please close this issue.

batata004, 2017-03-29:
@Tannoo I will give it a try when it's friday cause I am traveling today and I will just get home by friday. Anyway I was hoping @Roxy-3D  could provide his file version so I could compare with yours and see it they are similar.

Roxy-3D, 2017-03-29:
I have full confidence Tannoo's version is as good as mine would have been.   Please give it a try.

Tannoo, 2017-03-29:
> I was hoping @Roxy-3D could provide his file

FYI: _her_

batata004, 2017-03-29:
Great guys, I will gite it a try when I get hands on my printer! Thanks a lot and sorry for assuming you were a man, big thanks to all of you! I will close this for now, if some problem arises I open this again.

Bye!

batata004, 2017-04-04:
IT WORKED GUYS!!!!!!!!!!! THANKS A LOT, THE 2 FILES YOU ATTACHED ME WORKED PERFECTLY!!!

batata004, 2017-04-04:
Actually it worked at first but after I got some problems mainly with the bed leveling. I see there is no AUTO_BED_LEVELING_GRID option anymore. How did you specifify the grid? I think this may be a problem I am facing.

thinkyhead, 2017-04-04:
> How did you specifify the grid?

For `AUTO_BED_LEVELING_LINEAR` and `AUTO_BED_LEVELING_BILINEAR` you should set the following options according to your own dimensions:
```cpp
  #define ABL_GRID_MAX_POINTS_X 3
  #define ABL_GRID_MAX_POINTS_Y ABL_GRID_MAX_POINTS_X

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 15
  #define RIGHT_PROBE_BED_POSITION 170
  #define FRONT_PROBE_BED_POSITION 20
  #define BACK_PROBE_BED_POSITION 170
```

batata004, 2017-04-04:
GREAT MAN! Very easy to understand, so the reason why my printer is not self balacing before printing (I already added the extra gcode before print) is maybe beucase the files that @Tannoo gently provided does not activate that while the files that I provided (at the beginning of this thread) account for that. I see there are these lines in the files provided by @Tannoo:

> //#define AUTO_BED_LEVELING_3POINT
> //#define AUTO_BED_LEVELING_LINEAR
> //#define AUTO_BED_LEVELING_BILINEAR
> //#define AUTO_BED_LEVELING_UBL
> //#define MESH_BED_LEVELING

I have no idea which one should I uncomment PLUS I also dont know if any other change has to be done in the code in order to enable BAL exactly as in the attached files I provided. Could any of you help me with this cause I have no idea what shoudl I do.

OBS: you see that? It's not that easy to use the same configuration files accross diferent versions of marlin, definetelly should be a way to do this easier and error proof.

Roxy-3D, 2017-04-04:
> It's not that easy to use the same configuration files across different versions of marlin.

This is stabilizing fast now.   Most of the changes that are going to happen have been made.   It won't hurt to get caught up.  Because any new changes to options will just be that much easier to spot in a final catch up to the Golden Master.

batata004, 2017-04-04:
@Roxy-3D yeap I tottaly agree but for ignorant people like me, that is still trying to get in this hobby, this usage of marlin should be facilitated in order to try to convince the user to user marlin instead of so many other options out there :)

thinkyhead, 2017-04-07:
Someone made a tool to convert older Marlin configurations to the latest version, but at the moment I can't find the issue topic where the link is posted.

emartinez167, 2017-04-07:
https://github.com/cabbagecreek/Marlin3DprinterTool/tree/master/Marlin3DprinterTool

Regards/Saludos,

Ernesto.

batata004, 2017-04-07:
I will take  a look at it thanks!!!

batata004, 2017-04-08:
My friends I am trying to upload the code provided by @ghost but I am getting this error message:

exit status 1
#error "TEMP_SENSOR_1 shouldn't be set with only 1 extruder."

How can I solve this?

Bob-the-Kuhn, 2017-04-08:
In Configuration.h find TEMP_SENSOR_1 and set it to zero.  Zero tells Marlin that it's not used.

Make sure that TEMP_SENSOR_0 is set to the thermistor you're using on the extruder.

batata004, 2017-04-08:
Thanks my friend! Do you have any idea what termistor am I using based on the configuration files I attached in the top of this thread? Cause I thought @ghost had taken in account that when he kindly generated the 2 files to me but because there was this error that you pointed out in the @ghost file I think may be others.

Bob-the-Kuhn, 2017-04-08:
It's set to 1 which is `1 is 100k thermistor - best choice for EPCOS 100k (4.7k pullup)`

That's a very commonly used thermistor.

If the temperature reading you're getting at room temperature is reasonable then it's set correctly.

batata004, 2017-04-08:
Thanks my friends now it compiled correctly! I will install it right away! YOU ARE AWESOME!

batata004, 2017-04-08:
My friends, I uploaded the code and my LCD stopped working :( Any idea how to fix this?

batata004, 2017-04-08:
I also would like to point a bug: I setted the temperature of my hotend to 235 C degrees and when it was close to 150 C degrees I clicked the EXTRUDE button in the PRONTERFACE and it started spinning the EXTRUDER MOTOR! 

THIS WAS NOT SUPPOSED TO HAPPEN! I know that in the old version of marlin that this printer was using, if I tried to EXTRUDE while the hotend temperature was not at the target, the extruder would never turn. This was a safety measure but it looks like this has been disabled in the new version? Am I right?

batata004, 2017-04-08:
One more problem my dear friends: with the new configuration.h and configuration_adv.h provided by @ghost I cannot use this command below anymore:

M280 P0 S30

With the old firmware I could make the servo (that I use to level my heat bed) turn with the command above, but now I cant anymore. Could you please check what is going on?

batata004, 2017-04-08:
I dont want to bother you all, I took a long read at the file and I finally made the SERVO work! I uncommented some lines and changed the index of the servo to 0 so it worked perfectly!

I just want to know from you about the safety problem that I pointed earlier and to know how to fix the LCD?

Bob-the-Kuhn, 2017-04-08:
@ghost did a good job creating the new config files.  He was a bit conservative and didn't enable a few items that were in the old/original files.

Both the old & new configs had the minimum extrude temperature set to 170C.  The host has a choice of two commands to send to the controller when it sets the extruder temperature.  One won't allow any extrudes until the target temperature is hit and the other will.

Are you running X, Y or Z MAX endstops?  The new one has them disabled.

Bed leveling has changed a lot since the original was made.  Right now none are enabled.  Eventually you'll want to enable one of them.

The new one didn't have an LCD enabled.

The new one didn't have a Z servo probe enabled.   

The old one had servo angles of 105 & 10 for the extend and retract angles so I'm surprised that S30 worked for you.

Here's the link for an updated [Configuration.h ](https://www.dropbox.com/s/76np4g3gxsj7jgk/Configuration.h?dl=0).  I've turned on the LCD, enabled safe_homing and enabled the Z servo probe.  Better check that the servo settings match what you got to work.  If you want to run the MAX endstops then enable USE_XMAX_PLUG, USE_YMAX_PLUG & USE_ZMAX_PLUG.

SD card support isn't enabled.  It's a lot safer to print from the SD card than it is to print over the USB cable.

I didn't see anything that needs to be changed in Configuration_adv.h

thinkyhead, 2017-04-08:
> Both the old & new configs had the minimum extrude temperature set to 170C.

Both of these settings are needed. See your `Configuration_adv.h` file.
```cpp
#define PREVENT_COLD_EXTRUSION
#define EXTRUDE_MINTEMP 170
```
Note that this protection can be disabled from GCode with `M302 S0` or `M302 P1`.

Bob-the-Kuhn, 2017-04-08:
Yes, I left out an important item.

By default PREVENT_COLD_EXTRUSION is enabled.  That means Marlin won't allow an extrusion unless the extruder temp is above EXTRUDE_MINTEMP.

Pronterface probably sent the "set extruder temp to 230C but don't wait" command.  Which means you can do an extrude anytime after the temperature gets above 170C.

The slicer will use the "set temperature and wait" command.

batata004, 2017-04-09:
@Bob-the-Kuhn  I tried your code and after fixing some erros the compilator showed me I got an error that I cant solve by my own. This is the error:

```

sketch\ultralcd.cpp: In function 'void _lcd_move_xyz(const char*, AxisEnum)':

ultralcd.cpp:1414: error: 'min_software_endstops' was not declared in this scope

       float min = (soft_endstops_enabled && min_software_endstops) ? soft_endstop_min[axis] : current_position[axis] - 1000,

                                             ^

In file included from sketch\MarlinConfig.h:27:0,

                 from sketch\Marlin.h:36,

                 from sketch\ultralcd.h:26,

                 from sketch\ultralcd.cpp:23:

ultralcd.cpp:1431: error: 'max' was not declared in this scope

       if ((int32_t)encoderPosition > 0) NOMORE(current_position[axis], max);

                                                                        ^

sketch\macros.h:66:33: note: in definition of macro 'NOMORE'

 #define NOMORE(v,n) do{ if (v > n) v = n; }while(0)

                                 ^

exit status 1
'min_software_endstops' was not declared in this scope

```

batata004, 2017-04-09:
@thinkyhead  you were right, the safety feature was enabled and afer 170 celsius it allows extrude, sorry man for this misunderstanding, you are completely right :)

I am still facing the problem of compilin the code provided by @Bob-the-Kuhn if someone could help I appreciate!

Bob-the-Kuhn, 2017-04-09:
Looks like you have a configuration.h file that's newer than the software you downloaded.

About the 18th of March **min_software_endstops** changed to **MIN_SOFTWARE_ENDSTOPS**.

The quickest solution is to replace the new with the old.  It's not guaranteed that another error won't pop up.
DELETE
```
#define MIN_SOFTWARE_ENDSTOPS
#define MAX_SOFTWARE_ENDSTOPS
```

ADD
```
#define min_software_endstops true 
#define max_software_endstops true
```

The best solution is for you to download the latest software - [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix). There'll have to be a check of your current config files against the new.  If you drop them on your reply I'll do that in the morning.

Another solution would be to identify your flavor of software and then check the new vs. old config files.  I expect that you downloaded RC8.  In the **Version.h** file you'll probably find **#define SHORT_BUILD_VERSION "1.1.0-RC8"**

batata004, 2017-04-09:
Thanks a log @Bob-the-Kuhn but I am surelly not as smart as you think :( When you said I am using a new configuration file but an old software, you mean I am using an old Marlin firmware?

I downloaded the Marlin firmware from the GITHUB from here -> https://github.com/MarlinFirmware/Marlin

I took the configuration file that @ghost provided and worked from there. Maybe @ghost didnt use the newest Marlin firmware when he tried to migrate the confi file for me?

Anyway I appreciate all your kindness and my latest files that I have are those attached. When I try to uncomment the line 

//#define REPRAP_DISCOUNT_SMART_CONTROLLER

I got the error.

batata004, 2017-04-09:
@Bob-the-Kuhn YOU ARE A MASTER!!!! YOU ARE COMPLETELLY RIGHT! I dont know how you discovered that but you are 100% correct: the file that @ghost kindly provided me WAS NOT the configuration files of the latest MARLIN at -> https://github.com/MarlinFirmware/Marlin

Instead, he provided me configuration files for another Marlin "flavor" and that caused many problems that I was having! 

So I did this: I opened an online file comparator tool and changed everything that @ghost had set to me in his configs files and moved it to the new configuration.h and configuration_adv.h! It looks like I am not getting any more error messages!!! :) :) :)

Now I am gonna try this file in my printer when I get home and check if it works.

I appreciate all your attention guys, very nice community this one we have here.

batata004, 2017-04-09:
IT ALL WORKED! Finally after all your attention I could make some progress. Now I am gonna try to understand better some other things to dive in this whole new world of 3d printing!

Thanks to all specially to @Bob-the-Kuhn  :)

closed 2017-04-09 by batata004