MarlinFirmware/Marlin #6143: Feature request: Filament change menu option while idle

mattminuti, 2017-03-28:
There's one feature Sailfish has that I've long wished for in Marlin: a menu option for changing the filament. Not during a print, but while idle, and including preheating if necessary. It's particularly nice for new users and those who tend to use a machine more like an appliance than a hacker would.

For reference, the way Sailfish does it is preheat, make some beeping to alert the user it's ready, reverse the filament (guessing somewhere between 10-25mm/s), keep reversing until button press, then feeds new filament in until button press, then return to main status screen. There's a 5-minute timeout in there, too.

I tried adding it myself, but I got a bit lost with scoping issues when trying to mimic the behavior of lcd_filament_change_show_message().

Roxy-3D, 2017-03-28:
The filament change option is already there for you.   But it is not presented as an option unless the nozzle is at temperature.    If you are willing to turn the heat on for your nozzle and let it get hot...   Go look in the Prepare menu system.  It will be right after Disable Stepper Motors.

BUT...   You have to enable it in Configuration_adv.h or it won't show up.

mattminuti, 2017-03-28:
Ahh, I was looking at Marlin 1.1.0-RC8, not the RCBugFix branch. In RC, it never shows up in Prepare. Thanks! :+1: 

It'd still be nice to have filament change option available when cold, but have it preheat before trying anything. One less thing to dig through menus for.

closed 2017-03-28 by mattminuti

thinkyhead, 2017-04-02:
> but have it preheat before trying anything

That will probably be done, as we already have it cool down if you leave it idle for too long, then it waits to heat up again when you go to continue, with a helpful display of the temp.