MarlinFirmware/Marlin #6144: Fan and LED Pin Designation - help configuring h files

dvseliteus, 2017-03-28:
I am running Marlin 1.1.0-RCBugFix - 6 Dec 2016 with an Azteeg X3 [v1.1](http://files.panucatt.com/datasheets/x3v1_1_wiring_diagram.pdf) board.
I am having a hard time figuring out how to correctly configure the pins for the fans and leds.
Here are the pins I would like to configure;
HotEnd Fan = 4
Part Cooling Fan = 5
LED Case Light = 11

I am currently starting pins 5 and 11 through a startup gcode and pin 4 is just always on.
I get lost when going through the pin h files, can anyone help me learn how to correctly designate these pins?

Bob-the-Kuhn, 2017-03-29:
8, 9 & 10 are your power FETs.  They should be connected to extruder heaters, hotbed heater and high powered fans.

4, 5, 6 & 11 are lower power FETs.  They are meant for standard fans and LEDs.

The first step is to figure out what signals are currently tied to the pins you want to use.  In **Configuration_adv.h** enable **PINS_DEBUGGING**.  After that issue the command **M43 I1**.  That'll pop up a list of the pins and what signals are tied to them.

The next step is to modify your pin definitions in the **pins_AZTEEG_X3.h** file  In there you'll find **#include "pins_RAMPS_13.h"**.  After that line you can do your customization.

To disconnect the unwanted signals from the pins add the following lines:
```
#undef  name_of_unwanted_signal_1
#undef  name_of_unwanted_signal_2
#undef  name_of_unwanted_signal_3
```

To connect your fans add these lines:
```
#define E0_AUTO_FAN_PIN    4  //extruder 0 cooling fan
#define FAN_PIN            5  //part cooling fan
```

To connect your case light go to Configuration.h and find **//#define CASE_LIGHT_PIN 4**.  Change it to **#define CASE_LIGHT_PIN 11**.

dvseliteus, 2017-03-29:
@Bob-the-Kuhn 
Thanks for that information which was eye opening!
Found the //#define CASE_LIGHT_PIN 4 in the Configuration_adv.h file.
Everything is performing as needed now.
Thank you!

closed 2017-03-29 by dvseliteus