MarlinFirmware/Marlin #6146: Using Z probe as a limit switch as well?

gcormier, 2017-03-28:
Venturing out of uncharted waters....

I have a nozzle Z probe working quite well. Is it possible to have it also function as a limit switch, so if it's ever triggered (should never happen, other than probing operations), it would stop printing?

(I've been seeing all the sweet UBL merges, and I want to try it out. My plate is mounted on springs, so I want to change that to a rigid mounting so the nozzle probe doesn't have inconsistent results)

Roxy-3D, 2017-03-28:
> (I've been seeing all the sweet UBL merges, and I want to try it out. My plate is mounted on springs, so I want to change that to a rigid mounting so the nozzle probe doesn't have inconsistent results)

It should be OK to stay on springs.   I typically home (and probe) once in the center of the bed.   And if the height isn't exactly right as it is printing the skirt, a double click drops you into Z-BabyStepping where you can quickly get things perfect.

If you have a Z-Probe...   You shouldn't need to make any hardware changes to bring up UBL.

gcormier, 2017-03-28:
The issue is that the edges of the beds seem to have more give then the center of the bed, so I'm worried it will skew my resulting grid that Marlin uses. I could tighten the heck out of them to make them even stiffer I suppose?

Roxy-3D, 2017-03-28:
No...   I wouldn't do that.    I guess it depends on your Z-Probe.   But most of them are very gentle (unless you drive the nozzle into the bed).   I doubt you will see much deflection based on the probe pressing down on the bed.

But here is the thing...   It just doesn't matter.    The probe is used to help automate the mesh generation process.   Even if the edges of the mesh are off because the bed was deflecting a small amount...   It won't hurt you.   When you do the G26 P C O3.5  to validate the mesh, you will see that the edges need improvement.   And with the Mesh Validation Pattern still on the print bed, you can edit those mesh points up or down.

My startup GCode for the Slic3r doesn't even have a G29 in it.   The system loads the mesh and its active.   I home in the center of the bed and everything is alive and well.   Up until a week or so ago...  All of my machines had spring mounted beds.    I need that because its easy to make a firmware mistake and crash the nozzle into the bed.   But now I'm on a machine with solid mount for the print bed...  So no more nozzle crashes are allowed!

gcormier, 2017-03-28:
Hmmm. Okay, I will have to check it out :)

Where is the best place to read up on UBL? Is there any Github issue that has steps to follow? Or somewhere documented? :) Dare I ask. Do you think it's ready for testing/playing with yet or should I give it a few more weeks?

Roxy-3D, 2017-03-28:
There are bullet point lists in some of the UBL threads telling you the steps...   If you follow those lists, you will get a perfect mesh very quickly and will see very good results.    One thing that is important is get your Z_PROBE_OFFSET_FROM_EXTRUDER set very accurately.   That affects everything down stream in the process.    Make sure you can home (preferably in the center of the bed) and that you can print a calibration cube with no bed leveling active.   If you can do that...   Then just follow the bullet point list and you should get good adhesion across your entire bed.

Bob-the-Kuhn, 2017-03-29:
On one of the original questions ...

If you enable **Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN** (and disable **Z_MIN_PROBE_ENDSTOP**) and enable **ENDSTOPS_ALWAYS_ON_DEFAULT** (in Configuration_adv.h) then your nozzle/probe will act like an always on endstop.

You may have to re-define your Z_MIN_PIN to the one used by the probe.

gcormier, 2017-03-29:
@Bob-the-Kuhn Awesome! So I can just have the same pin doing both functions? And it will ignore that endstop for probing operations?

Bob-the-Kuhn, 2017-03-29:
It'll use the Z_MIN_PIN for both probing and homing.

closed 2018-02-06 by thinkyhead

derdigge, 2018-05-10:
@Bob-the-Kuhn 

Sorry for highjacking the ticket. I do not have found the option ENDSTOPS_ALWAYS_ON_DEFAULT in my tree. had it changed? I plan to use an microswitch for autoleveling. Both will be connected to z endstop pin, which is defined as 19 in my case.

thinkyhead, 2018-05-10:
[image]