MarlinFirmware/Marlin #6147: M84 ignores S directive, unsafe defaults

jwcrawley, 2017-03-28:
Problem: All motors are disabled after 120 seconds. In my case, a CoreXY, I have a 20 lb bed and the bed slams to the bottom. Previous stable releases did not to this. (Problem 1).

I notice there is a gcode instruction "M84 S##" where ## is the timeout in seconds when to turn the motors off. When I power on the printer, and immediately initiate a "M84 S0", which is noted to *turn off* the timeout time, the motors time out at 120 seconds, and the bed falls. Timeout changes with M84 S# are ignored (Problem 2)

__________________________

Marlin FW: I'm using RCBugFix from March 26 2017.

__________________________

Printer Type: Vulcanus 40, CoreXY. Bed homes to ZMax which is 400mm up. 

__________________________

Proposed Solution for Problem 1: 
File configuration_adv.h  change "true" to "false" for said lines:
360: #define DISABLE_INACTIVE_X false
361: #define DISABLE_INACTIVE_Y false
362: #define DISABLE_INACTIVE_Z false  // set to false if the nozzle will fall down on your printed part when print has finished.
363: #define DISABLE_INACTIVE_E false

Justification: Marlin 1.0.2-2 Has these vars as well and they are FALSE, thus keeping the motors on as expected. This is the known and right thing to do, given the stable branch's choices. M18/M84 is expected to disable steppers, and is an expected command to initiate when that is the desired result.

__________________________

Proposed solution for Problem 2:
File marlin_main.cpp remove STATIC keyword
460: millis_t stepper_inactive_time = (DEFAULT_STEPPER_DEACTIVE_TIME) * 1000UL;

Justification: C++ Static keyword. When included modifying a variable, and as global scope, the variable can no longer be referenced other than by the initial declaration. In the case of the default Marlin RCBugFix, the number is 120 seconds. Giving any M84 S## defaults to 120 seconds as per static keyword.

I also noticed this behavior as well in Marlin 1.0.2-2, with a similar STATIC declaration. However, since motors do not disable by default in Stable, did not notice it earlier.

jwcrawley, 2017-03-28:
FWIW, the proposed edits I have requested a PR for, are the edits that allow my Vulcanus to work properly. I made these changes at MRRF (Midwest RepRap Festival).

teemuatlut, 2017-03-28:
I tried this with my X axis and after M84 S10 the carriage could be moved after 10 seconds since a move had happened. Sending M84 S0 made the carriage stay still for as long as I could wait. This seems to work as intended. Is there something I'm missing?

jwcrawley, 2017-03-28:
Indeed.

I downloaded RCBugFix on March 24 while at MRRF. I configured it appropriately for a CoreXY and a mixing extruder (Diamond). The motion is CoreXY. The XY motion is fixed, and Z is the bed on 2 leadscrews. ZMin is the bottom, and ZMax is at the top, with the tip touching the bed in the usual manner. 

When I loaded the firmware after configuring, I brought Z halfway up as I manually moved the endstops to the appropriate position. "A bit later", the Z motors disengaged, with no notices or buttons pressed in Pronterface. I checked Pronterface, and saw no settings that would inject a M84. 

I then proceeded to power off everything, power back on and  use the LCD screen/dial to home Z. After 2 minutes (120 seconds) the bed fell, slamming into the base of the Vulcanus. 

I then found the relevant GCode to modify this, M84 (disables), M84 S## (sets motor timeout to ## seconds). I proceeded to set M84 S0 , which the documents state disables timeouts. After 120 seconds, the motors are cut and the bed falls as I described. 

By making the 2 edits in my PR, I have fixed the issue on my hardware.

I am running an Arduino Mega/RAMPS 1.4

teemuatlut, 2017-03-28:
Could you please attach your configuration.h and configuration_adv.h files here (rename to .txt) or tell me which settings to enable at minimum? I'd like to test this out.

Dreamloom, 2017-03-29:
In Configuration_adv.h, around line 359:
I set this to zero,
#define DEFAULT_STEPPER_DEACTIVE_TIME 0 

and this to false,
#define DISABLE_INACTIVE_Z false

jwcrawley, 2017-03-29:
[pins_RAMPS_13.txt](https://github.com/MarlinFirmware/Marlin/files/877759/pins_RAMPS_13.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/877758/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/877760/Configuration_adv.txt)

Files were .h renamed to .txt because of idiotic Github naming postfix conventions (yeah, never heard of file).

In all honesty, I would prefer the machine to be stupid and follow basic commands. If I give it an M84, I expect the motors to disengage. I also expect the motors wontt just disengage on their own due to bad settings.

teemuatlut, 2017-03-29:
Tried it with your configuration files. I had to change a few things but nothing major.

I reverted the DEFAULT_STEPPER_DEACTIVE_TIME and DISABLE_INACTIVE_Z back to upstream (120 and true), and the Z axis was free after 120s. I then sent M84 S0, moved the Z axis up a bit and I've now waited 10 minutes and the axis is still locked. Seems to be working as intended.

closed 2018-02-06 by thinkyhead