MarlinFirmware/Marlin #6151: Enabling servo support disables fan2 PWM on K8400/3DualDrag + Servo pin defs are wrong

SwiftNick, 2017-03-28:
K8400 pin definitions inherit from pins_Ramps.h which has conflicting pin definitions.

To resolve this add the following to pins_K8400.h
```C++
#undef HEATER_2_PIN     
#undef FAN1_PIN
#define FAN1_PIN         2  // Onboard MOSFET FAN2		

// Servos
#undef SERVO0_PIN 
#define SERVO0_PIN       7      // PH4 on expansion header

#undef SERVO1_PIN          
#define SERVO1_PIN       6    // PH3 on expansion header

#undef SERVO2_PIN          
#define SERVO2_PIN       -1 

#undef SERVO3_PIN          
#define SERVO3_PIN       -1
```

The K8400 board can only support 2 servos due to a lack of connected PWM capable outputs.

What I can't resolve is a PWM timer conflict on pin2 when enabling servo support.

M43 gives:
```
RECEIVED: PIN:  2        FAN1_PIN                    protected     TIMER3B    WGM: 0    TIMSK3: 2   Can't be used as a PWM because of counter mode**  
```

with servo support disabled the PWM works just fine.
```
RECEIVED: PIN:  2        FAN1_PIN                    protected     TIMER3B    WGM: 1    TIMSK3: 0   can be used as PWM     

```

Can the a different timer be assigned in the servo code for the K8400 board?
[K8400 Pin Debug.txt](https://github.com/MarlinFirmware/Marlin/files/877297/K8400.Pin.Debug.txt)

[image]

Bob-the-Kuhn, 2017-03-29:
My notes indicate that as soon as you turn on servo support the timer3 hardware PWMs are disabled.

I think you have **FAST_PWM_FAN** enabled.  That mode uses the hardware PWMs.  Disable that and enable **FAN_SOFT_PWM**.  That'll allow your fans to be functional along with having functional servos.

SwiftNick, 2017-03-29:
Thanks Bob that works a treat!

I hadn't enabled FAST_PWM_FAN, but FAN_SOFT_PWM wasn't enabled either.
Enabling FAN_SOFT_PWM and tweaking SOFT_PWM_SCALE gave me what I needed.
(I'm using the extra fan port to drive LED cobs to light the bed. default SOFT_PWM_SCALE was like a disco!)

Bob-the-Kuhn, 2017-03-29:
Old enough to remember disco?  That dates you!   ;-)

closed 2017-10-31 by thinkyhead