MarlinFirmware/Marlin #6153: How to save M218 in EEPROM?

karabas, 2017-03-29:
I use M218 in my start Gcode section to set offsets for my dual prusa like printer.
I think it's will be usefull to save it in EEPROM

Roxy-3D, 2017-03-29:
I agree this makes sense to store in the EEPROM.   But when I look at Configuration_store.cpp I see this code:
```cpp
    #if HOTENDS > 1
      // Skip hotend 0 which must be 0
      for (uint8_t e = 1; e < HOTENDS; e++)
        LOOP_XYZ(i) EEPROM_WRITE(hotend_offset[i][e]);
    #endif
```

Bob-the-Kuhn, 2017-03-29:
Later on in M501 is
```
      #if HOTENDS > 1
        // Skip hotend 0 which must be 0
        for (uint8_t e = 1; e < HOTENDS; e++)
          LOOP_XYZ(i) EEPROM_READ(hotend_offset[i][e]);
      #endif
```

M500 stores and M501 restores the M218 values.

Do a M503.  The M218 settings are printed out there.

You can also put them into your configuration.h file.   M502 followed by M500 will save them to EEPROM
```
#define HOTEND_OFFSET_X {0.0, 20.00} // (in mm) for each extruder, offset of the hotend on the X axis
#define HOTEND_OFFSET_Y {0.0, 5.00}  // (in mm) for each extruder, offset of the hotend on the Y axis
```

karabas, 2017-03-31:
New feature! I used RC7. Thank you  I will jump to bugfix branch

closed 2017-03-31 by thinkyhead