MarlinFirmware/Marlin #6155: Probe stops working as  ensdstop as soon as

emski0, 2017-03-29:
Hi
I am using Z-min probe, the probe works as endstop if no probe for autolevel is selected.
As soon as i define : 
#define FIX_MOUNTED_PROBE
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

The probe stops working as endstop and crash into the bed. However the Z-min pin shows triggered.
when i comment the #define FIX_MOUNTED_PROBE , it works again as endstop, but then there is no autolevel.
I tried playing with ENDSTOPPULLUPS ,but nothing changed

Roxy-3D, 2017-03-29:
Do you have a Fixed Probe?    If not...   Enabling that option is going to cause bad things to happen.   What kind of probe do you have?

emski0, 2017-03-29:
Yeh i have fixed probe, an inductive sensor

thinkyhead, 2018-02-06:
Presuming solved.

closed 2018-02-06 by thinkyhead