MarlinFirmware/Marlin #6156: Set nozzle temperature on graphical display menu item not working on RC8

JanVelleman, 2017-03-29:
Hi,

I'm working on a derivative board of the K8800 with a graphical display (based on ST7565 chip). Display is working as it should (albeit a bit slow on refresh but I guess that's just how marlin works with graphical displays). Firmware is a full RC8 install with proper config for the prototype that the board is driving.
Printer is a delta with autoleveling enabled, 1 hotend, no heated bed.

But when I try to change the temperature of the nozzle the display doesn't show anything (while printing or when idle). Then I have to wait until the blank menu times out back to the info screen to gain control again. I can however change the fan, contrast and so on... So the menu item to change a value seems to be working, just not for temperature of the nozzle...

See the video below for an illustration of the problem:

https://1drv.ms/v/s!AkB62gmPn-IZgpNQf_Y2zZqkG-83dA 

Any thoughts??

Bob-the-Kuhn, 2017-03-29:
Please update to the latest [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).  The LCD software has been improved a lot since RC8.

There's been a fair number of changes to the config files so you can't just drop your current ones into the new.

JanVelleman, 2017-03-29:
Hi, Thx for your answer

Is there a way to now for sure if this issue has been resolved as it always takes a long time to get all the configs and small tweaks I do correct between builds. I usually wait for an RC release before going through all that...

:(

StefanBruens, 2017-03-30:
@JanVelleman, if you use a proper git workflow, rebasing to the current RC is quite simple.

1. Create a local branch (e.g. my-K8800-rc8)
2. Commit your changes to Configuration.h (and Configuration_adv.h) to your local branch
  a) split you changes into semantical sets, e.g. one set for the names and boilerplate stuff, one commit for axis/endstops/dimensions, one commit for heater related stuff
  b) add a useful commit message and description, this will allow you to understand your changes later
3. Duplicate your local branch - as you are at the HEAD of your local branch, a simple "git branch my-K8800-rcBugfix" suffices
4. Rebase your branch on the current RCBugFix head - "git rebase origin/RCBugFix"

If you are lucky, git will merge all you changes with the changes done between your old branch point and the current RCBugFix. If not:
- likely some of the commits will merge without further changes, while others will fail - this is another reason for splitting your changes into independent commits, the size of failing commits will be quite small
- in case of merge conflicts, you have to do the conflict resolution yourself. There are two somewhat different approaches, either (the default) git will merge your changes into the file, and add conflict markers (triples of <<<, ===, >>>), or you use "git mergetool" whenever git reports a conflict. git mergetool is a wrapper for calling a visual diff tool, e.g. kdiff3, meld, vimdiff with the appropriate parameters.

JanVelleman, 2017-03-30:
Thx,

Switched to RCbugfix and indeed the problems are resolved. screen is also much (much!) faster except in the SDcard menu. Still slow as hell there, probably due to SD CARD reading also being SPI?

Anyway there seems to be a little bug still in the level bed menu when you did not previously auto home, some weird behaviour, but I will post that in a different issue. 

This one can be closed.

Kaibob2, 2017-04-12:
Please close this issue if it's solved.

closed 2017-10-31 by thinkyhead