MarlinFirmware/Marlin #6158: Stock Marlin FW LCD works but anything else doesn't??

Psyc0Flyer, 2017-03-29:
I am having a HELL of a time figuring this out. I have the stock Marlin build that came with my Melzi powered printer and as long as I use the included Marlin build, my RepRapDiscount FULL GRAPHIC Smart Controller will work but if I change to anything else, I just get a powered blank screen.. Can someone PLEASE help me figure this out?? I want to upload the latest version.
[MarlinSTOCK.zip](https://github.com/MarlinFirmware/Marlin/files/879977/MarlinSTOCK.zip)

Tannoo, 2017-03-29:
They do, you just have to configure it to work.
I will help you. Give me a few mins.

Psyc0Flyer, 2017-03-29:
TY! MUCH appreciated! I noticed after digging for a few, in latest, it refers to pins_SANGUINOLOLU_11.h So is THAT the file that needs edited?

Tannoo, 2017-03-29:
What board you have? The old file referred to a 63. And in the new `boards.h` file, that is:
`#define BOARD_MELZI             63   // Melzi`

Psyc0Flyer, 2017-03-29:
Correct

Tannoo, 2017-03-29:
`pins_MELZI.h` would be the new pins file for that board.

Psyc0Flyer, 2017-03-29:
but in that file, it refers to the previously mention pins file

Psyc0Flyer, 2017-03-29:
* Melzi pin assignments
 */

#define BOARD_NAME "Melzi"

#ifdef __AVR_ATmega1284P__
  #define LARGE_FLASH true
#endif

#define SANGUINOLOLU_V_1_2
#include "pins_SANGUINOLOLU_11.h"

Tannoo, 2017-03-29:
Getting your new config files ready for the newest firmware takes a little time. Things were re-arrainged quite a bit from the version you supplied.

Yeah, I saw that also just now. So, yeah. The `pins_SANGUINOLOLU_11.h` file is your pins file.

Psyc0Flyer, 2017-03-29:
That is pretty much the extent of where I am at. I am confused at how the pins are assigned in the original file and where to put them in the new version :/

Tannoo, 2017-03-29:
What are you looking for?

Tannoo, 2017-03-29:
You shouldn't have to change any pin configurations.

Tannoo, 2017-03-29:
Unless you are trying to change or add something that is not default.

Tannoo, 2017-03-29:
The screen didn't work most likely because it was not enabled in the FW.

Psyc0Flyer, 2017-03-29:
If I wouldn't have to change anything then why does the stock board not work in any other version besides stock? O_o I DO have it enabled, that is what I am saying.. LOL

Psyc0Flyer, 2017-03-29:
Somewhere the pin assignments are not correct in the new versions.

Tannoo, 2017-03-29:
Okay. I don't have that board.

Tannoo, 2017-03-29:
Maybe you can get a schematic of that board online and compare notes.

Psyc0Flyer, 2017-03-29:
That is why I was asking for someone that can read the old file and LMK what I need to change in the newer versions to make it work. Obviously the pins are proper in the stock version for it to work. I just need to know what to change.. ;)

Psyc0Flyer, 2017-03-29:
Here is the stock pins file that the display works on
`#ifndef PINS_H
#define PINS_H

#define X_MS1_PIN -1
#define X_MS2_PIN -1
#define Y_MS1_PIN -1
#define Y_MS2_PIN -1
#define Z_MS1_PIN -1
#define Z_MS2_PIN -1
#define E0_MS1_PIN -1
#define E0_MS2_PIN -1
#define E1_MS1_PIN -1
#define E1_MS2_PIN -1
#define DIGIPOTSS_PIN -1

#if MOTHERBOARD == 99
#define KNOWN_BOARD 1

#define X_STEP_PIN          2
#define X_DIR_PIN           3
#define X_ENABLE_PIN        -1
#define X_STOP_PIN          16

#define Y_STEP_PIN          5
#define Y_DIR_PIN           6
#define Y_ENABLE_PIN       -1
#define Y_STOP_PIN          67

#define Z_STEP_PIN          62
#define Z_DIR_PIN           63
#define Z_ENABLE_PIN       -1
#define Z_STOP_PIN          59

#define E0_STEP_PIN         65
#define E0_DIR_PIN          66
#define E0_ENABLE_PIN      -1

#define SDPOWER            -1
#define SDSS               53
#define LED_PIN            -1
#define FAN_PIN            -1
#define PS_ON_PIN           9
#define KILL_PIN           -1

#define HEATER_0_PIN        13
#define HEATER_1_PIN       -1
#define HEATER_2_PIN       -1
#define TEMP_0_PIN          6   // MUST USE ANALOG INPUT NUMBERING NOT DIGITAL OUTPUT NUMBERING!!!!!!!!!
#define TEMP_1_PIN         -1   // MUST USE ANALOG INPUT NUMBERING NOT DIGITAL OUTPUT NUMBERING!!!!!!!!!
#define TEMP_2_PIN         -1   // MUST USE ANALOG INPUT NUMBERING NOT DIGITAL OUTPUT NUMBERING!!!!!!!!!
#define HEATER_BED_PIN      4
#define TEMP_BED_PIN       10

#endif /* 99 */

/****************************************************************************************
* Sethi 3D_1 pin assignment - www.sethi3d.com.br
*
****************************************************************************************/

#if MOTHERBOARD == 20
#define KNOWN_BOARD

#if !defined(__AVR_ATmega644P__) && !defined(__AVR_ATmega644__) && !defined(__AVR_ATmega1284P__)
#error Oops! Make sure you have 'Sethi 3D' selected from the 'Tools -> Boards' menu.

#endif

#ifndef GEN7_VERSION
[... 2843 lines cut ...]
    #define LCD_PINS_D4         30
    #define LCD_PINS_D5         29
    #define LCD_PINS_D6         28
    #define LCD_PINS_D7         27

  #endif // !DOGLCD

  #define BTN_EN1               11
  #define BTN_EN2               10

  #if ENABLED(LCD_I2C_PANELOLU2)
    #if IS_MELZI
      #define BTN_ENC           29
      #define LCD_SDSS          30 // Panelolu2 SD card reader rather than the Melzi
    #else
      #define BTN_ENC           30
    #endif
  #else  // !Panelolu2
    #define BTN_ENC             16
    #define LCD_SDSS            28 // Smart Controller SD card reader rather than the Melzi
  #endif // !Panelolu2

  #define SD_DETECT_PIN         -1

#endif // ULTRA_LCD && NEWPANEL`

Psyc0Flyer, 2017-03-29:
# **Here is the NEW pins file that needs modified to make the display to work**
`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Sanguinololu board pin assignments
 */

#if !defined(__AVR_ATmega644P__) && !defined(__AVR_ATmega1284P__)
  #error "Oops!  Make sure you have 'Sanguino' selected from the 'Tools -> Boards' menu."
#endif

#ifndef BOARD_NAME
  #define BOARD_NAME "Sanguinololu <1.2"
#endif

#define IS_MELZI (MB(MELZI) || MB(MELZI_MAKR3D))

//
// Limit Switches
//
#define X_STOP_PIN         18
#define Y_STOP_PIN         19
#define Z_STOP_PIN         20

//
// Z Probe (when not Z_MIN_PIN)
//
#if ENABLED(Z_PROBE_SLED)
  #define SLED_PIN         -1
#endif

//
// Steppers
//
#define X_STEP_PIN         15
#define X_DIR_PIN          21

#define Y_STEP_PIN         22
#define Y_DIR_PIN          23

#define Z_STEP_PIN          3
#define Z_DIR_PIN           2

#define E0_STEP_PIN         1
#define E0_DIR_PIN          0

//
// Temperature Sensors
//
#define TEMP_0_PIN          7   // Analog Input (pin 33 extruder)
#define TEMP_BED_PIN        6   // Analog Input (pin 34 bed)

//
// Heaters / Fans
//
#define HEATER_0_PIN       13 // (extruder)

#if ENABLED(SANGUINOLOLU_V_1_2)

  #define HEATER_BED_PIN   12 // (bed)
  #define X_ENABLE_PIN     14
  #define Y_ENABLE_PIN     14
  #define Z_ENABLE_PIN     26
  #define E0_ENABLE_PIN    14

  #if ENABLED(LCD_I2C_PANELOLU2)
    #define FAN_PIN         4 // Uses Transistor1 (PWM) on Panelolu2's Sanguino Adapter Board to drive the fan
  #endif

#else

  #define HEATER_BED_PIN   14 // (bed)
  #define X_ENABLE_PIN     -1
  #define Y_ENABLE_PIN     -1
  #define Z_ENABLE_PIN     -1
  #define E0_ENABLE_PIN    -1

#endif

#if MB(AZTEEG_X1) || MB(STB_11) || IS_MELZI
  #define FAN_PIN           4 // Works for Panelolu2 too
#endif

//
// Misc. Functions
//
/**
 * On some broken versions of the Sanguino libraries the pin definitions are wrong,
 * which then needs SDSS as pin 24. But you should upgrade your Sanguino libraries! See #368.
 */
//#define SDSS               24
#define SDSS               31

#if IS_MELZI
  #define LED_PIN          27 // On some broken versions of the Sanguino libraries the pin definitions are wrong, so LED_PIN needs to be 28. But you should upgrade your Sanguino libraries! See #368.
#elif MB(STB_11)
  #define LCD_BACKLIGHT_PIN 17 // LCD backlight LED
#endif

//
// LCD / Controller
//
#if ENABLED(ULTRA_LCD) && ENABLED(NEWPANEL)

  #if ENABLED(DOGLCD)

    #if ENABLED(U8GLIB_ST7920) // SPI GLCD 12864 ST7920 ( like [www.digole.com] ) For Melzi V2.0

      #if IS_MELZI // Melzi board
        #define LCD_PINS_RS     30 // CS chip select /SS chip slave select
        #define LCD_PINS_ENABLE 29 // SID (MOSI)
        #define LCD_PINS_D4     17 // SCK (CLK) clock
        // Pin 27 is taken by LED_PIN, but Melzi LED does nothing with
        // Marlin so this can be used for BEEPER_PIN. You can use this pin
        // with M42 instead of BEEPER_PIN.
        #define BEEPER_PIN      27
      #else        // Sanguinololu 1.3
        #define LCD_PINS_RS      4
        #define LCD_PINS_ENABLE 17
        #define LCD_PINS_D4     30
        #define LCD_PINS_D5     29
        #define LCD_PINS_D6     28
        #define LCD_PINS_D7     27
      #endif

    #else // DOGM SPI LCD Support

      #define DOGLCD_A0         30
      #define LCD_CONTRAST       1

      #if ENABLED(MAKRPANEL)

        #define BEEPER_PIN      29
        #define DOGLCD_CS       17
        #define LCD_BACKLIGHT_PIN 28 // PA3

      #else // !MAKRPANEL

        #define DOGLCD_CS       29

      #endif

    #endif

    // Uncomment screen orientation
    #define LCD_SCREEN_ROT_0
    //#define LCD_SCREEN_ROT_90
    //#define LCD_SCREEN_ROT_180
    //#define LCD_SCREEN_ROT_270

  #else // !DOGLCD

    #define LCD_PINS_RS          4
    #define LCD_PINS_ENABLE     17
    #define LCD_PINS_D4         30
    #define LCD_PINS_D5         29
    #define LCD_PINS_D6         28
    #define LCD_PINS_D7         27

  #endif // !DOGLCD

  #define BTN_EN1               11
  #define BTN_EN2               10

  #if ENABLED(LCD_I2C_PANELOLU2)
    #if IS_MELZI
      #define BTN_ENC           29
      #define LCD_SDSS          30 // Panelolu2 SD card reader rather than the Melzi
    #else
      #define BTN_ENC           30
    #endif
  #else  // !Panelolu2
    #define BTN_ENC             16
    #define LCD_SDSS            28 // Smart Controller SD card reader rather than the Melzi
  #endif // !Panelolu2

  #define SD_DETECT_PIN         -1

#endif // ULTRA_LCD && NEWPANEL`

Tannoo, 2017-03-29:
Yeah, I was looking at that already, I think I have a working copy for you.

Tannoo, 2017-03-29:
[Marlin-RCBugFix-Psyc0Flyer.zip](https://github.com/MarlinFirmware/Marlin/files/880309/Marlin-RCBugFix-Psyc0Flyer.zip)

Tannoo, 2017-03-29:
That is the latest code as of a few hours ago.

Tannoo, 2017-03-29:
You will have to set any options and may have to define probe options (if you have one).

Psyc0Flyer, 2017-03-29:
Configuring now and uploading. Will confirm in a few ;)

Tannoo, 2017-03-29:
It should be configured as the original....just verify that.

Psyc0Flyer, 2017-03-29:
Still nothing :(

Psyc0Flyer, 2017-03-29:
well, it is showing something but not readable.

Tannoo, 2017-03-29:
Hmm. Power it down and re-seat the LCD cables and interface card (if you have one). Try again.

Psyc0Flyer, 2017-03-29:
I have :/ Here is what it looks like which is more than I had before!

Tannoo, 2017-03-29:
You can drag and drop a picture here to upload it.

Tannoo, 2017-03-29:
Is it full screen garbled stuff?
If so, this MAY help. 
In the `Configuration_adv.h` file there is this:
```
  // Enable this option and reduce the value to optimize screen updates.
  // The normal delay is 10µs. Use the lowest value that still gives a reliable display.
  //#define DOGM_SPI_DELAY_US 5
```
You can enabled that and change the value until it works for you.

Psyc0Flyer, 2017-03-29:
[image]

Tannoo, 2017-03-29:
At worse case, you could have a bad LCD and/or cables.

Psyc0Flyer, 2017-03-29:
Doubt it. It works fine with stock FW

Tannoo, 2017-03-29:
Ok. Then maybe there is a pin still not correct.

Tannoo, 2017-03-29:
Set that delay and see if it works.

Psyc0Flyer, 2017-03-29:
so far 20 and still same

Tannoo, 2017-03-29:
In the pins file, change:
from:
`#define LED_PIN          27`
to:
`#define LED_PIN          -1`

Tannoo, 2017-03-29:
Looks like I missed that one.

Psyc0Flyer, 2017-03-29:
comment the delay as before?

Tannoo, 2017-03-29:
You can.... especially as it made no difference.

Psyc0Flyer, 2017-03-29:
made no difference

Tannoo, 2017-03-29:
Well, I am at a loss.

Take a closer look at the pin definitions. I have to get ready for work, atm.

Psyc0Flyer, 2017-03-29:
I am using latest U8glib library also.

Tannoo, 2017-03-29:
Ok.

Tannoo, 2017-03-29:
Take a look at: #5756, #5720.

Not sure if any of that will help.

Bob-the-Kuhn, 2017-03-30:
I had a similar display on my Viki2 when the LCD register select pin was floating.

That would be pin 28 - LCD_PINS_RS

I don't see anything wrong in the files that @Tannoo supplied.

Perhaps there is a second signal assigned to pin 28.

Drop the following files into your new Marlin directory, enable PINS_DEBUGGING in configuration_adv.h and then issue the command M43 I1.  That'll pop up a list of all the pins and the signals assigned to them.

When you drop the files you'll be overwriting two files.
[pinsDebug.zip](https://github.com/MarlinFirmware/Marlin/files/881255/pinsDebug.zip)

Bob-the-Kuhn, 2017-03-30:
I thought I was on to something.  Turns out it was an error in the pin name list.

Please use this corrected file.
[pinsDebug_list new.zip](https://github.com/MarlinFirmware/Marlin/files/881314/pinsDebug_list.new.zip)

I downloaded the new files to my Sanguino and all looks good for the LCD.  Strange but the encoder button doesn't show up in the pin list.

Tannoo, 2017-04-01:
@Psyc0Flyer, any luck?

Take a look at a recent post, #6178.

closed 2017-10-31 by thinkyhead