MarlinFirmware/Marlin #6159: G29 P2 abnormal behavior

WheresWaldo, 2017-03-29:
Despite these lines on the branch readme
```
Note that the command tries to start you in a corner
of the bed where movement will be predictable.  You can force the location to be used in
the distance calculations by using the X and Y parameters.
``` 
The command only places you in the middle of the bed. Nor does the C parameter seem to work when using the B parameter.

If you are not using B the G29 P2 command by itself moves to the middle and allow you to move the hotend up and down. If you add the X and Y parameters no Z axis movement seems possible.

WheresWaldo, 2017-04-03:
I'm closing this until I can retest when UBL is more mature.

closed 2017-04-03 by WheresWaldo

Roxy-3D, 2017-04-03:
@WheresWaldo   The G29 P2 is the least 'mature' part of UBL.   Initially, I thought it would be important in order to fill in the unprobed areas of the mesh.   But because there are better solutions to that,  G29 P2 fell out of favor and I never use P2.   I'm actually thinking about removing the P2.   Especially since 'Manual Probing' is on a trajectory to be a 'Probe Type'.   (The problem is Marlin can only have 1 probe type defined right now.  We need to warm it over so we can do automatic probing with a real probe and use manual probing for the unprobed locations.)

But I'll try to remember to re-read this in the morning and see if I can get G29 P2 into compliance with your expectations.