MarlinFirmware/Marlin #6162: auto bed leveling corrects in the wrong direction

khschmidt, 2017-03-30:
installed Version: 1.1.0 RC8
Sensor BLTouch
Sensoroffset:
#define X_PROBE_OFFSET_FROM_EXTRUDER
#define Y_PROBE_OFFSET_FROM_EXTRUDER -41
#define Z_PROBE_OFFSET_FROM_EXTRUDER -2.08
tested with both of them:
AUTO_BED_LEVELING_LINEAR and AUTO_BED_LEVELING_BILINEAR
I have testet with debug Level on the G29:
in the x direction my bed is slightly increasing about 0,4mm to the right (increasing X)
measured results in x direction: 2,62  2,46  2,26
but if I move in X with G0 X150 starting from X50, Z axis is moving down instead up!

khschmidt, 2017-03-30:
I think I have found the Problem. My BLtouch sensor is -41mm in y-axis away from the nozzle. The 2 rods on the X-axis are not completly vertical. The tilt angle on the left side is 1.5° and on the right side 1.1°
That means: Delta Z = 41*(sin1,5 - sin1,1) = 0,3mm vertical movement of the BLTouch compared to the nozzle.

How to fix this Problem?
First way to put the BLTouch with y-Offset = 0 (same distance to the X-rods of sensor and nozzle)
second way. A linear correction Faktor in Software for Z over X.

khschmidt, 2017-04-06:
Problem is fixed, I mounted the BLTouch in the same Y distance as the nozzle. Y_PROBE_OFFSET_FROM_EXTRUDER 0
now the levelling is working as expected :-)

closed 2017-10-31 by thinkyhead

mzeve, 2017-11-26:
Hey guys, 
I have the same problem and wanted to use the second option to fix the problem: The software correction. Does anyone know the correct variable to be edited?

thinkyhead, 2018-01-26:
@mzeve We have no software correction at this time to fix readings resulting from tilted or twisted X rods. The solution of placing the probe at the same Y distance as the nozzle is the most correct one, as it will exactly compensate for the X rod tilt issue.