MarlinFirmware/Marlin #6163: Bed level menu not working when not previously homed.

JanVelleman, 2017-03-30:
Hi,

I have installed the latest RCbugfix on a delta prototype. The delta has auto bed leveling and a graphical display.

I included a small video to indicate the problem I describe below:

https://1drv.ms/v/s!AkB62gmPn-IZgpNR4gkiI_HDJAzClg

The problem occurs when a you reset the printer and you go to the "level bed" menu item directly (without homing first). When you select that menu item the printer will home first as expected but after the homing operation it will not go into the level bed routine. If you select the "level bed" menu item again (if the screen did not timeout to the info screen first) you will see the menu selector jump up to auto home item and will perform a home operation again without leveling the bed next. The only way to perform a bed level is to home first then exit the menu to either main menu or infoscreen and then select the "level bed" menu item.

Any ideas?

autonumous, 2017-03-30:
I have some very similar symptoms, issue: #6125 

In my case, with what i’ve seen, its like it’s taking a string of commands e.g. “G28\nG29” and somehow  executing the G29 before G28 has completed.

thinkyhead, 2018-02-06:
Presuming fixed.

closed 2018-02-06 by thinkyhead