MarlinFirmware/Marlin #6164: Starting a print from SD CARD but printer pauses the print immediately?

JanVelleman, 2017-03-30:
Hi fresh install of the latest RCBugfix on a Delta proto:

See the video below:

https://1drv.ms/v/s!AkB62gmPn-IZgpNSSaO_jHdU2XLi8w

Why does the print need to be resumed to start printing?

digi2life, 2017-03-30:
I'm having the same issue.

Jartza, 2017-04-01:
Confirmed. Print doesn't "start", instead selecting file brings the printer to "paused" state, and the print needs to be resumed after selecting file.

tjay31584, 2017-04-03:
I just tried to print from an SD card for the first time ever (always printed via USB before) and noticed it wouldn't print at all (or so I thought).  I found this thread which pointed out that it may simply be paused, so I pressed the menu key and sure enough, resume print was an option.  Pressed that and away it went!

Glad it works, but it's still strange behavior.  I'm sure they will get it worked out soon enough.

thinkyhead, 2017-04-03:
I just patched this issue yesterday, stemming from a bug I introduced last week. Are you saying it still exists in the current `RCBugFix`?

Jartza, 2017-04-03:
I just updated to latest commit an hour ago, I can confirm it's fixed.

tjay31584, 2017-04-03:
Ok, I stand corrected.  It is fixed in the most current download.  I just checked it.  Sorry about that!  At least I searched before just randomly asking "why won't it print from my SD card!?" and discovered it was simply paused immediately.

closed 2017-04-07 by thinkyhead

blueeagle69, 2019-06-27:
It appears it doing it on the latest 2.0 bugfix too :(

thinkyhead, 2019-06-28:
Open a new bug report, @blueeagle69. Otherwise we will not follow up.

blueeagle69, 2019-07-04:
Thanks thinkyhead.
The build I am running now starts the print corecctly, Thank you.
Although as stated the menu controls are backwards with a Smart Discount Graphic LCD.
I can open a new report for that if needed 👌

thinkyhead, 2019-07-11:
> menu controls are backwards

Which combination of `REVERSE_ENCODER_DIRECTION` and `REVERSE_MENU_DIRECTION` gives you the correct knob directions for both menus and value editing?

blueeagle69, 2019-07-11:
I have to enable;
#define REVERSE_ENCODER_DIRECTION
#define REVERSE_MENU_DIRECTION
to have menu navigation and value editing(z offset) etc to go in the right direction.
But then the new Start Print dialogue screen etc is backwards.
Hope this helps.

thinkyhead, 2019-07-11:
> But then the new Start Print dialogue screen etc is backwards.

Do you want these modal dialogs to always be:
```
NO <= counter-clockwise ... clockwise => YES
```
Or, would you prefer:
```
NO <= value-down-dir ... value-up-dir => YES
```
?

blueeagle69, 2019-07-11:
I think I would prefer NO counter clockwise, YES clockwise :)
I was trying to visualise it while typing :)
Cheers

thinkyhead, 2019-07-11:
Patched with e357268126.

blueeagle69, 2019-07-11:
Amazing, thank you 👌