MarlinFirmware/Marlin #6165: SD card filenames are capitalized unexpectedly with M20 M29 GCODE commands

denmorru, 2017-03-30:
Upload file named in lowercase to SD over Serial with M28, printer LCD has correct lower case name, but after M29, it is shown uppercase in the files list. The same illegal uppercase returned by M20.

Tested Marlin 1.1.0-RC8/RAMPS1.4./REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

denmorru, 2017-03-30:
Function 
```
void CardReader::ls()  {
  lsAction = LS_SerialPrint;
  root.rewind();
  lsDive("", root);
}
```
found in cardreader.cpp

Blue-Marlin, 2017-03-30:
8.3 filename https://en.wikipedia.org/wiki/8.3_filename
M33 https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L4783

Blue-Marlin, 2017-03-30:
External interface (serial/usb) is consequently 8.3.
For display purpose you can get the long filename by `M33 8.3-name`.

denmorru, 2017-03-30:
` #define LONG_FILENAME_HOST_SUPPORT` enabled for LCD, because (above): `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`.
Current serial speeds (250000) require another protocol for file exchange, making it possible to forget 8.3 and M28/M29.