MarlinFirmware/Marlin #6168: Print issue I can't explain

ruggb, 2017-03-30:
I was printing this, when at the beginning of the wall section the printer abruptly shifted the X&Y position about 1.5mm. Then, as it printed approx 10 more layers it gradually moved back to the correct position. The model did not move, the bed did not move, the belts are tight on the XY platform, power is supplied from a UPS, Slic3r shows the correct view without any jogs. I am using Repetier Host running from a laptop with a good battery. Any idea what I might look for? I just can't picture anything that would cause that immediate shift then a gradual return to the correct co-ords. Running RC8bf 11e892b
[image]

Angel996, 2017-03-31:
Might be your retract settings. When your retract setting is too high and the part of the model is thin, the extruder does not have time to really make it, before plastic is re-supplied to it.

However, on your foto it just looks like when of your steppers messed up. Maybe the feedrate is too high and it just skips steps. On top layer it skipped again, but in the other direction. That, actually, quite ordinary. Since it's working on its margin, it did it twice.

If you were around, you could just pause the print while it was shifted, execute G28 X0 Y0 (home X & Y) command and resume and see how it goes. I don't know if Repetier allows that, but Pronterface can do that.

ruggb, 2017-03-31:
@Angel996 thx - I have been running the same feed rate since day one - Max = 500mm/sec. I don't remember where it started its layer so I can't tell if it added or missed steps. I just finished printing another large model and did not have this issue. The strange thing is that it fully recovered, slowly, and on a layer by layer basis - IE, not while it printing a single layer. The steppers are about the highest torque you can get in a 17 size. If it were a mechanical, data, or program issue, I wouldn't expect it to fully recover and so smoothly. It could be a slicer issue, but it doesn't show in the preview. Since it is a coreXY, and the shift is diagonal, only one motor went off.
Still a total mystery!

ruggb, 2017-04-01:
Well, I tried again with commit 81136c1. Same problem, slightly different shift, same exact layer. What in Marlin could possible do this? That is the only variable. Time to dig out an old load.

Angel996, 2017-04-01:
Have you tried adding G28 command on layer change?

ruggb, 2017-04-01:
NO. My thinking is that since it only does it at that point in the print there must be a cause. It isn't random. If it were random, G28 might have some affect. I tried a newer Marlin load and it changed, but is still at the same layer. I will be trying an older Marlin load later to see if that affects it. If that doesn't change it I will try G28 at that layer.

ruggb, 2017-04-01:
I am printing with commit 11e892b. I am past layer 27, the verticals starts at layer 25. There is no sign of the shift yet. There is an extremely small indent at 26/27 in the face shown above but it does not propagate the perimeter and is about 0.2mm deep. I must look at it with a 3x glass to see it. The model was not re-sliced. Same slice as used previously. Can it be something in Marlin causing the previous anomaly? I can't imagine how, but I am not intimate with the code.

thinkyhead, 2017-04-02:
Check and make sure all the grubscrews on your pulleys are tight.

ruggb, 2017-04-02:
I'm thinking at least part of it is accel and jerk settings. But I am checking my Grubs

billyd60, 2017-04-05:
I had a stepper motor shaft shear off inside the stepper motor and it would work most of the time but occasionally slip.

ruggb, 2017-04-05:
that's sure strange. The issue here is it isn't random. It occurs in the same place to different degrees. Since I went back to 11... it is almost unnoticeable, but still there.

billyd60, 2017-04-05:
It wasn't random. The error seemed repeatable in a pattern, because it took a certain amount of force for the cracked shaft to slip. So certain print moves would cause it and didn't appear random. I suggest replacing the stepper, it's cheap, and they're a good spare to have on the shelf if it turns out to be something else.

ruggb, 2017-04-05:
I do have a spare. but I may be chasing an accel/jerk issue. The fact that it goes out then comes back to the exactly correct location is the mystery. I have some serious ringing around horiz holes, less obvious on corners, but there. I also have a purpose bulge on one side of a wall (which includes the holes) and I am getting a slight bulge on the other side of the wall that should not be there. The above artifact has still not returned, so I guest after I get rid of the ringing I will try the latest commit and see what happens.

billyd60, 2017-04-05:
I suppose if your speed and accelerations are high enough you could be causing slippage in one or more axes. Slow everything down and see if the problem goes away.

closed 2017-10-31 by thinkyhead