MarlinFirmware/Marlin #6169: Using extruder drivers as X and Y motor drivers for CNC

ikidd, 2017-03-30:
In the interests of keeping things cool, I’d like to run X and Y on the E0 and E1 drivers, since I have no need for extruders.

I’ve enabled the two options in configuration_adv.h (X_DUAL_STEPPER_DRIVERS andY_DUAL_STEPPER_DRIVERS) and set EXTRUDERS 0 and MOTHERBOARD BOARD_RAMPS_14_SF Configuration.h but get an array error (presumably because Marlin doesn’t account for not having an extruder?) when compiling.

I can't seem to find a version anyone has posted doing this, so I'm wondering if I'm asking too much of it?

Angel996, 2017-03-31:
Are you asking about DUAL stepper setup, or do you just plan to use extruder drivers for X and Y motors?
I've used DUAL Y setup for my laser cnc with Marlin and it works fine. If you just want to switch drivers (which makes no sense to me, but still), you can simply change pin assignments in the config file.

Bob-the-Kuhn, 2017-04-01:
Marlin insists on having one extruder and it must be assigned to E0.  After that the dual X is assigned to E1, dual Y to E2 & dual Z to E3.

Now the fun part - lying to Marling.

As @Angel996 stated, you can change the pin assignments.  The best place to do that is in the pins_YOUR_BOARD.h file.

Set all the E0 pins to an unused pin.  If you have a 2560 then pin 70 can be used for that.

Set all the E1 pins to the old E0 pins, the E2 pins to the old E1, ...

Angel996, 2017-04-01:
Actually, the 1st vacant extruder is used for whatever DUAL setup you are using. I use DUAL Y only, and it got assigned to E1.

ikidd, 2017-04-03:
Angel: yes, I'd like to use E0 for second X, E1 as second Y.  I get this working fine in Repetier since it doesn't insist on having an extruder, but would rather use Marlin because reasons.

Bob: that was kinda what I figured, so it's a matter of changing the pins you say.  So I have to move the E1 pins to force it to use the E0 as E1, etc.  I'll give that a try then, seems kludgy but makes sense given the insistence on having at least 1 extruder.  Seems odd when there's a board def for the hotend output to control a spindle.

Roxy-3D, 2017-04-03:
> , seems kludgy but makes sense given the insistence on having at least 1 extruder.

Come on now....  Be fair...    Marlin was one of the original firmware's to enable 3D-Printing.   I think I can understand how that assumption happened.

ikidd, 2017-04-03:
I don't think I was being unfair, seems like an assumption was also made when MOTHERBOARD BOARD_RAMPS_14_SF  assumed you were putting a spindle on D10, not a heater.  Someone's baking in CNC control, dropping the E0 for an extra motion motor on something as large as a CNC makes sense and seemed reasonable to me.

But I can work around it as Bob said, I really don't want to use Repetier despite the way it's already there.

Allted, 2017-09-02:
I have an ugly version working now. I'm not sure if I should put it on GitHub until someone takes a look at it as I am sure I messed up some functions. All 5 steppers on the MPCNC homing independently.
http://www.vicious1.com/auto-squaring/

thinkyhead, 2018-02-09:
When Marlin is able to have EXTRUDERS 0 this will be gained automatically. PRs for this are already submitted.

closed 2018-02-09 by thinkyhead