MarlinFirmware/Marlin #6170: UBL's Grid Based Leveling...

Roxy-3D, 2017-03-30:
@Bob-the-Kuhn    I don't know if this idea will interest you...   But the Grid Based Leveling of the UBL Mesh is not in place yet.   I have the code to do the very fast (with a small code size) Least Squares Fit done.   But the code to actually probe the bed has not been started...

You now have expertise at how the Grid Probing works...   If you got the UBL G29 G command to probe the bed (and collect the points in an array).   I'll wire up the Least Squares Fit code to tilt the mesh.

Its just an idea...  And it may not make sense...   But it seems like a valuable place to use the code you were thinking had no purpose.   And in fact...   If a person starts by doing a G29 P0 to zero the mesh,  doing a G29 G next would degenerate into a 'Normal' Grid based leveling scheme when the all zero mesh is tilted.

Bob-the-Kuhn, 2017-03-30:
I'll give it a whirl.  It'll be Friday or Saturday before I can clear up stuff at home and on my current PRs.

Roxy-3D, 2017-03-30:
> I'll give it a whirl. It'll be Friday or Saturday before I can clear up stuff at home and on my current PRs.

Not a problem!!!   And once you have it probing with nice options (like variable number of points and maybe supporting the E option...)   it will take me a few days to get started wiring up the least squares fit.

Bob-the-Kuhn, 2017-04-01:
I've got some time to work on this.

I'm not sure what G29 G does.  

Manually probing an existing mesh is done by G29 P2.

Automatically probing a mesh is done by G29 P1.

Does G29 Gn E do the following?
- Create an invalid mesh n x n on a side (ignores UBL_MESH_NUM_X_POINTS & UBL_MESH_NUM_Y_POINTS)
-- n is variable
- Automatically probe the reachable points & store in the mesh
-- E specifies if the probe is stowed after each probe
- Roxy adds code to do a AUTO_BED_LEVELING_LINEAR type of compensation

Roxy-3D, 2017-04-01:
I don't think G29 G creates a mesh...   I think it should use what ever mesh is loaded and active.   The reason is this:
- If the user has fine tuned mesh for their glass, they probably want to use it.  The reason you would tilt the mesh is because a small piece of dirt (or plastic) ends up under the glass and gives it a little bit of a tilt.   Or maybe even though you have the glass dialed in very accurately, your machine can not hold tight tolerances.  
- If the user does a G29 P0 initially, they will have a mesh filled with 0.000's.    Tilting this empty mesh would be very similar to the Grid Based leveling of today.   But instead of using a 'Correction matrix' to multiply the coordinates against, we would be obtaining the Z-Height correction from the (slightly) tilted mesh.
- The Actual G29 G 4 might probe a 4 x 4 grid across as much of the bed as can be reached.   And if the E is specified, the probe gets retracted and redeployed on each point probed.\

Bob-the-Kuhn, 2017-04-01:
UBL_MESH_NUM_X_POINTS 9
UBL_MESH_NUM_Y_POINTS 9

G29 P0 creates a 9x9 mesh filled with zeros
G29 G4 probes?
1. Probes the innermost 4x4 points that are reachable
2. Probes the outermost 4x deep points that are reachable
3. Probes on the diagonals
4. Probes the outermost reachable points and then tries to evenly space out the remaining points
A. Leaves the other points as is
B. Sets the other points to NAN

Roxy-3D, 2017-04-01:
I think G29 G2 would probe the 4 outer most corners of the bed that are reachable.    The Grid probing has nothing to do with the mesh that is going to be tilted.   We are just trying to get a quick understanding of how the surface of the bed looks.

The one gotcha here is when you probe a point, it could be at a dip or peak in the glass.  So we will need to correct those numbers before feeding them into the Least Square Best Fit algorithm.

I think G29 G3 would once again probe the 4 outer most corners that are reachable.  But then bi-sect each line and probe a point halfway in between.

Bob-the-Kuhn, 2017-04-02:
I'm still not understanding the philosophy behind the G subcommand.

Here's my understanding of UBL.

UBL is meant to replace **ALL** the old leveling systems.

When upgrading to UBL none of the grids/meshes/coefficients of the old systems can be imported into UBL.

**AUTO_BED_LEVELING_BILINEAR**
- Replaced by the G29 P1 auto probing system.
- G29 P1 creates the mesh and then populates as much of it as it can with the probe.
- UBL additionally offers the G29 P2 manually probing system to populate the remainder of the mesh and so cover the entire bed.
- Moves are broken up into sub-moves from one mesh boundary to the next.
- Z compensation is a bilinear calculation based on that rectangle only.
This is the crown jewel of the UBL system.

**MESH_BED_LEVELING** 
- Replaced by the G29 P2 manual probing system.  No auto probing.
- ?? creates the mesh.  Manual probing populates it.
- Moves are broken up into sub-moves from one mesh boundary to the next.
- Z compensation is a bilinear calculation based on that rectangle only.
The advantage of this system is that a probe is not needed. 

**AUTO_BED_LEVELING_LINEAR**
Replaced by the G29 G function
- Assumes the bed is a flat, tilted plane.
- Probe the reachable edges and the interior to populate a rectangular, evenly spaced mesh.
- No manual probing
- Use the mesh to calculate a best fit equation.
- Z compensation depends only on the equation (the mesh is not consulted/used)
The advantage of this method is ?? faster print speeds possible because fewer calculations are needed ??

**AUTO_BED_LEVELING_3POINT**
Replaced by the G29 T function.
- Assumes the bed is a flat, tilted plane.
- Probes 3 points that aren't collinear.
- No manual probing
- Uses the 3 points to calculate a best fit equation.
- Z compensation depends only on the equation (the 3 points are not consulted/used)
The advantage of this method is ?? faster print speeds possible because fewer calculations are needed ??

----

If I'm correct about the G function then we can use the G29 P1 code to populate the mesh. followed by your code to generate the coefficients which feeds a new version of ubl_line_to_destination (?maybe use the old system's compensation code?).

Roxy-3D, 2017-04-02:
> I'm still not understanding the philosophy behind the G subcommand.

and then went on to say:
> AUTO_BED_LEVELING_LINEAR
> - Replaced by the G29 G function
> - Assumes the bed is a flat, tilted plane.
> - Probe the reachable edges and the interior to populate a rectangular, evenly spaced mesh.
> - No [...]

One of the benefits of UBL is it is architected to let you combine the best features of the previous bed leveling schemes.   The original Mesh scheme let you compensate for a non-flat bed.   The 3-Point and Grid (or Auto Bed Leveling Linear)  let you compensate for a tilted bed.    Well...    There is no reason those ideas can not be combined.  What if the print bed is non-flat?  In that case you want a mesh to describe it.   What if the print bed surface can get tilted a little bit?  And what if the tilt varies either because of the printer's inability to hold tolerances or because some dirt gets under the glass bed?  In that case, you would want to tilt the print surface.

It is true the Linear Leveling implements its solution with a different set of math.  It multiplies each coordinate by the correction matrix to map it into a new space.   We wouldn't be doing the math that way.   But we would be mapping each mesh point that describes the print bed surface to a new height using very similar math.   And I'm not sure if you caught the 'error' that is introduced by doing this.  While we can adjust the mesh point height, we are not free to change the X & Y component of the new coordinate.   But this error is very small.   And in fact the person never quit clamps down the glass in the same spot twice anyway.  The very small X & Y error is not going to be visible in the print.

> If I'm correct about the G function then we can use the G29 P1 code to populate the mesh. followed by your code to generate the coefficients which feeds a new version of ubl_line_to_destination (?maybe use the old system's compensation code?).

Generating an accurate mesh to describe your print surface takes time.   So, that is only done when the user needs to calibrate a new piece of glass or there is some other reason to do it.  And those values are saved.     But when that piece of glass is put on the heat pad, it may not be in the exact same position.   On my gMax, I have a rubber heat pad.   I can see where it is supporting the glass because there is a different refraction of the light from where the rubber is pressing against the glass versus just very very close to the glass.    Those 'support' areas change from print to print.   The surface of the glass stays the same, but almost assuredly...   The tilt of the glass has changed a tiny amount.

The G29 G command would quickly probe the specified number of points, and tilt the finely tuned mesh to describe the current position of the glass's surface.

Does that help make things more clear?   We would still be using what ever mesh the user says to use.  It could be all 0.000's or it could be a finely tuned mesh.   But based on the grid probing (2x2, or 3x3 or 4x4 or 5x5) we would tilt that mesh.   And then do a normal mesh based Z-Height compensation.

Bob-the-Kuhn, 2017-04-02:
G29 G function
Assumes there is an existing valid/populated grid
Creates a second grid/array
a) all the X/Y coordinates come from the original grid
b) does a best effort to find the edges of the probeable area 
c) does a best effort to space out the interior points
Probes the points in the second grid/array and saves the Z values for each point
No manual probing
Roxy does her magic - uses second grid/array to modify the original grid
second grid/array goes away - not needed anymore
modified original grid now used like normal

Roxy-3D, 2017-04-02:
Yes...   That is the vision.    

- modified original grid now used like normal

Yes.   It even could be saved if that is desired by the user.   But I suspect it is best to just use it during the print.  And for the next print, you re-probe and adjust for the current tilt.   The reason is several fold.  First, with 4 byte floating point numbers, continued recalculation by re-tilting the mesh again and again, error is going to creep in.   But also...  You wouldn't want each print erasing and re-storing the 'new' mesh in the EEPROM.  The EEPROM does wear out.

Bob-the-Kuhn, 2017-04-04:
I've started working on G.  I have the indexing arrays working.  Next is to put the probing code in.

Roxy-3D, 2017-04-04:
> I have the indexing arrays working.

Hey...  I'm not 100% sure this is true.  But I think you can specify a variable in your declarations such as:
```cpp
float x[n_points][n_points], y[n_points][n_points], z[n_points][n_points];
```
I'm not 100% sure that is true...  But GCC seems to accept that even though that is not in the C standard definition.   If so...  that will allow the person to go as crazy as they want without worries.   If they are low on memory and say they want a 10 x 10 grid, we can try to do it... But the machine might crash.   And if the machine crashes... Well...  The obvious answer is you can't have quit as many points as you specified.

Can we give that a try?   Because then the user doesn't have to specify a max number of point.  It just automatically is what ever the machine can handle.

Bob-the-Kuhn, 2017-04-04:
Here's the first pass at the G29 G code.

There's two versions in the ZIP file.  One has all kinds of debug prints in it.  The other has none.  It's a lot easier to understand the code with all the debug prints removed.

The measured values go into a 2 dimensional array called **z_values_G**.

**grid_G_index_to_xpos** and **grid_G_index_to_ypos** are one dimensional arrays that contain the **MESH** indexes that will be probed.  

To probe the G grid point 1,2 you get the MESH X index from grid_G_index_to_xpos location 1 and the MESH Y index from grid_G_index_to_ypos location 2.   Now you use the standard MESH tools to get the bed X,Y locations and do the probing.  The measured value is then stored in z_values_G[1][2].

If you type G29 Gx O then the debug prints will be turned on.  At the very end are the three prints that will interest you the most.  The first is a standard MAP print but it only shows the z_values_G at the MESH locations.  The second is again a standard MAP but this time the probed points show the difference between the G probed values and the MESH values. The third is the standard MESH map.

----

I think I've implemented code_value per your request.  I've done thjs lots of times with the M43 commands with no problems.
```
if (code_seen('G')) {
      uint8_t grid_size_G = code_has_value() ? code_value_int() : 0;
```

----

I still need to do the following:
- Play with the E flag.  Right now it stows the probe after each point whether E is present or not.

----

Let me know if there's anything else you want me to work on.  

Are there any other options besides E that should be supported?

Roxy-3D, 2017-04-04:
> I think I've implemented code_value per your request. I've done this lots of times with the M43 commands with no problems.

THANK YOU!   I'll start wiring up the back end to it.   I tried to post a .pdf with a description of the new Least Squares Fit algorithm done by Ed Williams for you to look at.  But I couldn't get it to paste in.   But I think he posted it here in December or January if you are interested.   

```cpp
if (code_seen('G')) {
      uint8_t grid_size_G = code_has_value() ? code_value_int() : 0;
```

I'll probably edit this to a default of 3.  Historically...  That has usually been the default.

> Are there any other options besides E that should be supported?

I can't think of anything at this time.   But you know how that goes.  As soon as the code starts being used, there will be users that say "I need this or that."

> Let me know if there's anything else you want me to work on.

If you enjoy doing documentation (I most certainly do not!!!!),   UBL does need a 'Quick Start Guide'.

Bob-the-Kuhn, 2017-04-04:
Please take 15-30 minutes and write up a quick start guide/outline.  I'll take it from there.

I'll walk through it, re-write as needed and add pictures. 

I'll make up a .docx version for you to look over.  Once you're semi happy I'll start the process of adding it to the Wiki and I'll start posting it on issues as needed.

----

FYI - the following is still in UBL_line_to_destination.cpp
`SET_INPUT_PULLUP(66); // Roxy's Left Switch is on pin 66.  Right Switch is on pin 65`

Bob-the-Kuhn, 2017-04-05:
Here's the final line probe_pt line.
`const float measured_z = probe_pt(LOGICAL_X_POSITION(probeX), LOGICAL_Y_POSITION(probeY), code_seen('E'), code_seen('V') ? code_value_int() : 0 );  // takes into account the offsets`

The reason I'm not seeing the E flag have an effect is because I'm running a BLTouch.  The very lowest level routine automatically deploys before and stows after if you have BLTouch enabled.

Bob-the-Kuhn, 2017-04-05:
Doesn't take long to change the "final" version.  Added " && code_has_value()" in case they just send V without a number.

`const float measured_z = probe_pt(LOGICAL_X_POSITION(probeX), LOGICAL_Y_POSITION(probeY), code_seen('E'), (code_seen('V') && code_has_value()) ? code_value_int() : 0 );  // takes into account the offsets`

Roxy-3D, 2017-04-05:
Is it convenient to post an updated .ZIP file with the current state?

Bob-the-Kuhn, 2017-04-05:
Here's the code with the updates.

[G29 G rev B.zip](https://github.com/MarlinFirmware/Marlin/files/900674/G29.G.rev.B.zip)

Roxy-3D, 2017-04-05:
Thanks....    I'll do a diff and get the changes cut in.

Roxy-3D, 2017-04-05:
It's probing!   I don't have it doing anything with the points yet.

closed 2017-10-31 by thinkyhead