MarlinFirmware/Marlin #6171: UBL too large for ATmega1284P?

benlye, 2017-03-30:
I know it's still a work in progress, and not ready for prime-time yet, but I wanted to take a look at UBL on my printer, which has a Melzi-clone board.  When I try to compile with AUTO_BED_LEVELING_UBL instead of AUTO_BED_LEVELING_BILINEAR I have a few compilation issues with the current, but when I eventually get it to compile I get an error that the file is around 13KB to large for the flash on my ATmega1284P.

I've tried disabling nearly every space-using option (all the big stuff like DEBUG_LEVELING_FEATURE and SDSUPPORT is already disabled).  Can you suggest anything else I can do?  

Is there anything yet to be done which will optimize UBL to be smaller, or is it simply unlikely that it will work on an ATmega1284P based board?

Roxy-3D, 2017-03-30:
There are a lot of areas where UBL can be shrunk.   For example...  The G29 What_Command() can be stripped out.    You probably don't need the P0 and P2 commands.   With P0 and P1 gone you can strip out the functions

use_encoder_wheel_to_measure_point() 
measure_business_card_thickness(const float &height_value) 
void manually_probe_remaining_mesh()

But...   Really the right answer is to let us get options in place to turn on or off features in UBL.

benlye, 2017-03-30:
@Roxy-3D, That's cool - I just wanted to see if this was on the radar, and it sounds like it is.  I can see you've all been working hard on it and I'm happy to wait.  Feel free to ping me as/when you're ready for more testers.

trivalik, 2017-05-01:
I miss 1204 bytes without SD card support. It will be not so easy.

benlye, 2017-05-01:
For the last few weeks I had been able to compile with UBL enabled with about 2KB to spare on my 1284p board (with SD support and a number of other non-essential options disabled).  Now it's back to being about 5KB too large again.

Roxy-3D, 2017-05-01:
@benlye   Do you have your mesh fine tuned???   Because it is very easy to turn off the G26 code.  
That is controlled by UBL_G26_MESH_EDITING.    But once things stabilize...  We will be making individual features of UBL selectable.  That isn't hard to do.   Nobody has spent the time yet to do it.

Do you feel like doing that?   The first couple of things I would make selectable would be manual probing of the bed and possibly the G29 O  topography map.    Those should both be easy to wrap in an #ifdef and make the code go away if not enabled.

benlye, 2017-05-01:
@Roxy-3D, no - I compiled with UBL but didn't have the opportunity to actually start setting it up.  I'm not worried at this point, and I'm happy to let you all focus on getting the 1.1.0 release out the door :-)

closed 2017-08-03 by benlye