MarlinFirmware/Marlin #6173: problem with commit de9d2cd

ruggb, 2017-03-31:
commit de9d2cd issues
virgin load won't compile - gives this error
===============
sketch\Marlin_main.cpp:655:26: warning: 'busy_state' was declared 'extern' and later 'static' [-fpermissive]
   static MarlinBusyState busy_state = NOT_BUSY;
                          ^
In file included from sketch\Marlin_main.cpp:220:0:
sketch\Marlin.h:366:26: note: previous declaration of 'busy_state'
   extern MarlinBusyState busy_state;
=================

ALSO

configuration.h line 142 //#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"
Version.h line 86          #define DEFAULT_MACHINE_UUID "cede2a2f-41a2-4748-9b12-c55c62f367ff"

How about moving this from version.h and replacing line 142 with it. 
==================
  /**
   * Default generic printer UUID.
   */
  #define DEFAULT_MACHINE_UUID "cede2a2f-41a2-4748-9b12-c55c62f367ff"
==================
It complies much better when I change the UUID in configuration.h

ruggb, 2017-03-31:
@thinkyhead Thanks works fine now in 81136c1

closed 2017-03-31 by ruggb

thinkyhead, 2017-03-31:
A lot of changes this week, mostly related to Unified Bed Leveling. Sorry for the hiccups!