MarlinFirmware/Marlin #6175: Starting UBL with G29A P1 M hangs on first point

viper93458, 2017-03-31:
Hello,

it seems that when I issue G29A P1 M, the printer homes xyz and then goes to the first point and freezes. There is a message on the LCD that says "Doing G29 UBL !", the machine is frozen and won't respond to the LCD encoder or any further GCodes. It also freezes with the same symptom if I try something like G29 O.

When I watch the communications, it seems to be outputting the mesh over and over and basically staying all 0's.

READ:   0.000    0.000    0.000   -0.008   -0.000   -0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    134217776.000  
READ:  
READ:   0.000    0.000    0.000   -0.000   -0.000   -0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000   -0.000   -0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000echo:busy: processing
READ:     0.000       .        0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000       .        0.000   -0.000   -0.000    0.008    0.000    0.400    0.000  
READ:  
READ:   0.000       .        0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000       .        0.000    0.000   -0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000   -0.008   -0.000   -0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000   -0.000   -0.000   -0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:  -131512.000    0.000    0.000    0.000    0.000   -0.000   -0.000    0.000    0.000    0.000  
READ:  

Kind regards,
William

Roxy-3D, 2017-03-31:
What happens if you reset the printer...   Do a G28   and then give it a G29 W   ?
Hopefully, it will answer with some information that tells us what it thinks about the world.

viper93458, 2017-03-31:
G28 goes fine and then G29 W outputs the following (Also it also wipes out all text from the bottom of the LCD, then hangs:

SENT: G29 W
READ: ubl_eeprom_start=544
READ: Unified Bed Leveling System Version 1.00 Inctive.
READ: 
READ: No Mesh Loaded.
READ: g29_correction_fade_height : 10.00
READ: z_offset: 0.000000
SENT: M105
READ: X-Axis Mesh Points at: 1.0  24.7  48.3  72.0  95.7  119.3  143.0  166.7  190.3  214.0  
READ: Y-Axis Mesh Points at: 1.0  24.7  48.3  72.0  95.7  119.3  143.0  166.7  190.3  214.0  
READ: Kill pin on :41  state:1
READ: 
READ: ubl_state_at_invocation :0
READ: 
READ: ubl_state_recursion_chk :0
READ: 
READ: Free EEPROM space starts at: 0x0220
READ: end of EEPROM              : 0FFF
READ: sizeof(ubl) :  134
READ: 
READ: z_value[][] size: 400
READ: 
READ: EEPROM free for UBL: 0x0DDF
READ: EEPROM can hold 8 meshes.
READ: 
READ: sizeof(ubl.state) : 67
READ: UBL_MESH_NUM_X_POINTS  10
READ: UBL_MESH_NUM_Y_POINTS  10
READ: UBL_MESH_MIN_X         1
READ: UBL_MESH_MIN_Y         1
READ: UBL_MESH_MAX_X         214
READ: UBL_MESH_MAX_Y         214
READ: MESH_X_DIST        23.666666
READ: MESH_Y_DIST        23.666666
READ: Unified Bed Leveling sanity checks passed.
READ: ok

thinkyhead, 2017-03-31:
UBL is receiving a lot of attention right now. A couple of patches have been added in recent hours, , and some bugs were addressed in the process. Download the latest `RCBugFix` and give it another test. If the problem persists, I'm sure we can track it down quickly.

Roxy-3D, 2017-03-31:
> G28 goes fine and then G29 W outputs the following (Also it also wipes out all text from the bottom of the LCD, then hangs:

Can you give it a second G29 W ?   (G28  G29 W  G29 W)  Because the G29 W is one of the very last things that G29 checks to see if it should do.   There isn't much code between that and returning to the command processor.   If you can do two back to back G29's it probably is not the G29 hanging and we will have to look elsewhere for the explanation.

viper93458, 2017-03-31:
I issued the G28 and then sent (2) G29 W commands:

SENT: G29 W
READ: ubl_eeprom_start=544
READ: Unified Bed Leveling System Version 1.00 Inctive.
READ: 
READ: No Mesh Loaded.
READ: g29_correction_fade_height : 10.00
READ: z_offset: 0.000000
SENT: M105
READ: X-Axis Mesh Points at: 1.0  24.7  48.3  72.0  95.7  119.3  143.0  166.7  190.3  214.0  
READ: Y-Axis Mesh Points at: 1.0  24.7  48.3  72.0  95.7  119.3  143.0  166.7  190.3  214.0  
READ: Kill pin on :41  state:1
READ: 
READ: ubl_state_at_invocation :0
READ: 
READ: ubl_state_recursion_chk :0
READ: 
READ: Free EEPROM space starts at: 0x0220
READ: end of EEPROM              : 0FFF
READ: sizeof(ubl) :  134
READ: 
READ: z_value[][] size: 400
READ: 
READ: EEPROM free for UBL: 0x0DDF
READ: EEPROM can hold 8 meshes.
READ: 
READ: sizeof(ubl.state) : 67
READ: UBL_MESH_NUM_X_POINTS  10
READ: UBL_MESH_NUM_Y_POINTS  10
READ: UBL_MESH_MIN_X         1
READ: UBL_MESH_MIN_Y         1
READ: UBL_MESH_MAX_X         214
READ: UBL_MESH_MAX_Y         214
READ: MESH_X_DIST        23.666666
READ: MESH_Y_DIST        23.666666
READ: Unified Bed Leveling sanity checks passed.
READ: ok
SENT: G29 W
READ: ubl_eeprom_start=544
READ: Unified Bed Leveling System Version 1.00 Inctive.
READ: 
READ: No Mesh Loaded.
READ: g29_correction_fade_height : 10.00
READ: z_offset: 0.000000
READ: X-Axis Mesh Points at: 1.0  24.7  48.3  72.0  95.7  119.3  143.0  166.7  190.3  214.0  
READ: Y-Axis Mesh Points at: 1.0  24.7  48.3  72.0  95.7  119.3  143.0  166.7  190.3  214.0  
READ: Kill pin on :41  state:1
READ: 
READ: ubl_state_at_invocation :0
READ: 
READ: ubl_state_recursion_chk :0
READ: 
READ: Free EEPROM space starts at: 0x0220
READ: end of EEPROM              : 0FFF
READ: sizeof(ubl) :  134
READ: 
READ: z_value[][] size: 400
READ: 
READ: EEPROM free for UBL: 0x0DDF
READ: EEPROM can hold 8 meshes.
READ: 
READ: sizeof(ubl.state) : 67
READ: UBL_MESH_NUM_X_POINTS  10
READ: UBL_MESH_NUM_Y_POINTS  10
READ: UBL_MESH_MIN_X         1
READ: UBL_MESH_MIN_Y         1
READ: UBL_MESH_MAX_X         214
SENT: M105
READ: UBL_MESH_MAX_Y         214
READ: MESH_X_DIST        23.666666
READ: MESH_Y_DIST        23.666666
READ: Unified Bed Leveling sanity checks passed.
READ: ok

I will pull down the code that was updated a couple hours after my current version (from last night).

William

viper93458, 2017-03-31:
Latest code produced no joy. I did G28 then G29 W twice just for grins, then G29 P1 M and it is stuck sitting at the first point just like the original description.

SENT: G29 W
READ: ubl.eeprom_start=32
READ: Unified Bed Leveling System Version 1.00 Inctive.
READ: 
READ: No Mesh Loaded.
READ: g29_correction_fade_height : 10.00
READ: z_offset: 0.000000
READ: X-Axis Mesh Points at: 1.0  24.7  48.3  72.0  95.7  119.3  143.0  166.7  190.3  214.0  
READ: Y-Axis Mesh Points at: 1.0  24.7  48.3  72.0  95.7  119.3  143.0  166.7  190.3  214.0  
READ: Kill pin on :41  state:1
READ: 
READ: ubl_state_at_invocation :0
READ: 
READ: ubl_state_recursion_chk :0
READ: 
READ: Free EEPROM space starts at: 0x0020
READ: end of EEPROM              : 0x0FFF
READ: sizeof(ubl) :  1
READ: 
READ: z_value[][] size: 400
READ: 
READ: EEPROM free for UBL: 0x0FDF
READ: EEPROM can hold 10 meshes.
READ: 
READ: sizeof(ubl.state) : 64
READ: UBL_MESH_NUM_X_POINTS  10
READ: UBL_MESH_NUM_Y_POINTS  10
READ: UBL_MESH_MIN_X         1
READ: UBL_MESH_MIN_Y         1
READ: UBL_MESH_MAX_X         214
READ: UBL_MESH_MAX_Y         214
READ: MESH_X_DIST        23.666666
READ: MESH_Y_DIST        23.666666
READ: Unified Bed Leveling sanity checks passed.
READ: ok
SENT: M105
READ: ok T:21.2 /0.0 B:20.6 /0.0 @:0 B@:0
SENT: M105
READ: ok T:21.2 /0.0 B:20.7 /0.0 @:0 B@:0
SENT: M105
READ: ok T:21.2 /0.0 B:20.4 /0.0 @:0 B@:0
SENT: G29 W
READ: ubl.eeprom_start=32
READ: Unified Bed Leveling System Version 1.00 Inctive.
READ: 
READ: No Mesh Loaded.
READ: g29_correction_fade_height : 10.00
READ: z_offset: 0.000000
READ: X-Axis Mesh Points at: 1.0  24.7  48.3  72.0  95.7  119.3  143.0  166.7  190.3  214.0  
READ: Y-Axis Mesh Points at: 1.0  24.7  48.3  72.0  95.7  119.3  143.0  166.7  190.3  214.0  
READ: Kill pin on :41  state:1
READ: 
READ: ubl_state_at_invocation :0
READ: 
READ: ubl_state_recursion_chk :0
READ: 
READ: Free EEPROM space starts at: 0x0020
READ: end of EEPROM              : 0x0FFF
READ: sizeof(ubl) :  1
READ: 
READ: z_value[][] size: 400
READ: 
READ: EEPROM free for UBL: 0x0FDF
READ: EEPROM can hold 10 meshes.
READ: 
READ: sizeof(ubl.state) : 64
READ: UBL_MESH_NUM_X_POINTS  10
READ: UBL_MESH_NUM_Y_POINTS  10
[... 86 lines cut ...]
READ:  
READ:  -0.000    0.000    0.000    0.000    0.000echo:busy: processing
READ:     0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
SENT: M105
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:  
READ:   0.000   -0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000  
READ:

Roxy-3D, 2017-03-31:
Are you using Repetier Host?      If so...  You can not use the M option.   You need to use O to get the map.

viper93458, 2017-03-31:
I am actually using Simplify 3D. Even with not using the M (I.E. G29 P1), the same symptom occurs and the display no longer works and no additional GCodes are processed for any move operations. :(

Roxy-3D, 2017-03-31:
OK...  Let me check things out...

Roxy-3D, 2017-03-31:
@viper93458    When you get a chance...  Can you try changing the very last lines of the gcode_G29() function to this?   The code to change is about 655 lines into UBL_G29.cpp
```cpp
  void gcode_G29() {
  ...
    lcd_reset_alert_level();
    lcd_setstatus("                         ", 0 );
    lcd_quick_feedback();
    ubl_has_control_of_lcd_panel = false;
  }
```

viper93458, 2017-03-31:
I sure can in a couple hours.

viper93458, 2017-04-01:
I don't think my code looked exactly like you are mentioning. I added the lines that are highlighted here, but these seem to be after the closing brace (Image attached).
[image]

Roxy-3D, 2017-04-01:
Those lines were supposed to go at the very very end of the function.   Search for the LEAVE: label duplicate this:
```cpp
    LEAVE:

    lcd_reset_alert_level();
    strcpy(lcd_status_message,"");
    lcd_quick_feedback();

    ubl_has_control_of_lcd_panel = false;
  }
```

viper93458, 2017-04-01:
OK, sorry about that, does this look better?
[image]

Roxy-3D, 2017-04-01:
No...  Take out the #if ENABLED and everything between it and the #endif.   Take out the #endif also.

viper93458, 2017-04-01:
You got it. Here is where it stands, I will compile and test:
[image]

viper93458, 2017-04-01:
Unfortunately the same issue occurs, nothing has changed. The LCD that says "Doing G29 UBL !", the machine is frozen and won't respond to the LCD encoder or any further GCodes. The output just keeps scrolling the mesh with mostly all zeros.

William

Roxy-3D, 2017-04-01:
OK....   I'll have to find a way to duplicate it here.   What code base are you using?   There have been some changes today.

viper93458, 2017-04-01:
Latest from 8 hours ago. :)

 I can redownload now and re run my settings in the configuration.h and retry if you would like.

William

Roxy-3D, 2017-04-01:
> Don't know if it matters but my LCD is the reprap discount full graphic smart controller unit.

Yes...  I'll try to duplicate on a machine with the full graphic smart controller.

viper93458, 2017-04-01:
Just re downloaded the present code and the issue remains as I suspected since I don't think there were any changes between the version I downloaded this morning and the version presently posted. :(

William

Roxy-3D, 2017-04-01:
Please upload your Configuration.h files...  That will give me the best chance of duplicating the problem.   And to cause the problem, you are doing: G28 followed by G29 P1 O Right?

viper93458, 2017-04-01:
I have been doing G28 followed by G29 P1 M and just G29 P1. We tried G29 P1 O earlier in the testing with the same result.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/887271/Configuration.zip)

Tannoo, 2017-04-02:
I got `G29 O1` serial output to display as it's supposed to.
I replaced the whole display_map function with a previous working one:
```
  void unified_bed_leveling::display_map(const int map_type) {
    float f, current_xi, current_yi;
    int8_t i, j;
    UNUSED(map_type);

    if (map_type==0) {
      SERIAL_PROTOCOLLNPGM("\nBed Topography Report:\n");

      SERIAL_ECHOPAIR("(", 0);
      SERIAL_ECHOPAIR(", ", UBL_MESH_NUM_Y_POINTS - 1);
      SERIAL_ECHOPGM(")    ");
    }

    current_xi = ubl.get_cell_index_x(current_position[X_AXIS] + (MESH_X_DIST) / 2.0);
    current_yi = ubl.get_cell_index_y(current_position[Y_AXIS] + (MESH_Y_DIST) / 2.0);

    if (map_type==0) {
      for (i = 0; i < UBL_MESH_NUM_X_POINTS - 1; i++) {
        SERIAL_ECHOPGM("            ");
        #if TX_BUFFER_SIZE>0
          MYSERIAL.flushTX();
        #endif
        delay(15);
      }
      
      SERIAL_ECHOPAIR("(", UBL_MESH_NUM_X_POINTS - 1);
      SERIAL_ECHOPAIR(",", UBL_MESH_NUM_Y_POINTS - 1);
      SERIAL_ECHOLNPGM(")");

      SERIAL_ECHOPAIR("(", UBL_MESH_MIN_X);
      SERIAL_ECHOPAIR(",", UBL_MESH_MAX_Y);
      SERIAL_CHAR(')');
      delay(15);

      for (i = 0; i < UBL_MESH_NUM_X_POINTS - 1; i++) {
        SERIAL_ECHOPGM("            ");
        #if TX_BUFFER_SIZE>0
          MYSERIAL.flushTX();
        #endif
        delay(15);
      }

      SERIAL_ECHOPAIR("(", UBL_MESH_MAX_X);
      SERIAL_ECHOPAIR(",", UBL_MESH_MAX_Y);
      SERIAL_ECHOLNPGM(")");
      delay(15);
    }

    for (j = UBL_MESH_NUM_Y_POINTS - 1; j >= 0; j--) {
      for (i = 0; i < UBL_MESH_NUM_X_POINTS; i++) {
        f = z_values[i][j];

        // is the nozzle here?  if so, mark the number
        if (map_type==0) 
          SERIAL_CHAR(i == current_xi && j == current_yi ? '[' : ' ');

        if (isnan(f))
          if (map_type==0) {
            SERIAL_PROTOCOLPGM("    .    ");
          } else 
            SERIAL_PROTOCOLPGM("NAN");
        else {
          // if we don't do this, the columns won't line up nicely
          if (f>=0.0 && map_type==0) SERIAL_CHAR(' ');
          SERIAL_PROTOCOL_F(f, 3);
          idle();
        }
        if (map_type!=0 && i<UBL_MESH_NUM_X_POINTS-1) 
         SERIAL_PROTOCOLPGM(",");

        #if TX_BUFFER_SIZE>0
          MYSERIAL.flushTX();
        #endif
        delay(15);
        if (map_type==0) {
          if (i == current_xi && j == current_yi) // is the nozzle here? if so, finish marking the number
            SERIAL_CHAR(']');
          else
            SERIAL_PROTOCOL("  ");
          SERIAL_CHAR(' ');
        }
      }
      SERIAL_EOL;
      if (j && map_type==0) { // we want the (0,0) up tight against the block of numbers
        SERIAL_CHAR(' ');
        SERIAL_EOL;
      }
    }

    if (map_type==0) {
      SERIAL_ECHOPAIR("(", int(UBL_MESH_MIN_X));
      SERIAL_ECHOPAIR(",", int(UBL_MESH_MIN_Y));
      SERIAL_ECHOPGM(")    ");

      for (i = 0; i < UBL_MESH_NUM_X_POINTS - 1; i++)  {
        SERIAL_ECHOPGM("            ");
        #if TX_BUFFER_SIZE>0
          MYSERIAL.flushTX();
        #endif
        delay(15);
      }
      SERIAL_ECHOPAIR("(", int(UBL_MESH_MAX_X));
      SERIAL_ECHOPAIR(",", int(UBL_MESH_MIN_Y));
      SERIAL_CHAR(')');
      SERIAL_EOL;

      SERIAL_ECHOPAIR("(", 0);
      SERIAL_ECHOPAIR(",", 0);
      SERIAL_ECHOPGM(")       ");

      for (i = 0; i < UBL_MESH_NUM_X_POINTS - 1; i++) {
        SERIAL_ECHOPGM("            ");
        #if TX_BUFFER_SIZE>0
          MYSERIAL.flushTX();
        #endif
        delay(15);
      }
      SERIAL_ECHOPAIR("(", UBL_MESH_NUM_X_POINTS-1);
      SERIAL_ECHOPAIR(",", 0);
      SERIAL_ECHOLNPGM(")");
    }
  }
```

viper93458, 2017-04-02:
Just tried it and my machine is now moving from point to point. Not sure if it will continue to work as I try UBL or not. :)

Thanks!

-William

thinkyhead, 2017-04-02:
Thanks, testers. I've been working on Marlin without a lot of sleep, and I'm sure mistakes are bound to occur. I have been testing as I go along, but only with my own configuration, and not with delta. So your feedback is essential to finding these errors.

Roxy-3D, 2017-04-02:
@viper93458   Please download and flash the latest RCBugFix.  I believe the Topographical Map is going to work for you.    I've been beating on it for last couple of hours and am currently doing a difficult print.  Everything seems correct.

viper93458, 2017-04-02:
Downloading now. Results in about 15 minutes. :)

Thanks for the hard work!

viper93458, 2017-04-02:
Yep, it is indeed probing now with the current code. I didn't have to run the code fix mentioned previously by Tannoo. I am still trying to fully understand how to get my mesh clean, but at least step 1 is working. :) Thanks!

viper93458, 2017-04-02:
When its done with phase 1 probing it leaves a message on the LCD that says "DEFAULT_". Not sure it matters. I also see that the work inactive is misspelled with you first run G29 W with a mesh inactive. It says "inctive". :)

Roxy-3D, 2017-04-03:
> I also see that the work inactive is misspelled with you first run G29 W with a mesh inactive. It says "inctive". :)

Yes...   I'll fix that right now!   Thanks!

viper93458, 2017-04-03:
This issue appears to be resolved. Thank you very much.

William

viper93458, 2017-04-03:
@Tannoo I totally understand and completely appreciate what you provided previously. :)

William

closed 2017-10-31 by thinkyhead