MarlinFirmware/Marlin #6180: Request: M3 and M5 command support

Angel996, 2017-03-31:
Many people are using Marlin to control CNC routers and laser cutters, however, they have to modify it to support spindle/fire enable/PCM outputs. Maybe, we could make this a feature? It could go something like this:

Marlin_main.cpp
```
inline void gcode_M3() { // enable laser/spindle

  pinMode(LASER_PWM, OUTPUT);
  pinMode(LASER_EN, OUTPUT);

  if (code_seen('S')) {
	uint8_t s = code_value_int(); // ushort
	analogWrite(LASER_PWM, s);
	}

  digitalWrite(LASER_EN, 0);
  }

  inline void gcode_M5() { // disable laser
  pinMode(LASER_EN, OUTPUT);
  pinMode(LASER_PWM, OUTPUT);

  digitalWrite(LASER_EN, 255);

  if (code_seen('S')) {
	uint8_t s = code_value_int(); // ushort
	analogWrite(LASER_PWM, s);
	}
  }

    case 'M': switch (codenum) {
      #if ENABLED(ULTIPANEL) || ENABLED(EMERGENCY_PARSER)
        case 0: // M0: Unconditional stop - Wait for user button press on LCD
        case 1: // M1: Conditional stop - Wait for user button press on LCD
          gcode_M0_M1();
          break;
      #endif // ULTIPANEL

	case 3: // M3: Enable Laser
	  gcode_M3();break;
	case 5: // M5: Disable Laser
	  gcode_M5();break;

```
A typical laser power supply has two control pins: one to enable the laser (active low), another pin to set its power (analog, 0-5v). So, the M3 and M5 commands can be used both to fire the laser (spindle) and set intensity (rpm): `M3 S127` (fire laser at 50% power).

denmorru, 2017-03-31:
Is it new idea? https://github.com/MarlinFirmware/Marlin/pull/5145 look in there.

thinkyhead, 2017-04-01:
I have code for multiple tool support that I've been working on for a client, which will migrate into the main branch eventually, but not before the 1.1.1 release. We're focused on patching up existing issues, catching up on old pull requests, and finalizing things for a 1.1.0 release before the end of the month.

Roxy-3D, 2017-04-01:
> We're focused on patching up existing issues, catching up on old pull requests, and finalizing things for a 1.1.0 release before the end of the month.

You mean before the end of April, right?   Because if you mean the end of March, you only have a few hours left and not time for testers to reply.

closed 2017-10-31 by thinkyhead

luisluna2, 2018-05-02:
@Angel996 you know how change S255 to S24000 for simulate 24000RPM with S????