MarlinFirmware/Marlin #6182: [FR] autocalibration for delta machines

LVD-AC, 2017-03-31:
The calibration of a delta is an iteration process that can be automated. Here's a routine to calibrate the delta height, end-stop adjustments and delta radius. It has been tested thoroughly by a multitude of users in the Flsun FB group. It should work on other deltas as well.

First one needs to enable changing and storing in EEPROM of the delta height like I explained in #4061

Here's the auto-calibration routine to determine the correct delta height, end-stop adjustments and delta radius. It iterates using a convergence matrix until the iteration process overshoots the solution and then rolls back one step.

In configuration.h
 ... add in delta settings:

  #define ENABLE_DELTA_AUTO_CALIBRATION  // enable G33 auto delta calibration (EEPROM must be turned on in order to store results)
  #ifdef ENABLE_DELTA_AUTO_CALIBRATION
    #define DELTA_CALIBRATION_RADIUS (DELTA_PRINTABLE_RADIUS - 10) // set the radius for the calibration probe points
    #define CONVERGENCE_FACTOR 0.75 // <1.00 lower to prevent overshooting solution, higher to speed up convergence
  #endif

In marlin_main.cpp:
... add

  #ifdef ENABLE_DELTA_AUTO_CALIBRATION
    /**
     * G33: Delta '4-point' auto calibration iteration
     * Usage:
     *   G33 <Cn> <Vn>
     *     V (default) = verify calibration (n= 1-4 : n*n probe points, default 4)
     *     C = Calibrate endstops, height and delta radius (n= 1-4 : n*n probe points, default 4)
     *         with n=1 sets height only - usefull when calibrated naked bed and tape is added
     */
    inline void gcode_G33() 
    {
      // **************************************
      // * written by Luc Van Daele 2016-2017 *
      // **************************************
      
        stepper.synchronize();

        #if PLANNER_LEVELING
          set_bed_leveling_enabled(false);
        #endif

        float delta_calibration_radius = DELTA_CALIBRATION_RADIUS;
        float test_precision;
        float Zero_StdDev = 0.0;
        int probe_points = 4;
               
        if(code_seen('C')) {
          probe_points = code_value_int();
          if (probe_points < 1 || probe_points > 4) probe_points = 4;
          Zero_StdDev = 999.00;
        }
        
        if(code_seen('V')) {
          probe_points = code_value_int();
          if (probe_points < 1 || probe_points > 4) probe_points = 4;
          Zero_StdDev = 0.00;
          
        }

        SERIAL_PROTOCOLLNPGM(" ");
        no_report = true;
        gcode_G28();
        
        float abl_probe_Pt_X[13] = {0.0,-SIN_60*delta_calibration_radius,-COS_60*delta_calibration_radius,0.0,COS_60*delta_calibration_radius,
                                   SIN_60*delta_calibration_radius,delta_calibration_radius,SIN_60*delta_calibration_radius,COS_60*delta_calibration_radius,
                                   0.0,-COS_60*delta_calibration_radius,-SIN_60*delta_calibration_radius,-delta_calibration_radius};            
        float abl_probe_Pt_Y[13] = {0.0,-COS_60*delta_calibration_radius,-SIN_60*delta_calibration_radius,-delta_calibration_radius,-SIN_60*delta_calibration_radius,
                                   -COS_60*delta_calibration_radius,0.0,COS_60*delta_calibration_radius,SIN_60*delta_calibration_radius,
                                   delta_calibration_radius,SIN_60*delta_calibration_radius,COS_60*delta_calibration_radius,0.0};
        int iterations = 0;
        
        float Ex_old = endstop_adj[0];
[... 209 lines cut ...]
        clean_up_after_endstop_or_probe_move();
        stepper.synchronize();

        #if ENABLED(Z_PROBE_SLED)
          DEPLOY_PROBE(); }
        #endif // Z_PROBE_SLED
        gcode_G28();      
      }
      while ((Zero_StdDev < test_precision && iterations < 31)); // stop iterations
      no_report = false;
      report_current_position();
    }
  #endif // ENABLE_DELTA_AUTO_CALIBRATION

... and add

        #ifdef ENABLE_DELTA_AUTO_CALIBRATION
        
          case 33: // G33 - Delta '4-point' auto calibration iteration
            gcode_G33();
[Marlin-1.1.0-RC8-AC.zip](https://github.com/MarlinFirmware/Marlin/files/886055/Marlin-1.1.0-RC8-AC.zip)

            break;
            
        #endif // ENABLE_DELTA_AUTO_CALIBRATION

teemuatlut, 2017-03-31:
Can you make this into a pull request against RCBugFix?

LVD-AC, 2017-03-31:
I'll give it a try later on, but I'm new to GitHub and still rather confused by it...

LVD-AC, 2017-03-31:
I might have done it to my own astonishment

thinkyhead, 2017-04-01:
Thanks! We'll continue over on that PR….

closed 2017-04-01 by thinkyhead