MarlinFirmware/Marlin #6183: Step Motors only move one direction

riojaonly, 2017-03-31:
Just completed building a RepRap Prusa I3 printer for my son.

Had no problems with the build and I tuned all the stepper motors to .550 
Inputted all the settings info from Reprapguru instruction manual into Repetier software and then tried to manually control stepper….this is when the problems started.

Both Z stepper motors will only move in a clockwise direction.  And the X stepper motor when I turned on the first time would move in both directions until I hit the X home button.  Now it will only move in one direction as well.

I’m new to this and really don’t have a clue, but reading post for a couple of hours I’d say I have an issue with my endstops?  If I input code M119 I get the following:

09:04:46.020 : Reporting endstop status 
09:04:46.020 : x_min: open 
09:04:46.020 : x_max: TRIGGERED 
09:04:46.020 : y_min: open 
09:04:46.020 : y_max: open 
09:04:46.020 : z_min: open 
09:04:46.020 : z_max: TRIGGERED 
When I try to move the X or Z away from home position via the manual controls in Repetier I get the following error code (it is basically different measurement every time):

08:12:40.150 : echo:endstops hit:  Z:-0.30 
08:12:51.353 : echo:endstops hit:  Z:-0.60 
08:13:02.010 : echo:endstops hit:  Z:-0.66 
08:13:12.432 : echo:endstops hit:  X:0.00 
08:13:31.793 : echo:endstops hit:  X:-16.99

I tried disconnecting my Z endstop from the control board and it made no difference.

Is there a way to re-set the Endstops or something?  Any advice/direction would be greatly appreciated.

Thank you

Roxy-3D, 2017-03-31:
Do you really have both Min and Max endstops on the X, Y & Z axis?   You should only define endstops that exist.

riojaonly, 2017-03-31:
I guess I only have max enstops.  3 mechanical endstops for each axis.  But I did not define these myself....are these definitions set in adrino software?

Roxy-3D, 2017-03-31:
The Configuration.h files come pre-configured for a generic printer.   Most printers use Min endstops...  So those are turned on by default.    Please turn those off and probably things will behave better for you.

ruggb, 2017-03-31:
yes, that is what happens when the endstops are not defined properly. been there, done that.
3 endstops usually are defined as MIN endstops, but you could try to confuse yourself and define them as MAX, then you would have to turn other things around. Best to go with MIN

thinkyhead, 2017-03-31:
Please post your configuration files and someone will check them over for you. You can ZIP them and just drop the ZIP file onto the textfield in your next reply.

riojaonly, 2017-03-31:
I disabled the max endstops and now all the stepper motors are moving in both directions.  Tested the endstops and they're reporting correctly

On to the next the configurations.  If I run into any problems, I will include my config files as thinkyhead suggested.  

Thank you for helping out the newbie

closed 2017-04-03 by riojaonly

ruggb, 2017-04-03:
a pleasure - I've been there and still am sort of.