MarlinFirmware/Marlin #6186: OPPS, babystepping appears to be broken now

ruggb, 2017-03-31:
@Roxy-3D RC8bf commit 81136c1
I can no longer dbl click into babystepping
babystepping and dbl click to babystepping are both enabled.
it is on the tune menu but I can't get to that until it starts moving

Roxy-3D, 2017-04-01:
What kind of electronics board do you have?   Mostly, that commit is for TinyBoy hardware.   I didn't see anything that should affect Double Click Baby Stepping...

ruggb, 2017-04-01:
have a ramps 1.4. It worked in a previous commit.

ruggb, 2017-04-01:
It works in commit 11e892b
I can dbl click when the machine is idle - actually no power except USB
something broke between then and now.

Roxy-3D, 2017-04-01:
Can you do a Binary Search and find the 'bad' commit?   That would help me a lot.   (But I understand if the answer is "No!)     It is my belief the code 3 days ago worked.   I just brought that up my Frankenstein printer.

ruggb, 2017-04-01:
I have a reprap discount full graphic
there is significant code change in ultralcd.cpp on your dbl click routine

this is 11
#if ENABLED(BABYSTEPPING)
  long babysteps_done = 0;
  millis_t status_screen_click_time = 0; <<<<<<<<<< this is missing in 81136c1
  static void lcd_babystep_z();
#endif
............................
this is 11
  /**
   * General function to go directly to a screen
   */
  void lcd_goto_screen(screenFunc_t screen, const uint32_t encoder = 0) {
    #ifdef DOUBLE_CLICK_JUMPS_TO_Z_BABYSTEPPING
      #if ENABLED(BABYSTEPPING) 
        if (currentScreen==lcd_status_screen && screen==lcd_main_menu)  // We are in leaving the status screen to goto the main_menu  
          status_screen_click_time = millis();       // screen.  Mark the time so we know how quick the user is 
                     // pressing buttons. 
        if (currentScreen==lcd_main_menu)  { 
          if ( screen==lcd_status_screen && status_screen_click_time+DOUBLE_CLICK_TIME_WINDOW>millis() ) { 
            lcdDrawUpdate = LCDVIEW_CLEAR_CALL_REDRAW; 
            status_screen_click_time = 0; 
            lcd_babystep_z(); 
          return; 
          } 
        } 
      #endif 
    #endif 
    if (currentScreen != screen) {
      currentScreen = screen;
      encoderPosition = encoder;
      if (screen == lcd_status_screen) {
        defer_return_to_status = false;
        screen_history_depth = 0;
      }
      lcd_implementation_clear();
      #if ENABLED(LCD_PROGRESS_BAR)
        // For LCD_PROGRESS_BAR re-initialize custom characters
        lcd_set_custom_characters(screen == lcd_status_screen);
      #endif
      lcdDrawUpdate = LCDVIEW_CALL_REDRAW_NEXT;
      screen_changed = true;
      #if ENABLED(DOGLCD)
        drawing_screen = false;
      #endif
    }
  ...........................................
this is 81
  /**
   * General function to go directly to a screen
   */
  void lcd_goto_screen(screenFunc_t screen, const uint32_t encoder = 0) {
    if (currentScreen != screen) {

      #if ENABLED(DOUBLECLICK_FOR_Z_BABYSTEPPING) && ENABLED(BABYSTEPPING)
        static millis_t doubleclick_expire_ms = 0;
        // Going to lcd_main_menu from status screen? Remember first click time.
        // Going back to status screen within a very short time? Go to Z babystepping.
        if (screen == lcd_main_menu) {
          if (currentScreen == lcd_status_screen)
            doubleclick_expire_ms = millis() + DOUBLECLICK_MAX_INTERVAL;
        }
        else if (screen == lcd_status_screen && currentScreen == lcd_main_menu && PENDING(millis(), doubleclick_expire_ms))
          screen = lcd_babystep_z;
      #endif

      currentScreen = screen;
      encoderPosition = encoder;
      if (screen == lcd_status_screen) {
        defer_return_to_status = false;
        screen_history_depth = 0;
      }
      lcd_implementation_clear();
      #if ENABLED(LCD_PROGRESS_BAR)
        // For LCD_PROGRESS_BAR re-initialize custom characters
        lcd_set_custom_characters(screen == lcd_status_screen);
      #endif
      lcdDrawUpdate = LCDVIEW_CALL_REDRAW_NEXT;
      screen_changed = true;
      #if ENABLED(DOGLCD)
        drawing_screen = false;
      #endif
    }
  }

Roxy-3D, 2017-04-01:
> I have a reprap discount full graphic
there is significant code change in ultralcd.cpp on your dbl click routine

THANK YOU!!!   That analysis will save me a pile of time tomorrow!

Kaibob2, 2017-04-12:
I think this one can be closed. See #6298

closed 2017-04-12 by thinkyhead