MarlinFirmware/Marlin #6188: Marlin 1.1.0-GM

thinkyhead, 2017-04-01:
Let's make a new branch as a copy of `RCBugFix` to prepare the code for version 1.1.0 release.

Let's also move the `RCBugFix` code into a 1.1.x-dev branch.

The `1.1.0-dino-egg` (or whatever name we like) release-preparation branch will get all the bug-fixes and cleanup patches that we receive (and so will the concurrent 1.1.1 branch). We won't allow any commits of new features into the `1.1.0-dino-egg` branch, and that will keep us more disciplined.

Expediting the release means neglecting some PRs and extant issues, but we can return to them in a couple of weeks after we have a solid release.

And, ya know… we should also start on the 1.2.x branch at or shortly before the 1.1.0 release also, and focus more energy on that, specifically because the file hierarchy will be so much less painful to work with.

Thoughts?

Roxy-3D, 2017-04-01:
Let me change the words to something I can understand.

- Let's do a new feature freeze.
- We can still accommodate new features in a parallel branch.
- We will still make changes to RCBugFix, but those changes will be strictly bug fixes.
- We will target the end of April as a time to tag the existing RCBugFix branch as 'Stable'

Marlin Users...   Please cast a vote if you have an opinion.

emartinez167, 2017-04-01:
+1 for that plan!

Regards,

Ernesto

mperdue, 2017-04-01:
I'm not a contributor, but that sounds like a workable plan.

Bob-the-Kuhn, 2017-04-01:
I support this plan.  I understand the need to delay new features otherwise you'll always be delaying the 1.1.0 release.

I'm not happy that a couple of my PRs that required weeks of effort won't make it into 1.1.0 but ... there isn't enough time to get them out before the users and get enough testing on them before the end of April.

Now if they still haven't made it into release 1.8.0 then ...   ;-)

phil-barrett, 2017-04-01:
Agree. You have to draw a line. Everything else goes into the next release.

christianh17, 2017-04-01:
I totally agree. Without feature freeze you will never get a stable release.

JohnnyTheOne, 2017-04-01:
+1 sounds like a good plan.

apballard, 2017-04-01:
+1 agree on the approach

boelle, 2017-04-01:
+reduce issue list

autonumous, 2017-04-01:
I  disagree.......

Nah  not really :-)  April 1st  lol

Although not a contributor (yet) maybe one day, i agree with that approach.

thinkyhead, 2017-04-02:
> I'm not happy that a couple of my PRs that required weeks of effort

Well, some of the PRs that have been lingering a while, and which seem non-problematic, might still make it in. Just make sure to get them rebased and follow up on review comments.

thinkyhead, 2017-04-02:
> Without feature freeze you will never get a stable release.

Truly. And it's not like it's anyone's fault per se. Most of the PRs we get offer good improvements, and it's simply hard to resist accepting those improvements. The good thing about adding a 1.1 branch separately is we can still move forward on those PRs concurrently, but also call a freeze on 1.1.0-GM.

Bob-the-Kuhn, 2017-04-02:
> I'm not happy that a couple of my PRs that required weeks of effort

What I should have said was "weeks of my time, hours of thinkyhead time". 

I'll add comments to my PRs as to their current state.

thinkyhead, 2017-04-02:
> hours of thinkyhead time

LOL. Sometimes I rush too much for some, apparently!

thinkyhead, 2017-04-02:
> I'm not happy that a couple of my PRs that required weeks of effort

@Bob-the-Kuhn
Well, some of the PRs that have been lingering a while, and which seem non-problematic, might still make it in. Just make sure to get them rebased and follow up on review comments.

thinkyhead, 2017-04-02:
@Bob-the-Kuhn Let's have a chat with @Roxy-3D about getting you some credentials, yes?

Bob-the-Kuhn, 2017-04-02:
I don't know what credentials are.

Roxy-3D, 2017-04-03:
@thinkyhead    If you are OK with it...   I'll give him administrator privileges to MarlinFirmware/Marlin.

Roxy-3D, 2017-04-03:
@thinkyhead    Our 'Teams' are back.   It looks like we can easily invite @Bob-the-Kuhn to the MarlinFirmware organization and pre-select a team.   How about general-maintainers  ?   He will be able to Merge and Revert with that?

Roxy-3D, 2017-04-03:
@Bob-the-Kuhn   Bob, please check your email...   You should have an 'Invite' to the MarlinFirmware organization.     Please be very careful as you click on buttons.   I can speak first hand to how easy it is to make a horrible mistake!

Bob-the-Kuhn, 2017-04-03:
I'm not comfortable doing merges into the main code base.  I still want someone reviewing before they get merged.

I'm OK with being able to update issues & PRs.

Roxy-3D, 2017-04-03:
That is fine.   That is how I try to operate.   The exception is, I'm fully confident on the UBL area and will merge if appropriate.   The thing is, there are some things that mitigate the situation.  For example, if you are correcting an obvious logic error that only affects one part of a seldom used function...  Well...  The amount of damage that can be done is pretty minor.    If you are changing the the way endstops get checked...  It might be best to go VERY SLOWLY.    You get what I'm saying!

Bob-the-Kuhn, 2017-04-03:
I'm in and DANGEROUS!!!

Thanks for the confidence.

Roxy-3D, 2017-04-04:
So...   I need closure.   Are we going to agree that we only put bug fixes into RCBugFix and no new features that disrupt thing?  

Or are we not going to all agree to that?

If we aren't going to agree to that, somebody is going to have to convince me that idea doesn't make sense.   And you better be ready for some strong arguments.

bgort, 2017-04-19:
What was decided here?  

Makes sense to me to put only bug fixes, etc. in RCBugFix and have a separate dev branch for new features, for whatever it's worth.

Roxy-3D, 2017-04-19:
> What was decided here?

> Makes sense to me to put only bug fixes, etc. in RCBugFix and have a separate dev branch for new features, for whatever it's worth.

I think @thinkyhead is still planning on doing the release on April 30th.   So, if that is true, that would imply we be  very careful to only fix things and not break anything for the next 2 weeks.

bgort, 2017-04-19:
Got it.  Thanks.

thinkyhead, 2017-04-22:
Making another branch is always a little bit dangerous 😜  because things get out of sync easily. So I'm just marking all current PRs with the 1.1.1 milestone unless they fix or improve something. Nothing too fancy will be merged before the end of the month. Whatever we have in the `RCBugFix` branch on April 30th will be released as Marlin 1.1.0.

The next trick after release will be keeping `RCBugFix` around a little longer (so we can bring in some of the extant PRs). Then pretty soon afterward we'll rename the branch and re-sort the files to start the 1.2.x branch.

When bug reports start to come in, it would be nice if there was only a single code-base to receive patches. But as soon as 1.2 starts fixes will need to be applied to both. So I think it will be a good idea to do the initial project file rearrangement without making a lot of changes to the code itself and do an initial 1.2.0 release as quickly as possible. Then the 1.1.0 branch can receive only the most serious bug fixes, while we can emphasize 1.2 as the future of cutting-edge development and new features.

Roxy-3D, 2017-04-29:
It looks like we are still on track to promote RC-8's RCBugFix to 'Golden Status' on Sunday, right?

thinkyhead, 2017-05-01:
Ran out of time yesterday. Gathering release notes and other materials, reviewing last-minute patches, etc., and aiming to finish the release before 2am CDT.

thinkyhead, 2017-05-04:
Marlin 1.1.0 is out now… Yieeeee! ✌️ 🏆 🍻 
Excellent work everyone. Take a day off.

https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.0

CONSULitAS, 2017-05-04:
@thinkyhead Congratulations! Great work! 🏁

emartinez167, 2017-05-04:
Great work all!!!

Regards,

Ernesto.

Grogyan, 2017-05-04:
Forget a day off.  Take the rest of the week off.
You guys have done an amazing job over these last two years.  And this latest firmware feels so radically different from even a few months ago, goes to show the amazing work that has gone into it.

Well done everyone!

TGMods, 2017-05-04:
Congratulations to all the contributors and especially to @thinkyhead for putting in so much time and effort to this project.
If I could I'd buy you all a drink.

mroote, 2017-05-04:
Great work everyone!  This release looks amazing and your hard work is appreciated.

stigjoergensen, 2017-05-04:
Great job.

Just one quirk so far, not related to the firmware but the front page og
marlin

There is a link

[image: Billede indsat i tekst 1]
it points to

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/MarlinFirmware/Marlin/releases

Which dosnt exist

Marlin 1.1.0 is out now… Yieeeee! ✌️ 🏆 🍻
> Excellent work everyone. Take a day off.
>
> https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.0
>

CONSULitAS, 2017-05-14:
@thinkyhead Hi Thinkyhead,

just recognized: It has been a really long journey!
https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.0-RC1

Regards

emartinez167, 2017-05-14:
Wow.... Indeed. Thanks for the commitment @thinkyhead and the rest of the team!

thinkyhead, 2017-05-15:
Thanks everyone for your participation and everything you've brought to this event!

When I started making small tweaks to the firmware in late 2014 I had no idea what it would spawn. At that time the project was relatively dormant. The elders of legend had moved on to other projects, and so Marlin languished, awaiting some new fools to fall under its spell. As changes started getting introduced, and as the project became more visible, it drew increasing attention. Since that time, Marlin has continued to receive more and more contributions, adding up to an immensely better user experience and —most importantly— better print results.

Think of all the plastic trees we've saved.

In the lead-up to the release of 1.1.0, with issues and PRs piling up, my role became increasingly patching reported issues, and especially helping get PRs into shape for merging. Holding them for changes, helping the author to fix them up, and then, most of the time copying the branch into my fork for a rebase, additions, changes, and a new PR. I find it's easier to just make the changes than to explain them — the code explains itself. This technique preserves the author on the commits, which is important to tracking user contributions.

It has entailed a huge amount of work, and a daily effort by more and more active contributors. Throughout this very long 28 month sprint it has always felt like a gradual, incremental, and _necessary_ evolution of all the constituent parts of the firmware. There was a long list of requests for long-delayed features, performance improvement, and to iron out bugs. And, really, there just hadn't ever been much of a total cleanup and re-think of the code. So that was all due.

Anyway, Marlin wouldn't be where it is today without everyone who gave their time and expertise. So, thank you all from the bottom of my _quality-obsessed_ heart! The next chapter is going to have a much more solid footing thanks to everyone who took the time to really science the shit out of this thing and redesign its innards to work so much more consistently and efficiently.

The long-awaited code re-organization is coming soon. But first, I need to integrate as many lingering PRs as possible, decide how to cope with the rest, and prepare a 1.1.2 release to get the latest patches out to the world.

This past week I've been filling out the [website](http://marlinfw.org/) (Markdown with Jekyll is fast!). The [GCode section](http://marlinfw.org/meta/gcode/) is 99.9% done and all `Configuration.h` options are [now covered](http://marlinfw.org/docs/configuration/configuration.html).

I just have to add all the `Configuration_adv.h` options to that page, and fill in a few more GCode pages, and the core foundation of the site will be done! I'll continue to add more pages over the coming weeks and polish up the pages pertaining to the most important features of Marlin.

We need more step-by-step guides with images, photographs, illustrations, tables, graphs, etc. If anyone wants to contribute photos or other imagery for step-by-step procedures like bed leveling, filament change, and so on, please feel free to drop them in an issue!

landodragon141, 2018-03-18:
Is there any reason this is still open?

closed 2018-03-19 by thinkyhead