MarlinFirmware/Marlin #6189: RE LCD backlight

ruggb, 2017-04-01:
I have a reprap full graphic display
It would be nice if LCD activity kept the backlight on but I don't know if that can be implemented with this display since the delay is adjusted on the unit and it is turned on with button on unit.
Anybody know?

Tannoo, 2017-04-01:
I have the same display and my backlight is on all the time.

I did not do anything special. So, I'm not quite sure why mine is on all the time, nor why yours would not.

ruggb, 2017-04-01:
there is an adjustment for the delay. If I turn it all the way up (or down maybe) it is on permanently.
Actually, I could not get it to adjust where I want it and had to add a resistor. I thought once I saw provision for a BL logic input, but maybe not

Tannoo, 2017-04-01:
I found that I don't have that display. Mine doesn't have the "light" button, nor the timeout delay pot.

I have found though, that on those models, there is a jumper on the back that will keep the backlight on all the time.

thinkyhead, 2017-04-02:
There's no software interface to the backlight on that display.

closed 2017-04-02 by thinkyhead

ruggb, 2017-04-02:
Thx. Maybe I will just increase that resistor value. Another days project.